Ysurac/openmptcprouter #3091: Apparent Connection Leak / Excessive "Established Connections"

xzjq, 2023-12-27:
## Expected Behavior
"Fewer" connections between OMR and the VPS during normal, routine usage of 6.1-based OMR.

## Current Behavior
The established connections page indicates thousands of connections are being used. As of the moment of this ticket 5269 connections are listed. Once I saw approximately 7500 connections. This appears to be a connection leak, though periodically it will reduce down to "only" about 1500 connections.

This was tangentially referenced in #3068.

No strange use (i.e. no P2P/torrenting)

## Possible Solution
This seems to be related to the 6.1-based MPTCP implementation. The same local hardware & VPS provider shows a reasonable set of connections on a 5.4-based instance.

## Steps to Reproduce the Problem
Connections apparently leak over time/usage. This happens invariably after uptime. It has recurred on multiple local OMR VMs and multiple VPSs.

## Context (Environment)
Here are some statistics from the current state of the OMR/VPS:
Current connections: 5269
Current connections to port 65101 on the VPS: 5268
Current connections using mptcp designated master WAN: 5085
Current connections using WAN_2: 184
Current connections using WAN_3: 0
Current connections in ESTAB state: 5245 (remaining connections are in FIN-WAIT-1, -2, or SYN-SENT... all are using master WAN)

## Specifications

  - OpenMPTCProuter version: 6.1-based snapshot, of the version of the file that was available for download on 20 Dec 2023 (I can give you the shasum of the snapshot file I used, if you prefer)
  - OpenMPTCProuter VPS version: test version, installed on Debian 12 using the version of the script that was available on 14 Dec 2023
  - OpenMPTCProuter platform: x86_64

Ysurac, 2023-12-27:
What is the proxy used ?

xzjq, 2023-12-27:
The basic configuration Shadowsocks

xzjq, 2023-12-27:
I don't know if this is related, but per `journalctl` on the VPS there were 4484 lines of this type in the past 24 hr period:
Dec 26 23:57:20 vps ss-server[738]: server recv: Broken pipe

(Nothing special about that 24 hr period, just using it as a statistic to give an idea of rate... there are 24,586 "broken pipe" entries since Dec 19)

xzjq, 2023-12-28:
I wonder how many connections your system is creating before the VPS provider decides it is DDOS.

My `'server recv: Broken pipe'` count in `journalctl` is up to 29,015 since 19 Dec... another ~5k in the last 24 hrs.

The established connections page currently shows 1,232 current connections, though `ss -4 | grep -c "mptcp ESTAB"` is 2,344.

Ysurac, 2023-12-28:
@xzjq You can try Shadowsocks-Rust to check if you have same issue or not.

xzjq, 2023-12-28:
Okay, just did that and rebooted the OMR instance and VPS. In the first couple minutes of uptime the connection count has swelled from 780 to 1300 and now 2225. So, yes, same issue.  

Interestingly, ss-rust has almost all of these connections in `FIN-WAIT-2` rather than `ESTAB`.
<hr>
Also, the logging for ss-rust in `journalctl` on the VPS is "copious" (i.e. is too fast to read). There are ~25 log entries per second related to the following:

Dec 28 01:48:05 vps shadowsocks-go[520]: 2023-12-28T01:48:05.349-0800 INFO zerocopy/tcp.go:106 Forcing RST on TCP connection {"server": "ss-2022", "listener": 0, "listenAddress": "0.0.0.0:65280", "clientAddress": "9>
Dec 28 01:48:05 vps shadowsocks-go[520]: 2023-12-28T01:48:05.357-0800 INFO service/tcp.go:263 Two-way relay completed {"server": "ss-2022", "listener": 0, "listenAddress": "0.0.0.0:65280", "clientAddress": "98.97.x.x>
Dec 28 01:48:05 vps shadowsocks-go[520]: 2023-12-28T01:48:05.394-0800 INFO service/tcp.go:263 Two-way relay completed {"server": "ss-2022", "listener": 0, "listenAddress": "0.0.0.0:65280", "clientAddress": "98.97.x.x>
Dec 28 01:48:05 vps shadowsocks-go[520]: 2023-12-28T01:48:05.432-0800 INFO service/tcp.go:263 Two-way relay completed {"server": "ss-2022", "listener": 0, "listenAddress": "0.0.0.0:65280", "clientAddress": "98.97.x.x>
Dec 28 01:48:05 vps shadowsocks-go[520]: 2023-12-28T01:48:05.444-0800 INFO service/tcp.go:246 Two-way relay started {"server": "ss-2022", "listener": 0, "listenAddress": "0.0.0.0:65280", "clientAddress": "98.97.x.x>
Dec 28 01:48:05 vps shadowsocks-go[520]: 2023-12-28T01:48:05.465-0800 INFO service/tcp.go:263 Two-way relay completed {"server": "ss-2022", "listener": 0, "listenAddress": "0.0.0.0:65280", "clientAddress": "98.97.x.x>
Dec 28 01:48:05 vps shadowsocks-go[520]: 2023-12-28T01:48:05.505-0800 INFO service/tcp.go:263 Two-way relay completed {"server": "ss-2022", "listener": 0, "listenAddress": "0.0.0.0:65280", "clientAddress": "98.97.x.x>
Dec 28 01:48:05 vps shadowsocks-go[520]: 2023-12-28T01:48:05.546-0800 INFO service/tcp.go:263 Two-way relay completed {"server": "ss-2022", "listener": 0, "listenAddress": "0.0.0.0:65280", "clientAddress": "98.97.x.x>
Dec 28 01:48:05 vps shadowsocks-go[520]: 2023-12-28T01:48:05.585-0800 INFO service/tcp.go:263 Two-way relay completed {"server": "ss-2022", "listener": 0, "listenAddress": "0.0.0.0:65280", "clientAddress": "98.97.x.x>
Dec 28 01:48:05 vps shadowsocks-go[520]: 2023-12-28T01:48:05.621-0800 INFO service/tcp.go:263 Two-way relay completed {"server": "ss-2022", "listener": 0, "listenAddress": "0.0.0.0:65280", "clientAddress": "98.97.x.x>
Dec 28 01:48:05 vps shadowsocks-go[520]: 2023-12-28T01:48:05.721-0800 INFO service/tcp.go:246 Two-way relay started {"server": "ss-2022", "listener": 0, "listenAddress": "0.0.0.0:65280", "clientAddress": "98.97.x.x>
Dec 28 01:48:06 vps shadowsocks-go[520]: 2023-12-28T01:48:06.081-0800 INFO service/tcp.go:246 Two-way relay started {"server": "ss-2022", "listener": 0, "listenAddress": "0.0.0.0:65280", "clientAddress": "98.97.x.x>
Dec 28 01:48:06 vps shadowsocks-go[520]: 2023-12-28T01:48:06.232-0800 INFO service/tcp.go:246 Two-way relay started {"server": "ss-2022", "listener": 0, "listenAddress": "0.0.0.0:65280", "clientAddress": "98.97.x.x>
Dec 28 01:48:06 vps shadowsocks-go[520]: 2023-12-28T01:48:06.361-0800 INFO service/tcp.go:246 Two-way relay started {"server": "ss-2022", "listener": 0, "listenAddress": "0.0.0.0:65280", "clientAddress": "98.97.x.x>
Dec 28 01:48:06 vps shadowsocks-go[520]: 2023-12-28T01:48:06.716-0800 INFO service/tcp.go:246 Two-way relay started {"server": "ss-2022", "listener": 0, "listenAddress": "0.0.0.0:65280", "clientAddress": "98.97.x.x>
Dec 28 01:48:06 vps shadowsocks-go[520]: 2023-12-28T01:48:06.944-0800 INFO service/tcp.go:246 Two-way relay started {"server": "ss-2022", "listener": 0, "listenAddress": "0.0.0.0:65280", "clientAddress": "98.97.x.x>
Dec 28 01:48:07 vps shadowsocks-go[520]: 2023-12-28T01:48:07.006-0800 INFO service/tcp.go:246 Two-way relay started {"server": "ss-2022", "listener": 0, "listenAddress": "0.0.0.0:65280", "clientAddress": "98.97.x.x>
Dec 28 01:48:07 vps shadowsocks-go[520]: 2023-12-28T01:48:07.166-0800 INFO service/tcp.go:246 Two-way relay started {"server": "ss-2022", "listener": 0, "listenAddress": "0.0.0.0:65280", "clientAddress": "98.97.x.x>

Ysurac, 2023-12-28:
The lines are not complete, what do you have after "clientAddress" part ? the targetAddress should be what you try to reach.
It's seems that omr-tracker is doing too much requests  in my case.

Ysurac, 2023-12-28:
You can also put a lower timeout for these connections like ```sysctl -w net.ipv4.tcp_fin_timeout=30```

xzjq, 2023-12-28:
Here is one second worth of VPS logs. I redacted the public IP of the WAN. If you need a different sample let me know, but this was already 60 lines:

Dec 28 02:45:26 vps shadowsocks-go[520]: 2023-12-28T02:45:26.043-0800 INFO service/tcp.go:246 Two-way relay started {"server": "ss-2022", "listener": 0, "listenAddress": "0.0.0.0:65280", "clientAddress": "98.97.x.x:47109", "username": "openmptcprouter", "targetAddress": "98.97.x.x:5900", "client": "direct", "initialPayloadLength": 0}
Dec 28 02:45:26 vps shadowsocks-go[520]: 2023-12-28T02:45:26.052-0800 INFO service/tcp.go:263 Two-way relay completed {"server": "ss-2022", "listener": 0, "listenAddress": "0.0.0.0:65280", "clientAddress": "98.97.x.x:46919", "username": "openmptcprouter", "targetAddress": "98.97.x.x:5900", "client": "direct", "nl2r": 0, "nr2l": 0}
Dec 28 02:45:26 vps shadowsocks-go[520]: 2023-12-28T02:45:26.100-0800 INFO service/tcp.go:263 Two-way relay completed {"server": "ss-2022", "listener": 0, "listenAddress": "0.0.0.0:65280", "clientAddress": "98.97.x.x:46719", "username": "openmptcprouter", "targetAddress": "98.97.x.x:5900", "client": "direct", "nl2r": 0, "nr2l": 0}
Dec 28 02:45:26 vps shadowsocks-go[520]: 2023-12-28T02:45:26.133-0800 INFO service/tcp.go:246 Two-way relay started {"server": "ss-2022", "listener": 0, "listenAddress": "0.0.0.0:65280", "clientAddress": "98.97.x.x:47213", "username": "openmptcprouter", "targetAddress": "98.97.x.x:8078", "client": "direct", "initialPayloadLength": 56}
Dec 28 02:45:26 vps shadowsocks-go[520]: 2023-12-28T02:45:26.133-0800 INFO service/tcp.go:246 Two-way relay started {"server": "ss-2022", "listener": 0, "listenAddress": "0.0.0.0:65280", "clientAddress": "98.97.x.x:47205", "username": "openmptcprouter", "targetAddress": "98.97.x.x:8078", "client": "direct", "initialPayloadLength": 16}
Dec 28 02:45:26 vps shadowsocks-go[520]: 2023-12-28T02:45:26.134-0800 INFO service/tcp.go:263 Two-way relay completed {"server": "ss-2022", "listener": 0, "listenAddress": "0.0.0.0:65280", "clientAddress": "98.97.x.x:46559", "username": "openmptcprouter", "targetAddress": "98.97.x.x:5900", "client": "direct", "nl2r": 0, "nr2l": 0}
Dec 28 02:45:26 vps shadowsocks-go[520]: 2023-12-28T02:45:26.180-0800 INFO service/tcp.go:263 Two-way relay completed {"server": "ss-2022", "listener": 0, "listenAddress": "0.0.0.0:65280", "clientAddress": "98.97.x.x:46387", "username": "openmptcprouter", "targetAddress": "98.97.x.x:5900", "client": "direct", "nl2r": 0, "nr2l": 0}
Dec 28 02:45:26 vps shadowsocks-go[520]: 2023-12-28T02:45:26.180-0800 INFO service/tcp.go:246 Two-way relay started {"server": "ss-2022", "listener": 0, "listenAddress": "0.0.0.0:65280", "clientAddress": "98.97.x.x:47121", "username": "openmptcprouter", "targetAddress": "98.97.x.x:5900", "client": "direct", "initialPayloadLength": 0}
Dec 28 02:45:26 vps shadowsocks-go[520]: 2023-12-28T02:45:26.218-0800 INFO service/tcp.go:263 Two-way relay completed {"server": "ss-2022", "listener": 0, "listenAddress": "0.0.0.0:65280", "clientAddress": "98.97.x.x:46229", "username": "openmptcprouter", "targetAddress": "98.97.x.x:5900", "client": "direct", "nl2r": 0, "nr2l": 0}
Dec 28 02:45:26 vps shadowsocks-go[520]: 2023-12-28T02:45:26.239-0800 WARN service/tcp.go:149 Failed to complete handshake with client {"server": "ss-2022", "listener": 0, "listenAddress": "0.0.0.0:65280", "clientAddress": "98.97.x.x:47217", "error": "EOF"}
Dec 28 02:45:26 vps shadowsocks-go[520]: 2023-12-28T02:45:26.240-0800 INFO zerocopy/tcp.go:106 Forcing RST on TCP connection {"server": "ss-2022", "listener": 0, "listenAddress": "0.0.0.0:65280", "clientAddress": "98.97.x.x:47217"}
Dec 28 02:45:26 vps shadowsocks-go[520]: 2023-12-28T02:45:26.250-0800 INFO service/tcp.go:263 Two-way relay completed {"server": "ss-2022", "listener": 0, "listenAddress": "0.0.0.0:65280", "clientAddress": "98.97.x.x:47109", "username": "openmptcprouter", "targetAddress": "98.97.x.x:5900", "client": "direct", "nl2r": 0, "nr2l": 0}
Dec 28 02:45:26 vps shadowsocks-go[520]: 2023-12-28T02:45:26.260-0800 INFO service/tcp.go:246 Two-way relay started {"server": "ss-2022", "listener": 0, "listenAddress": "0.0.0.0:65280", "clientAddress": "98.97.x.x:47139", "username": "openmptcprouter", "targetAddress": "98.97.x.x:5900", "client": "direct", "initialPayloadLength": 0}
Dec 28 02:45:26 vps shadowsocks-go[520]: 2023-12-28T02:45:26.261-0800 INFO service/tcp.go:246 Two-way relay started {"server": "ss-2022", "listener": 0, "listenAddress": "0.0.0.0:65280", "clientAddress": "98.97.x.x:47167", "username": "openmptcprouter", "targetAddress": "98.97.x.x:5901", "client": "direct", "initialPayloadLength": 0}
Dec 28 02:45:26 vps shadowsocks-go[520]: 2023-12-28T02:45:26.264-0800 INFO service/tcp.go:246 Two-way relay started {"server": "ss-2022", "listener": 0, "listenAddress": "0.0.0.0:65280", "clientAddress": "98.97.x.x:47141", "username": "openmptcprouter", "targetAddress": "98.97.x.x:5900", "client": "direct", "initialPayloadLength": 0}
Dec 28 02:45:26 vps shadowsocks-go[520]: 2023-12-28T02:45:26.264-0800 INFO service/tcp.go:263 Two-way relay completed {"server": "ss-2022", "listener": 0, "listenAddress": "0.0.0.0:65280", "clientAddress": "98.97.x.x:46097", "username": "openmptcprouter", "targetAddress": "98.97.x.x:5900", "client": "direct", "nl2r": 0, "nr2l": 0}
[... cut ...]
Dec 28 02:45:26 vps shadowsocks-go[520]: 2023-12-28T02:45:26.943-0800 INFO service/tcp.go:246 Two-way relay started {"server": "ss-2022", "listener": 0, "listenAddress": "0.0.0.0:65280", "clientAddress": "98.97.x.x:47267", "username": "openmptcprouter", "targetAddress": "98.97.x.x:5900", "client": "direct", "initialPayloadLength": 0}
Dec 28 02:45:26 vps shadowsocks-go[520]: 2023-12-28T02:45:26.980-0800 INFO service/tcp.go:263 Two-way relay completed {"server": "ss-2022", "listener": 0, "listenAddress": "0.0.0.0:65280", "clientAddress": "98.97.x.x:46891", "username": "openmptcprouter", "targetAddress": "98.97.x.x:3128", "client": "direct", "nl2r": 0, "nr2l": 0}

xzjq, 2023-12-28:
Per the Established Connections page, there are currently 2026 connections in FIN-WAIT. The default system net.ipv4.tcp_fin_timeout is 60. I will set it to 30.

Ysurac, 2023-12-28:
```net.ipv4.tcp_fin_timeout``` was 80 by default on one of my VPS, I will fix it to 30 via sysctl for next VPS install.
Many connections have targetAddress that seems to be same as clientAddress for port 5900 (VNC ?) there is also some connections to 3128 (proxy ?).
Maybe there is a loop somewhere...

xzjq, 2023-12-28:
It doesn't appear that setting it to 30 has made an appreciable difference. Here's the current breakdown of the states listed on the "Established Connections" page:
`1674 FIN-WAIT-2` 
`  149 ESTAB `
` 125 CLOSE-WAIT` 
`   8 FIN-WAIT-1` 

<hr>
As for the connections to certain ports, I am unaware of why that would be. This connection leak doesn't occur on the 5.4-based OMR/VPS installations. I don't use VNC. Things that do get installed on the VPS/OMR (both on 5.4 and 6.1-based) are basic utilities: vim, screen, htop, bmon, atop, and qemu-ga (if virtualized; the VPS in this ticket is *not*).

xzjq, 2023-12-29:
FWIW, the OMR isn't listening on port 5900 or 3128 either.

I switched to v2ray/VMESS; here's another 1 second worth of VPS logs (no log activity on OMR at this time and again the IP being referenced is the master WAN on OMR):
Dec 29 04:55:21 vps v2ray[523]: 2023/12/29 04:55:21 98.97.x.x:10371 accepted tcp:98.97.x.x:5900 [direct] email: openmptcprouter
Dec 29 04:55:21 vps v2ray[523]: 2023/12/29 04:55:21 98.97.x.x:10369 accepted tcp:98.97.x.x:5900 [direct] email: openmptcprouter
Dec 29 04:55:21 vps v2ray[523]: 2023/12/29 04:55:21 98.97.x.x:10381 accepted tcp:98.97.x.x:5900 [direct] email: openmptcprouter
Dec 29 04:55:21 vps v2ray[523]: 2023/12/29 04:55:21 98.97.x.x:10395 accepted tcp:98.97.x.x:5900 [direct] email: openmptcprouter
Dec 29 04:55:21 vps v2ray[523]: 2023/12/29 04:55:21 98.97.x.x:10399 accepted tcp:98.97.x.x:5900 [direct] email: openmptcprouter
Dec 29 04:55:21 vps v2ray[523]: 2023/12/29 04:55:21 98.97.x.x:10403 accepted tcp:98.97.x.x:5900 [direct] email: openmptcprouter
Dec 29 04:55:21 vps v2ray[523]: 2023/12/29 04:55:21 98.97.x.x:10405 accepted tcp:98.97.x.x:5900 [direct] email: openmptcprouter
Dec 29 04:55:21 vps v2ray[523]: 2023/12/29 04:55:21 98.97.x.x:10413 accepted tcp:98.97.x.x:5900 [direct] email: openmptcprouter
Dec 29 04:55:21 vps v2ray[523]: 2023/12/29 04:55:21 98.97.x.x:10431 accepted tcp:98.97.x.x:7170 [direct] email: openmptcprouter
Dec 29 04:55:21 vps v2ray[523]: 2023/12/29 04:55:21 98.97.x.x:10423 accepted tcp:98.97.x.x:5900 [direct] email: openmptcprouter
Dec 29 04:55:21 vps v2ray[523]: 2023/12/29 04:55:21 98.97.x.x:10425 accepted tcp:98.97.x.x:5900 [direct] email: openmptcprouter
Dec 29 04:55:21 vps v2ray[523]: 2023/12/29 04:55:21 98.97.x.x:10433 accepted tcp:98.97.x.x:5900 [direct] email: openmptcprouter
Dec 29 04:55:21 vps v2ray[523]: 2023/12/29 04:55:21 98.97.x.x:10447 accepted tcp:98.97.x.x:5900 [direct] email: openmptcprouter
Dec 29 04:55:21 vps v2ray[523]: 2023/12/29 04:55:21 98.97.x.x:10461 accepted tcp:98.97.x.x:7170 [direct] email: openmptcprouter
Dec 29 04:55:21 vps v2ray[523]: 2023/12/29 04:55:21 98.97.x.x:10465 accepted tcp:98.97.x.x:5900 [direct] email: openmptcprouter
Dec 29 04:55:21 vps v2ray[523]: 2023/12/29 04:55:21 98.97.x.x:10475 accepted tcp:98.97.x.x:5900 [direct] email: openmptcprouter
Dec 29 04:55:21 vps v2ray[523]: 2023/12/29 04:55:21 98.97.x.x:10487 accepted tcp:98.97.x.x:7170 [direct] email: openmptcprouter
Dec 29 04:55:21 vps v2ray[523]: 2023/12/29 04:55:21 98.97.x.x:10495 accepted tcp:98.97.x.x:5900 [direct] email: openmptcprouter
Dec 29 04:55:21 vps v2ray[523]: 2023/12/29 04:55:21 98.97.x.x:10525 accepted tcp:98.97.x.x:7170 [direct] email: openmptcprouter
Dec 29 04:55:21 vps v2ray[523]: 2023/12/29 04:55:21 98.97.x.x:10509 accepted tcp:98.97.x.x:5900 [direct] email: openmptcprouter
Dec 29 04:55:21 vps v2ray[523]: 2023/12/29 04:55:21 98.97.x.x:10535 accepted tcp:98.97.x.x:5900 [direct] email: openmptcprouter
Dec 29 04:55:21 vps v2ray[523]: 2023/12/29 04:55:21 98.97.x.x:10541 accepted tcp:98.97.x.x:7170 [direct] email: openmptcprouter
Dec 29 04:55:21 vps v2ray[523]: 2023/12/29 04:55:21 98.97.x.x:10551 accepted tcp:98.97.x.x:5900 [direct] email: openmptcprouter
Dec 29 04:55:21 vps v2ray[523]: 2023/12/29 04:55:21 98.97.x.x:10555 accepted tcp:98.97.x.x:5900 [direct] email: openmptcprouter

Ysurac, 2023-12-29:
As I said there is a loop: connections from LAN exit via Proxy to reach Master WAN IP.
Something from your internal network is doing many connections to port 5900 of master wan IP.
This will be the same with all proxy.
You can try to disconnect all clients one by one to check which one is doing these connections.

xzjq, 2023-12-29:
Ah, so if I understand correctly you are saying that the OMR routing table is configured to route traffic to the local WAN IP via the tunnel and have it bounce back, which doesn't work. This could potentially explain why the 5.4-based OMR does not exhibit this behavior.

BTW, a salient fact here is that this WAN IP is a public IP rather than a reserved private network subnet IP w/ NAT like most (including the other two WANs used here). That is to say, the IP address & WAN address are the same (c.f. diagram w/ Starlink_2). Does OMR implicitly expect WAN & public IP to be different values?

I switched master and the logflood continues to involve just this one WAN IP address (which is no longer the master).

I will try manually adjusting the routing table and/or firewall.
[image]

xzjq, 2023-12-30:
Okay, I moved all my client traffic over to the 5.4-based instance of OMR & VPS and rebooted the 6.1-based OMR. Now, there is no LAN client traffic routed via this 6.1-based OMR & VPS at all. The VPS is spewing all the log messages to that public/WAN IP still.

The *only* connections from my LAN subnet to this 6.1-based OMR are two: one to the OMR:443 (luci, to monitor status) and one to OMR:22 (SSH).

Therefore, these thousands of loop/requests must be generated within the OMR instance. I have no clue why the 6.1-based OMR instances open thousands of connections to try to access port 5900 on this one public WAN IP, and furthermore route that via tun0.

xzjq, 2023-12-30:
To further confirm: I opened a serial console to the 6.1-based OMR and took the LAN interface offline. The VPS log message flood continues unchanged, meaning the loop traffic is generated within OMR.

xzjq, 2023-12-30:
It appears that by default the OMR dnsmasq will bind to the WAN IPs. That is undesirable for a public WAN IP. I changed it to only listen on LAN (I didn't check if there was a firewall rule protecting port 53 on the public WAN IP).

I added a ss-rust redir rule that excluded the public WAN IP as a destination and the traffic flood on the VPS stopped. I will need to ascertain how to make that rule dynamic. The WAN IP is public, but not static.

Ysurac, 2023-12-30:
There is nothing on OMR by default that can connect to port 5900.
Do you have installed or configured something other than via the Wizard ?
DNSMasq bind on all IPs but firewall rules block traffic on WAN.

xzjq, 2023-12-30:
Okay, I created a brand new 6.1-based OMR VM using the snapshot image that was available on 20 Dec 2023.

*No* additional software is installed.
This OMR was solely configured using the wizard, except to disable DHCP on the LAN and WANs in luci. 
There is *no* client traffic being routed via this OMR. 
The OMR LAN interface was taken DOWN by ifconfig to ensure that no local traffic could be inadvertently routed.
The OMR is configured to use ss-rust 2022.

It still floods connections via the master interface, and *only* to the WAN that has a public IP (i.e. where the ip address and WAN address are the same/not NAT). When I first configured the system, I gave it only the other two WANs. No flood. When I configured the third WAN, via the wizard, it immediately started the flood.

<hr>
Per VPS journalctl, here are the connection counts / ports on that one public WAN IP that have been initiated by this brand new, pristine OMR in the past ~20 minutes has existed/been configured.

`conn count / Public WAN IP:port`
`   7658  98.97.x.x:5900`
`   1558  98.97.x.x:38520`
`    924  98.97.x.x:17010`
`    826  98.97.x.x:47000`
`    563  98.97.x.x:9085`
`    485  98.97.x.x:2057`
`    460  98.97.x.x:8277`
`    336  98.97.x.x:21`
`    322  98.97.x.x:55552`
`    318  98.97.x.x:6000`
`    289  98.97.x.x:43080`
`    222  98.97.x.x:12228`
`    190  98.97.x.x:5800`
`    186  98.97.x.x:12177`
`    157  98.97.x.x:19225`
`    149  98.97.x.x:3396`
`    146  98.97.x.x:179`
`    128  98.97.x.x:5902`
`    124  98.97.x.x:5901`
`    120  98.97.x.x:2011`
`     96  98.97.x.x:31134`
`     55  98.97.x.x:6625`
`     36  98.97.x.x:427`
`     28  98.97.x.x:3306`
`     26  98.97.x.x:9300`
`     25  98.97.x.x:4782`
`     18  98.97.x.x:22`
`     16  98.97.x.x:2525`
`      4  98.97.x.x:3389`
`      4  98.97.x.x:23`
`      2  98.97.x.x:5577`

xzjq, 2023-12-30:
Changing `ss_redir.hi1` to listen solely on the LAN interface rather than `::` stops the flood and eliminates the thousands of extraneous connections cited in this ticket. 

E.g. Established Connections page shows 13 connections now, after this OMR/VPS has been in use & routing packets for my home network, and the `journalctl` on the VPS no longer shows target address entries for the public IP of the one of my three WANs that has that.

Based on this and the prior experiments, I surmise that either:

1. OMR is generating traffic internally, or 
2. For some reason the publicly accessible IP of the WAN was not firewalled by OMR, was listening, and the sockets listed in my prior posts were somehow generated by port scans from the internet.

Ysurac, 2023-12-30:
1. OMR is not generating traffic on port 5900 by default
2. You can try this easily ith something like: ```curl wanip:8554``` and check if you find this in log
Check that WAN port are in WAN firewall zone.

xzjq, 2023-12-30:
Okay, I confirmed this is caused by port scans, using your `curl` test and a few arbitrary ports which then show up in the VPS logs. The WAN with the publicly routable IP is in the WAN firewall zone.

Is there any reason for `ss_redir.hi1` to listen on `::` rather than just the LAN interface IP? Also, adding an OMR `iptables` rule on INPUT to drop all traffic with the public WAN IP destination stops the flood too, but that breaks things.

Ysurac, 2023-12-30:
Some rules are not applied by firewall. I'm checking why.

Ysurac, 2023-12-30:
Seems to be applied in fact...
What is the result of ```nft list ruleset``` via SSH on the router ?

Ysurac, 2023-12-30:
Your test is done from your LAN or from an external server ? Should be done from an external server (else it's normal).

xzjq, 2023-12-30:
The `curl` test was done from a different VPS from a different provider, so accurately represents an external/foreign host.

xzjq, 2023-12-30:
Here is the `nft list ruleset` output (public IP redacted):
<hr>
table inet fw4 {
	ct helper amanda {
		type "amanda" protocol udp
		l3proto inet
	}

	ct helper ftp {
		type "ftp" protocol tcp
		l3proto inet
	}

	ct helper RAS {
		type "RAS" protocol udp
		l3proto inet
	}

	ct helper Q.931 {
		type "Q.931" protocol tcp
		l3proto inet
	}

	ct helper irc {
		type "irc" protocol tcp
		l3proto ip
	}

	ct helper pptp {
		type "pptp" protocol tcp
		l3proto ip
	}

	ct helper sip {
		type "sip" protocol udp
		l3proto inet
	}

	ct helper snmp {
		type "snmp" protocol udp
		l3proto ip
	}

	ct helper tftp {
		type "tftp" protocol udp
		l3proto inet
	}

	chain input {
		type filter hook input priority filter; policy drop;
		iifname "lo" accept comment "!fw4: Accept traffic from loopback"
		ct state established,related accept comment "!fw4: Allow inbound established and related flows"
		tcp flags syn / fin,syn,rst,ack jump syn_flood comment "!fw4: Rate limit TCP syn packets"
		iifname "eth0.1024" jump input_lan comment "!fw4: Handle lan IPv4/IPv6 input traffic"
		iifname { "eth0.300", "eth0.400", "eth0.500" } jump input_wan comment "!fw4: Handle wan IPv4/IPv6 input traffic"
		iifname "tun0" jump input_vpn comment "!fw4: Handle vpn IPv4/IPv6 input traffic"
		jump handle_reject
	}

	chain forward {
		type filter hook forward priority filter; policy drop;
		ct state established,related accept comment "!fw4: Allow forwarded established and related flows"
		meta l4proto tcp ip daddr 98.97.x.x counter packets 0 bytes 0 jump reject_to_vpn comment "!fw4: Block flood"
		meta l4proto udp ip daddr 98.97.x.x counter packets 0 bytes 0 jump reject_to_vpn comment "!fw4: Block flood"
		icmp type echo-request limit rate 1000/second counter packets 60 bytes 5456 accept comment "!fw4: Allow-All-Ping"
		icmpv6 type echo-request limit rate 1000/second counter packets 0 bytes 0 accept comment "!fw4: Allow-All-Ping"
		udp dport 443 counter packets 12 bytes 16536 drop comment "!fw4: Block QUIC All"
		iifname "eth0.1024" jump forward_lan comment "!fw4: Handle lan IPv4/IPv6 forward traffic"
		iifname { "eth0.300", "eth0.400", "eth0.500" } jump forward_wan comment "!fw4: Handle wan IPv4/IPv6 forward traffic"
		iifname "tun0" jump forward_vpn comment "!fw4: Handle vpn IPv4/IPv6 forward traffic"
[... 582 lines cut ...]
		xt match "set" counter packets 0 bytes 0 xt target "MARK"
		meta mark 0x00005398 counter packets 0 bytes 0 return
		xt match "set" counter packets 0 bytes 0 xt target "MARK"
		meta mark 0x00005399 counter packets 0 bytes 0 return
		xt match "set" counter packets 0 bytes 0 xt target "MARK"
		meta mark 0x00005397 counter packets 0 bytes 0 return
		xt match "set" counter packets 0 bytes 0 xt target "MARK"
		meta mark 0x00005396 counter packets 0 bytes 0 return
		xt match "set" counter packets 0 bytes 0 xt target "MARK"
		meta mark 0x00005395 counter packets 0 bytes 0 return
		xt match "set" counter packets 13 bytes 832 return
		xt match "set" counter packets 0 bytes 0 xt target "MARK"
		xt match "set" counter packets 0 bytes 0 return
		xt match "set" counter packets 0 bytes 0 return
		meta mark 0x00000539 counter packets 0 bytes 0 return
		meta l4proto tcp counter packets 3192 bytes 185577 jump ssr_def_src
	}

	chain ssr_def_src {
		xt match "set" counter packets 0 bytes 0 return
		xt match "set" counter packets 0 bytes 0 jump ssr_def_forward
		xt match "set" counter packets 0 bytes 0 jump ssr_def_dst
		xt match "comment" counter packets 3192 bytes 185577 jump ssr_def_forward
	}
}

xzjq, 2023-12-30:
For now I have set ss-rust to only listen on the LAN address. Naturally, I changed it back to the default `::` prior to the `curl` test from the external/uninvolved VPS, and that immediately resumed the flood in the logs and the established connections.

Ysurac, 2023-12-31:
I'm not able to reproduce the issue for now.
You have 2 starlink, one with a router and one without ? And it's this last one that make problem ?

xzjq, 2023-12-31:
The issue is more precisely described as "WAN with a publicly routable IP". Both my starlinks have equivalent hardware/VLAN configuration.

Presently, almost all ISPs give IPs that are behind layers of CG-NAT. Most consumer grade WAN devices also perform NAT themselves; two of my WANs are this way (c.f. the private subnet addresses for the WANs "TMo" and "Starlink_1"). In these cases the status page shows a different IP address and WAN address.

One of my Starlinks (the affected one) is configured by Starlink, on their network side, to give a public IP and be a transparent router (this is a starlink premium feature). In this case, WAN address is the IP address. No NAT. This public IP gets port scanned and OMR is listening on that public IP by default with `ss_redir.hi1` configured to bind to `local_address: ::` rather than solely the LAN interface IP of `172.31.31.8`.

<hr>
Here's how I confirmed this is port scan related:

1. Logged into an unrelated VPS, external to my home network, w/ IP `23.94.x.x`
2. From that VPS, `curl 98.97.x.x:44444` (etc)... i.e. the public IP of the OMR WAN (btw, that WAN IP responds on ping too)
3.  Following `journalctl` on the VPS used by OMR (IP `38.72.x.x`), see lines generated with matching entries for those arbitrary port numbers **_but_** that show the same `clientAddress` and `targetAddress` as the WAN: e.g. `Dec 31 12:53:29 vps shadowsocks-go[526]: 2023-12-31T12:53:29.205-0800 INFO service/tcp.go:263 Two-way relay completed {"server": "ss-2022", "listener": 0, "listenAddress": "[::]:65280", "clientAddress": "[::ffff:98.97.x.x]:52973", "username": "openmptcprouter", "targetAddress": "98.97.x.x:44444", "client": "direct", "nl2r": 82, "nr2l": 0}`

[image]

xzjq, 2023-12-31:
Ah, the `clientAddress` is actually prefixed by `[::ffff:`, so is this a 6-to-4 type thing? Note that my `curl` tests were using the manually entered IPv4 address of the public WAN.

xzjq, 2023-12-31:
FWIW, I tried the `curl` test locally from another device that was a peer on the VLAN for Starlink_1 (i.e. behind all the NAT and that WAN is using a private address). So, `curl` was targeting the private IP for the WAN on the OMR from a peer on the same `192.168.1.0/24` subnet.

The connection was refused, so the firewall works to protect that, but not in the case of a publicly routable IP.

xzjq, 2024-01-01:
Further confirmation: when the `ss_redir.hi1` is reconfigured to bind only to `local address: 172.31.31.8` (i.e. the LAN interface IP) rather than `::`, then the port scan test with `curl` using that unrelated, external VPS => the publicly routable IPv4 of the Starlink_2 (described in prior post) results in an immediate `Connection refused`.

Ysurac, 2024-01-01:
It's strange, wan traffic should be blocked the same way with a public IP or a private IP as it's based on interface.
If you do a curl on a private IP ```192.168.1.22``` for example from a client in same subnet, it's blocked ?
I will compile a new image with iptables-nft used for all firewall traffic, maybe a conflict between iptables-legacy and iptables-nft.

xzjq, 2024-01-01:
>If you do a curl on a private IP `192.168.1.22` for example from a client in same subnet, it's blocked ? 

Correct. The address is pingable, but the `curl` test results in instant `Connection refused`.

xzjq, 2024-01-02:
Just confirmed this "connection leak" (i.e. thousands of established connections) caused by port scanning on a public WAN IP is exhibited by V2Ray/XRay as well — not just shadowsocks.

closed 2024-04-07 by github-actions[bot]