Ysurac/openmptcprouter #3093: VPN offen tun off in the last snapshot

fareign, 2023-12-30:
## Expected Behavior
prox:ss+glorytuntcp
PRox2:xray tojan+openvpn

internet:one fiber + 5G
I'm sure the internet is good.

`Dec 30 09:40:18 OpenMPTCProuter daemon.notice netifd: Network device 'tun0' link is up
Dec 30 09:40:18 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' has link connectivity 
Dec 30 09:40:18 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is setting up now
Dec 30 09:40:18 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is now up
Dec 30 09:40:19 OpenMPTCProuter user.notice firewall: Reloading firewall due to ifup of omrvpn (tun0)
Dec 30 09:40:19 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Dec 30 09:40:19 OpenMPTCProuter user.notice mptcp: Reloading mptcp config due to ifup of omrvpn (tun0)
Dec 30 09:40:19 OpenMPTCProuter user.notice MPTCP: Set tun0 to off
Dec 30 09:40:20 OpenMPTCProuter user.notice MPTCP: Set eth0 to off
Dec 30 09:40:20 OpenMPTCProuter user.notice MPTCP: Set usb0 to on
Dec 30 09:40:20 OpenMPTCProuter user.notice MPTCP: Set usb2 to master
Dec 30 09:40:20 OpenMPTCProuter user.notice post-tracking-010-services: Set firewall on server vps
Dec 30 09:40:20 OpenMPTCProuter user.notice MPTCP: Set eth1 to on
Dec 30 09:40:20 OpenMPTCProuter user.notice MPTCP: Set usb1 to on
Dec 30 09:40:20 OpenMPTCProuter user.notice MPTCP: Set tun0 to off
Dec 30 09:40:23 OpenMPTCProuter daemon.warn openvpn(omr)[21036]: sitnl_send: rtnl: generic error (-17): File exists
Dec 30 09:40:23 OpenMPTCProuter daemon.notice openvpn(omr)[21036]: Initialization Sequence Completed
Dec 30 09:40:25 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Tunnel up : Replace default route by 10.255.252.5 dev tun0`

`来自 223.5.5.5 的回复: 字节=32 时间=42ms TTL=119
来自 223.5.5.5 的回复: 字节=32 时间=41ms TTL=119
来自 223.5.5.5 的回复: 字节=32 时间=41ms TTL=119
请求超时。
请求超时。
请求超时。
来自 223.5.5.5 的回复: 字节=32 时间=978ms TTL=119
来自 223.5.5.5 的回复: 字节=32 时间=47ms TTL=119
请求超时。
请求超时。
请求超时。
请求超时。
请求超时。
请求超时。
请求超时。
请求超时。
请求超时。
请求超时。
请求超时。
请求超时。
来自 223.5.5.5 的回复: 字节=32 时间=22ms TTL=114
来自 223.5.5.5 的回复: 字节=32 时间=21ms TTL=114
来自 223.5.5.5 的回复: 字节=32 时间=21ms TTL=114
来自 223.5.5.5 的回复: 字节=32 时间=21ms TTL=114
来自 223.5.5.5 的回复: 字节=32 时间=23ms TTL=114
来自 223.5.5.5 的回复: 字节=32 时间=23ms TTL=114
来自 223.5.5.5 的回复: 字节=32 时间=22ms TTL=114
来自 223.5.5.5 的回复: 字节=32 时间=21ms TTL=114
请求超时。
来自 223.5.5.5 的回复: 字节=32 时间=45ms TTL=119`

## Context (Environment)

## Specifications

  - OpenMPTCProuter version: 0.61 snap 12.28
  - OpenMPTCProuter VPS version: 0.1029
  - OpenMPTCProuter VPS provider:aliyun
  - OpenMPTCProuter platform: /x86_64
  - Country: cn

fareign, 2023-12-30:
Dec 30 09:37:17 OpenMPTCProuter daemon.notice netifd: Interface 'wan3' is now up
Dec 30 09:37:18 OpenMPTCProuter user.notice firewall: Reloading firewall due to ifup of wan3 (usb2)
Dec 30 09:37:18 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Dec 30 09:37:19 OpenMPTCProuter user.notice post-tracking-010-services: Set firewall on server vps
Dec 30 09:37:20 OpenMPTCProuter daemon.info vnstatd[20909]: Info: Traffic rate for "usb2" higher than set maximum 1000 Mbit (20s->2673868800, r4294943271 t4294881027, 64bit:0), syncing.
Dec 30 09:37:24 OpenMPTCProuter user.notice MPTCP: Set eth1 to on
Dec 30 09:37:24 OpenMPTCProuter user.notice MPTCP: Set usb1 to on
Dec 30 09:37:24 OpenMPTCProuter user.notice MPTCP: Set tun0 to off
Dec 30 09:37:25 OpenMPTCProuter user.notice mptcp: Reloading mptcp config due to ifup of wan3 (usb2)
Dec 30 09:37:25 OpenMPTCProuter user.notice MPTCP: Set usb2 to master
Dec 30 09:37:25 OpenMPTCProuter user.notice MPTCP: Set eth0 to off
Dec 30 09:37:25 OpenMPTCProuter user.notice MPTCP: Set usb0 to on
Dec 30 09:37:25 OpenMPTCProuter user.notice MPTCP: Set usb2 to master
Dec 30 09:37:26 OpenMPTCProuter user.notice MPTCP: Set eth1 to on
Dec 30 09:37:26 OpenMPTCProuter user.notice MPTCP: Set usb1 to on
Dec 30 09:37:26 OpenMPTCProuter user.notice MPTCP: Set tun0 to off
Dec 30 09:38:50 OpenMPTCProuter daemon.notice netifd: Network device 'usb2' link is down
Dec 30 09:38:50 OpenMPTCProuter daemon.notice netifd: Interface 'wan3' has link connectivity loss
Dec 30 09:38:50 OpenMPTCProuter daemon.notice ttyd[30330]: [2023/12/30 17:38:50:2236] N: rops_handle_POLLIN_netlink: DELADDR
Dec 30 09:38:50 OpenMPTCProuter daemon.notice ttyd[30330]: [2023/12/30 17:38:50:2242] N: rops_handle_POLLIN_netlink: DELADDR
Dec 30 09:38:50 OpenMPTCProuter daemon.notice netifd: Interface 'wan3' is disabled
Dec 30 09:38:50 OpenMPTCProuter daemon.notice netifd: wan3 (10568): udhcpc: SIOCGIFINDEX: No such device
Dec 30 09:38:50 OpenMPTCProuter daemon.notice netifd: wan3 (10568): udhcpc: received SIGTERM
Dec 30 09:38:50 OpenMPTCProuter daemon.notice netifd: wan3 (10568): udhcpc: unicasting a release of 192.168.225.20 to 192.168.225.1
Dec 30 09:38:50 OpenMPTCProuter daemon.notice netifd: wan3 (10568): udhcpc: sending release
Dec 30 09:38:50 OpenMPTCProuter daemon.notice netifd: wan3 (10568): udhcpc: can't bind to interface usb2: No such device
Dec 30 09:38:50 OpenMPTCProuter daemon.notice netifd: wan3 (10568): udhcpc: bindtodevice: No such device
Dec 30 09:38:50 OpenMPTCProuter daemon.notice netifd: wan3 (10568): udhcpc: entering released state
Dec 30 09:38:50 OpenMPTCProuter daemon.notice netifd: wan3 (10568): Command failed: ubus call network.interface notify_proto { "action": 0, "link-up": false, "keep": false, "interface": "wan3" } (Permission denied)
Dec 30 09:38:50 OpenMPTCProuter daemon.notice netifd: Interface 'wan3' is now down
Dec 30 09:38:50 OpenMPTCProuter user.notice mptcp: Set multipath off on  (wan3) due to ifdown
Dec 30 09:38:50 OpenMPTCProuter daemon.info vnstatd[20909]: Info: Interface "usb2" disabled.
Dec 30 09:38:51 OpenMPTCProuter daemon.notice netifd: Interface 'wan3' is enabled
Dec 30 09:38:51 OpenMPTCProuter daemon.notice netifd: Network device 'usb2' link is up
Dec 30 09:38:51 OpenMPTCProuter daemon.notice netifd: Interface 'wan3' has link connectivity 
Dec 30 09:38:51 OpenMPTCProuter daemon.notice netifd: Interface 'wan3' is setting up now
Dec 30 09:38:51 OpenMPTCProuter daemon.notice netifd: wan3 (19266): udhcpc: started, v1.36.1
Dec 30 09:38:51 OpenMPTCProuter daemon.notice netifd: wan3 (19266): udhcpc: broadcasting discover
Dec 30 09:38:51 OpenMPTCProuter user.notice MPTCP: Set eth0 to off
Dec 30 09:38:51 OpenMPTCProuter user.notice MPTCP: Set usb0 to on
Dec 30 09:38:51 OpenMPTCProuter user.notice MPTCP: Set usb2 to master
Dec 30 09:38:54 OpenMPTCProuter daemon.notice netifd: wan3 (19266): udhcpc: broadcasting discover
Dec 30 09:38:55 OpenMPTCProuter daemon.info vnstatd[20909]: Info: Interface "usb2" enabled.
Dec 30 09:38:57 OpenMPTCProuter daemon.notice netifd: wan3 (19266): udhcpc: broadcasting select for 192.168.225.20, server 192.168.225.1
Dec 30 09:38:57 OpenMPTCProuter daemon.notice netifd: wan3 (19266): udhcpc: lease of 192.168.225.20 obtained from 192.168.225.1, lease time 43200
Dec 30 09:38:57 OpenMPTCProuter daemon.notice netifd: Interface 'wan3' is now up
Dec 30 09:38:58 OpenMPTCProuter user.notice firewall: Reloading firewall due to ifup of wan3 (usb2)
Dec 30 09:38:58 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Dec 30 09:38:59 OpenMPTCProuter user.notice post-tracking-010-services: Set firewall on server vps
Dec 30 09:39:01 OpenMPTCProuter user.notice MPTCP: Set eth1 to on
Dec 30 09:39:01 OpenMPTCProuter user.notice MPTCP: Set usb1 to on
Dec 30 09:39:01 OpenMPTCProuter user.notice MPTCP: Set tun0 to off
Dec 30 09:39:16 OpenMPTCProuter user.notice mptcp: Reloading mptcp config due to ifup of wan3 (usb2)
Dec 30 09:39:17 OpenMPTCProuter user.notice MPTCP: Set usb2 to master
Dec 30 09:39:17 OpenMPTCProuter user.notice MPTCP: Set eth0 to off
Dec 30 09:39:17 OpenMPTCProuter user.notice MPTCP: Set usb0 to on
Dec 30 09:39:17 OpenMPTCProuter user.notice MPTCP: Set usb2 to master
Dec 30 09:39:17 OpenMPTCProuter user.notice MPTCP: Set eth1 to on
Dec 30 09:39:18 OpenMPTCProuter user.notice MPTCP: Set usb1 to on
Dec 30 09:39:18 OpenMPTCProuter user.notice MPTCP: Set tun0 to off
[... cut ...]
Dec 30 09:41:29 OpenMPTCProuter user.notice mptcp: Reloading mptcp config due to ifup of wan3 (usb2)
Dec 30 09:41:29 OpenMPTCProuter user.notice MPTCP: Set usb2 to master
Dec 30 09:41:30 OpenMPTCProuter user.notice MPTCP: Set eth0 to off
Dec 30 09:41:30 OpenMPTCProuter user.notice MPTCP: Set usb0 to on
Dec 30 09:41:30 OpenMPTCProuter user.notice MPTCP: Set usb2 to master
Dec 30 09:41:30 OpenMPTCProuter user.notice MPTCP: Set eth1 to on
Dec 30 09:41:30 OpenMPTCProuter user.notice MPTCP: Set usb1 to on
Dec 30 09:41:31 OpenMPTCProuter user.notice MPTCP: Set tun0 to off

Ysurac, 2023-12-30:
Can I have the result of ```ip r``` via SSH on the router ?
Do you know why usb2 seems to be up/down/up/... ?

fareign, 2023-12-30:
USB2 no card.

fareign, 2023-12-30:
Dec 30 17:12:11 OpenMPTCProuter daemon.err openvpn(omr)[22702]: read TCPv4_CLIENT []: Socket not connected (fd=5,code=107)
Dec 30 17:12:11 OpenMPTCProuter daemon.err openvpn(omr)[22702]: read TCPv4_CLIENT []: Socket not connected (fd=5,code=107)
Dec 30 17:12:11 OpenMPTCProuter daemon.err openvpn(omr)[22702]: read TCPv4_CLIENT []: Socket not connected (fd=5,code=107)
Dec 30 17:12:11 OpenMPTCProuter daemon.err openvpn(omr)[22702]: read TCPv4_CLIENT []: Socket not connected (fd=5,code=107)
Dec 30 17:12:11 OpenMPTCProuter daemon.err openvpn(omr)[22702]: read TCPv4_CLIENT []: Socket not connected (fd=5,code=107)
Dec 30 17:12:11 OpenMPTCProuter user.notice post-tracking-001-post-tracking: OpenVPN down, restart it
Dec 30 17:12:11 OpenMPTCProuter daemon.err openvpn(omr)[22702]: read TCPv4_CLIENT []: Socket not connected (fd=5,code=107)
Dec 30 17:12:11 OpenMPTCProuter daemon.err openvpn(omr)[22702]: read TCPv4_CLIENT []: Socket not connected (fd=5,code=107)
Dec 30 17:12:11 OpenMPTCProuter daemon.err openvpn(omr)[22702]: read TCPv4_CLIENT []: Socket not connected (fd=5,code=107)
Dec 30 17:12:11 OpenMPTCProuter daemon.err openvpn(omr)[22702]: read TCPv4_CLIENT []: Socket not connected (fd=5,code=107)
Dec 30 17:12:11 OpenMPTCProuter daemon.err openvpn(omr)[22702]: read TCPv4_CLIENT []: Socket not connected (fd=5,code=107)
Dec 30 17:12:11 OpenMPTCProuter daemon.err openvpn(omr)[22702]: read TCPv4_CLIENT []: Socket not connected (fd=5,code=107)
Dec 30 17:12:11 OpenMPTCProuter daemon.err openvpn(omr)[22702]: read TCPv4_CLIENT []: Socket not connected (fd=5,code=107)
Dec 30 17:12:11 OpenMPTCProuter daemon.err openvpn(omr)[22702]: read TCPv4_CLIENT []: Socket not connected (fd=5,code=107)
Dec 30 17:12:11 OpenMPTCProuter daemon.err openvpn(omr)[22702]: read TCPv4_CLIENT []: Socket not connected (fd=5,code=107)
Dec 30 17:12:11 OpenMPTCProuter daemon.err openvpn(omr)[22702]: read TCPv4_CLIENT []: Socket not connected (fd=5,code=107)
Dec 30 17:12:11 OpenMPTCProuter daemon.err openvpn(omr)[22702]: read TCPv4_CLIENT []: Socket not connected (fd=5,code=107)
Dec 30 17:12:11 OpenMPTCProuter daemon.err openvpn(omr)[22702]: read TCPv4_CLIENT []: Socket not connected (fd=5,code=107)
Dec 30 17:12:11 OpenMPTCProuter daemon.err openvpn(omr)[22702]: read TCPv4_CLIENT []: Socket not connected (fd=5,code=107)
Dec 30 17:12:11 OpenMPTCProuter daemon.err openvpn(omr)[22702]: read TCPv4_CLIENT []: Socket not connected (fd=5,code=107)
Dec 30 17:12:11 OpenMPTCProuter daemon.err openvpn(omr)[22702]: read TCPv4_CLIENT []: Socket not connected (fd=5,code=107)
Dec 30 17:12:11 OpenMPTCProuter daemon.err openvpn(omr)[22702]: read TCPv4_CLIENT []: Socket not connected (fd=5,code=107)
Dec 30 17:12:11 OpenMPTCProuter daemon.err openvpn(omr)[22702]: read TCPv4_CLIENT []: Socket not connected (fd=5,code=107)
Dec 30 17:12:11 OpenMPTCProuter daemon.err openvpn(omr)[22702]: read TCPv4_CLIENT []: Socket not connected (fd=5,code=107)
Dec 30 17:12:11 OpenMPTCProuter daemon.err openvpn(omr)[22702]: read TCPv4_CLIENT []: Socket not connected (fd=5,code=107)
Dec 30 17:12:11 OpenMPTCProuter daemon.err openvpn(omr)[22702]: read TCPv4_CLIENT []: Socket not connected (fd=5,code=107)
Dec 30 17:12:11 OpenMPTCProuter daemon.err openvpn(omr)[22702]: read TCPv4_CLIENT []: Socket not connected (fd=5,code=107)
Dec 30 17:12:11 OpenMPTCProuter daemon.err openvpn(omr)[22702]: read TCPv4_CLIENT []: Socket not connected (fd=5,code=107)
Dec 30 17:12:11 OpenMPTCProuter daemon.err openvpn(omr)[22702]: read TCPv4_CLIENT []: Socket not connected (fd=5,code=107)
Dec 30 17:12:11 OpenMPTCProuter daemon.err openvpn(omr)[22702]: read TCPv4_CLIENT []: Socket not connected (fd=5,code=107)
Dec 30 17:12:11 OpenMPTCProuter daemon.err openvpn(omr)[22702]: read TCPv4_CLIENT []: Socket not connected (fd=5,code=107)
Dec 30 17:12:11 OpenMPTCProuter daemon.err openvpn(omr)[22702]: read TCPv4_CLIENT []: Socket not connected (fd=5,code=107)
Dec 30 17:12:11 OpenMPTCProuter daemon.err openvpn(omr)[22702]: read TCPv4_CLIENT []: Socket not connected (fd=5,code=107)
Dec 30 17:12:11 OpenMPTCProuter daemon.err openvpn(omr)[22702]: read TCPv4_CLIENT []: Socket not connected (fd=5,code=107)
Dec 30 17:12:11 OpenMPTCProuter daemon.err openvpn(omr)[22702]: read TCPv4_CLIENT []: Socket not connected (fd=5,code=107)
Dec 30 17:12:11 OpenMPTCProuter daemon.err openvpn(omr)[22702]: read TCPv4_CLIENT []: Socket not connected (fd=5,code=107)
Dec 30 17:12:11 OpenMPTCProuter daemon.err openvpn(omr)[22702]: read TCPv4_CLIENT []: Socket not connected (fd=5,code=107)
Dec 30 17:12:11 OpenMPTCProuter daemon.err openvpn(omr)[22702]: read TCPv4_CLIENT []: Socket not connected (fd=5,code=107)
Dec 30 17:12:11 OpenMPTCProuter daemon.err openvpn(omr)[22702]: read TCPv4_CLIENT []: Socket not connected (fd=5,code=107)
Dec 30 17:12:11 OpenMPTCProuter daemon.err openvpn(omr)[22702]: read TCPv4_CLIENT []: Socket not connected (fd=5,code=107)
Dec 30 17:12:11 OpenMPTCProuter daemon.err openvpn(omr)[22702]: read TCPv4_CLIENT []: Socket not connected (fd=5,code=107)
Dec 30 17:12:11 OpenMPTCProuter daemon.err openvpn(omr)[22702]: read TCPv4_CLIENT []: Socket not connected (fd=5,code=107)
Dec 30 17:12:11 OpenMPTCProuter daemon.err openvpn(omr)[22702]: read TCPv4_CLIENT []: Socket not connected (fd=5,code=107)
Dec 30 17:12:11 OpenMPTCProuter daemon.err openvpn(omr)[22702]: read TCPv4_CLIENT []: Socket not connected (fd=5,code=107)
[... cut ...]
Dec 30 17:12:13 OpenMPTCProuter user.notice MPTCP: Set eth1 to on
Dec 30 17:12:14 OpenMPTCProuter user.notice MPTCP: Set usb1 to master
Dec 30 17:12:14 OpenMPTCProuter user.notice MPTCP: Set tun0 to off
Dec 30 17:12:16 OpenMPTCProuter daemon.warn openvpn(omr)[3972]: sitnl_send: rtnl: generic error (-17): File exists
Dec 30 17:12:16 OpenMPTCProuter daemon.notice openvpn(omr)[3972]: Initialization Sequence Completed
Dec 30 17:12:18 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Tunnel up : Replace default route by 10.255.252.9 dev tun0

fareign, 2023-12-30:
I use 5G only and close OMR, it works very well ,can download 15MB,UPload 5MB with low latancy.
if open OMR, afew minutes,will brokendown. even set vpn to none is the same

Dec 30 17:27:40 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can't ping server vps on 112.74.39.***, no server API answer on 112.74.39.***)
Dec 30 17:27:42 OpenMPTCProuter daemon.info omr-tracker-xray: xray is down (can't contact via http 198.41.212.162, 212.27.48.10, 198.27.92.1)
Dec 30 17:27:56 OpenMPTCProuter user.notice xray: Rules DOWN

fareign, 2023-12-30:
maybe xray-trojan no-stable? change proxy to xray-ss seems much better

fareign, 2023-12-30:
xray-ss cant work download well ,but cant  aggregation UDP，xray vmess can aggreate UDP without VPN,but if download speed is up,will break down.

closed 2024-04-04 by github-actions[bot]