Ysurac/openmptcprouter #3094: Add support for Marvell CN9130 : QNAP QHora-322/IEi PUZZLE-M902

xillibit, 2023-12-30:
**Is this request related to a missing driver support for a particular hardware platform, SoC or board? Please describe.**
Request for support Marvell CN9130 which is present in [QNAP QHora-322](https://www.qnap.com/fr-fr/product/qhora-322/specs/hardware) and [IEi PUZZLE-M902](https://www.ieiworld.com/fr/product/model.php?II=748), the both seems to share teh same hardware.

**Describe why you are asking for this support?**
This router is interesting because it has six 2.5GbE RJ-45 ports and three 10GbE RJ-45. 
OpenWrt is preinstalled on it : https://openwrt.org/toh/iei/puzzle-m902

xillibit, 2024-04-05:
The support of this hardware has been really done or the issue is just closed automatically ?

Ysurac, 2024-04-05:
It's closed automatically.
I forgot this issue sorry.
As it's supported by OpenWRT, you can try to compile it yourself: https://github.com/Ysurac/openmptcprouter/wiki/Create-image-for-unsupported-platform

I support officially only platform I own and these are not really cheap.