Ysurac/openmptcprouter #3101: How to solve the problem of recompiling from beginning by running 'make'

atyouspace, 2024-01-03:
## Expected Behavior

## Current Behavior
Every time I run make, recompile rather than compiling the modified files. I did not modify the Makefile. How to solve this problem?

## Specifications

  - OpenMPTCProuter version: 
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform:

Ysurac, 2024-01-03:
You can do a ```make -j$(nproc)```, it's faster than a -j1. It's not recompiling from beginning if run in source directory.
You can also compile a specific package with ```make package/network/utils/nftables/compile``` for example and install it with ```make package/install && make target/install```

atyouspace, 2024-01-03:
> You can do a `make -j$(nproc)`, it's faster than a -j1. It's not recompiling from beginning if run in source directory. You can also compile a specific package with `make package/network/utils/nftables/compile` for example and install it with `make package/install && make target/install`

Does it have anything to do with whether the platform is not supported?
I found `rm -f ...` will run firstly after `make`

Ysurac, 2024-01-03:
This doesn't have to do anything with supported or not supported platform.
If you do only a make some part are recompiled and some not. It use OpenWRT build system.

closed 2024-04-07 by github-actions[bot]