Ysurac/openmptcprouter #3102: LTE Modem configuration

tsh2001, 2024-01-03:
## Expected Behavior
I have 2 4G LTE Dongle to connect to the router. i connected one of the Dongle to the RPI4 USB port and another to a USB Hub.

## Current Behavior
I cant seem to get it to work. its stating Duplicate Network interface
[image]

## Specifications

  - OpenMPTCProuter version: 0.59.1
  - OpenMPTCProuter VPS version: 0.1028
  - OpenMPTCProuter VPS provider: DigitalOcean
  - OpenMPTCProuter platform: RPI4

Ysurac, 2024-01-03:
It's working with only "modem2" connected ?

tsh2001, 2024-01-04:
Hi, 
With only 'modem2' connected, it still state Network Interface Duplicated.

Additionally, I tried using another Dongle, connected it to the RPI along with one of the 4G LTE Dongle. The router recognise the new dongle and set it as wwan0. However, it doesn't seem to be connected to the internet. I have set it as DHCP 'wwan0' but its stating no IP address or Gateway found.

[image]

Ysurac, 2024-01-05:
What is the result of ```uci show network``` and ```ip a``` ?
What do you have in Network->Interfaces for wwan0 interface ?

closed 2024-04-09 by github-actions[bot]