Ysurac/openmptcprouter #3103: impossible to config 4g backup

gmorlet, 2024-01-03:
## Expected Behavior

## Current Behavior

Multipath TCP : backup
OK to red cross in status

log 
Wed Jan  3 09:11:39 2024 user.notice omr-tracker: Launched
Wed Jan  3 09:11:39 2024 user.notice omr-bypass: Starting OMR-ByPass...
Wed Jan  3 09:11:42 2024 user.notice post-tracking-post-tracking: 4G (4G) switched off because check error and ping from 192.168.2.1 error (4.2.2.1,8.8.8.8,80.67.169.12)
Wed Jan  3 09:11:42 2024 user.notice post-tracking-post-tracking: Delete default route to 51.210.149.28 via 192.168.2.254 dev 4G
Wed Jan  3 09:11:47 2024 user.notice omr-bypass: Restart dnsmasq...
Wed Jan  3 09:11:48 2024 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Wed Jan  3 09:11:48 2024 user.notice omr-bypass: OMR-ByPass is running
Wed Jan  3 09:11:48 2024 daemon.info dnsmasq[1]: started, version 2.85 cachesize 150

## Specifications

  - OpenMPTCProuter version: 5.4.86-mptcp 0.1024
  - OpenMPTCProuter VPS version: 5.4.86-mptcp 0.1024
  - OpenMPTCProuter VPS provider: OVH
  - OpenMPTCProuter platform: RPI 4B

Ysurac, 2024-01-03:
When the interface is put again in active Multipath TCP it's working again ?

gmorlet, 2024-01-03:
at the begin : active ==> Yes
modify to backup ==> red cross
modify to active ==> red cross

gmorlet, 2024-01-03:
quelqu'un a t il un mode opératoire pour bien déclarer un dongle 4G et l'ajouter en interface backup ?

closed 2024-04-07 by github-actions[bot]