Ysurac/openmptcprouter #3108: OMR status no increct with CN VPS in proxy only mode

fareign, 2024-01-04:
if I set VPN to none. use aliyu VPS in China .the proxy status show Direct output. but it aggrated.
if I use aliyun in Hongkong. it is correct.
what's the diffent? how to solve it.

</body>
</html>

## Context (Environment)

## Specifications

  - OpenMPTCProuter version: last snap 6.1
  - OpenMPTCProuter VPS version: 0.1029 test
  - OpenMPTCProuter VPS provider: aliyun
  - OpenMPTCProuter platform: x86_64
  - Country: CN,HK

[Direct output](http://192.168.100.1/cgi-bin/luci/admin/system/openmptcprouter/wizard)

Ysurac, 2024-01-05:
When you see "Direct Output", the IP displayed for the VPS in status page is the same as the IP set in the Wizard ?

fareign, 2024-01-06:
> When you see "Direct Output", the IP displayed for the VPS in status page is the same as the IP set in the Wizard ?

Direct OUt IP is the maseter wanIP. but speedtest ,ip check is VPS IP, SO I think only statu display erro. but HK VPS
 seems have no this problem

fareign, 2024-01-09:
@Ysurac 
sometime it cant get public IP like this
Direct output (120.228.****.58)
Can't get public IP address from XRay

if I save and reply OMR wizard it can get public IP still dispaly directout
Direct output (120.228.113.58
版本 0.1029-test 6.1.0-13-amd64

fareign, 2024-01-09:
so it need to restart XRAY

Jan  9 07:10:32 OpenMPTCProuter daemon.info xray[614]: Xray 1.8.6 (Xray, Penetrates Everything.) Custom (go1.21.4 linux/amd64)
Jan  9 07:10:32 OpenMPTCProuter daemon.info xray[614]: A unified platform for anti-censorship.
Jan  9 07:10:32 OpenMPTCProuter daemon.info xray[614]: 2024/01/09 07:10:32 [Info] infra/conf/serial: Reading config: /var/etc/xray/xray.main.json
Jan  9 07:10:32 OpenMPTCProuter daemon.info xray[614]: Failed to start: main: failed to load config files: [/var/etc/xray/xray.main.json] > infra/conf: Failed to build REALITY config. > infra/conf: empty "publicKey"
Jan  9 07:10:32 OpenMPTCProuter daemon.info procd: Instance xray::xray.main s in a crash loop 6 crashes, 0 seconds since last crash
Jan  9 07:11:06 OpenMPTCProuter daemon.err uhttpd[8347]: [info] luci: accepted login on / for root from 192.168.100.166

closed 2024-04-14 by github-actions[bot]