Ysurac/openmptcprouter #3110: OMR unable to establish new connections while the Master WAN is without internet.

vempire-ghost, 2024-01-04:
## Expected Behavior
Capable of continuing to initiate new connections while the Master WAN is without internet if there are other WANs with functional internet.

## Current Behavior
When the Master WAN loses internet, established connections continue to function normally. However, no new connections can be initiated until the Master WAN regains internet.
I tried configuring the OMR tracker as ping and DNS, and even so, it is not detecting the lack of internet on the master WAN, even if it stays without it for minutes. The log shows only this:
Jan  4 22:29:46 OpenMPTCProuter user.notice MPTCP: Set eth0 to off
Jan  4 22:29:46 OpenMPTCProuter user.notice MPTCP: Set eth1 to on
Jan  4 22:29:46 OpenMPTCProuter user.notice MPTCP: Set eth2 to on
Jan  4 22:29:46 OpenMPTCProuter user.notice MPTCP: Set eth3 to on
Jan  4 22:29:46 OpenMPTCProuter user.notice MPTCP: Set eth4 to master
Jan  4 22:30:25 OpenMPTCProuter user.notice OMR-Tracker-Server: Master server vps down (X.X.X.X)
Jan  4 22:30:35 OpenMPTCProuter user.notice OMR-Tracker-Server: Master server vps down (X.X.X.X)
Jan  4 22:30:45 OpenMPTCProuter user.notice OMR-Tracker-Server: Master server vps down (X.X.X.X)
Jan  4 22:30:56 OpenMPTCProuter daemon.info omr-tracker-xray: xray is down (can't contact via http 77.88.55.77, 1.1.1.1, 74.82.42.42)
Jan  4 22:30:58 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later
Jan  4 22:30:58 OpenMPTCProuter user.notice xray: Rules DOWN
Jan  4 22:31:00 OpenMPTCProuter daemon.info omr-tracker-xray: Server (X.X.X.X) seems down, no answer to ping
Jan  4 22:31:05 OpenMPTCProuter user.notice OMR-Tracker-Server: Master server vps down (X.X.X.X)
Jan  4 22:31:07 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can't ping server vps on vps.ddnsfree.com, no server API answer on vps.ddnsfree.com)
Jan  4 22:31:13 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can't ping server vps on vps.ddnsfree.com, no server API answer on vps.ddnsfree.com)
Jan  4 22:31:16 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can't ping server vps on vps.ddnsfree.com, no server API answer on vps.ddnsfree.com)
Jan  4 22:31:21 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can't ping server vps on vps.ddnsfree.com, no server API answer on vps.ddnsfree.com)
Jan  4 22:31:24 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can't ping server vps on vps.ddnsfree.com, no server API answer on vps.ddnsfree.com)
Jan  4 22:31:26 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can't ping server vps on vps.ddnsfree.com, no server API answer on vps.ddnsfree.com)
Jan  4 22:31:30 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can't ping server vps on vps.ddnsfree.com, no server API answer on vps.ddnsfree.com)
Jan  4 22:31:36 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can't ping server vps on vps.ddnsfree.com, no server API answer on vps.ddnsfree.com)
Jan  4 22:31:38 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can't ping server vps on vps.ddnsfree.com, no server API answer on vps.ddnsfree.com)
Jan  4 22:31:41 OpenMPTCProuter user.notice OMR-Tracker-Server: Master server vps down (X.X.X.X)
Jan  4 22:32:00 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can't ping server vps on vps.ddnsfree.com, no server API answer on vps.ddnsfree.com)
Jan  4 22:32:20 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can't ping server vps on vps.ddnsfree.com, no server API answer on vps.ddnsfree.com)
Jan  4 22:32:40 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can't ping server vps on vps.ddnsfree.com, no server API answer on vps.ddnsfree.com)
Jan  4 22:32:56 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can't ping server vps on vps.ddnsfree.com, no server API answer on vps.ddnsfree.com)
Jan  4 22:33:03 OpenMPTCProuter daemon.info omr-tracker-xray: xray is up (can contact via http 198.41.212.162)
Jan  4 22:33:03 OpenMPTCProuter daemon.info omr-tracker-xray: Reload xray rules
Jan  4 22:33:03 OpenMPTCProuter user.notice xray: Rules UP
Jan  4 22:33:03 OpenMPTCProuter user.notice xray: Reload omr-bypass rules
Jan  4 22:33:04 OpenMPTCProuter user.notice omr-bypass: Starting OMR-ByPass...
Jan  4 22:33:04 OpenMPTCProuter user.notice omr-bypass: OMR-ByPass is running

Throughout the period that the OMR log indicates it is unable to connect to the VPS, I cannot establish any new connections. However, the existing connections continue to function normally.

## Possible Solution
I imagine the solution would be for the OMR tracker, when unable to make contact with the VPS through the master WAN, to attempt to use the other WANs instead of preventing new connections.

## Steps to Reproduce the Problem
To reproduce the issue, I deliberately make the master WAN lose internet without disconnecting the network cable until the OMR tracker starts giving errors about being unable to connect to the VPS. Then I try to initiate a new connection, whether opening a webpage or starting an online game, and I receive an error indicating no internet access.

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.60beta1-5.4 r0+16862-170d9e447d snap 01/01
  - OpenMPTCProuter VPS version: 0.1029 test
  - OpenMPTCProuter VPS provider: GCP
  - OpenMPTCProuter platform: x86_64
  - Country: Brazil

This is a significant issue that I don't recall having until recently. I remember having issues with the OMR tracker not accurately determining whether the WAN has internet or not, but I didn't have problems accessing the internet during this outage if other WANs had functional internet.

Ysurac, 2024-01-05:
omr-tracker doesn't detect the connection lost, only omr-server, so it switch to backup server but as connection is not detected as lost it can't reach it correctly.
What is the result of ```uci show omr-tracker``` via SSH on the router ?

vempire-ghost, 2024-01-05:
root@OpenMPTCProuter:~# uci show omr-tracker
omr-tracker.defaults=defaults
omr-tracker.defaults.enabled='1'
omr-tracker.defaults.hosts='4.2.2.1' '8.8.8.8' '80.67.169.12' '8.8.4.4' '9.9.9.9' '1.0.0.1' '114.114.115.115' '1.2.4.8' '80.67.169.40' '114.114.114.114' '1.1.1.1'
omr-tracker.defaults.hosts6='2606:4700:4700::1111' '2606:4700:4700::1001' '2620:fe::fe' '2620:fe::9' '2001:4860:4860::8888' '2001:4860:4860::8844'
omr-tracker.defaults.interval_tries='1'
omr-tracker.defaults.type='ping'
omr-tracker.defaults.wait_test='0'
omr-tracker.defaults.server_http_test='0'
omr-tracker.defaults.restart_down='0'
omr-tracker.defaults.mail_alert='0'
omr-tracker.defaults.timeout='1'
omr-tracker.defaults.tries='1'
omr-tracker.defaults.count='1'
omr-tracker.defaults.interval='1'
omr-tracker.proxy=proxy
omr-tracker.proxy.enabled='1'
omr-tracker.proxy.hosts='212.27.48.10' '198.27.92.1' '151.101.129.164' '77.88.55.77' '1.1.1.1' '74.82.42.42' '198.41.212.162'
omr-tracker.proxy.timeout='10'
omr-tracker.proxy.tries='3'
omr-tracker.proxy.wait_test='0'
omr-tracker.proxy.interval_tries='1'
omr-tracker.proxy.interval='10'
omr-tracker.proxy.mail_alert='0'
omr-tracker.server=server
omr-tracker.server.enabled='1'
omr-tracker.server.tries='3'
omr-tracker.server.timeout='10'
omr-tracker.server.wait_test='0'
omr-tracker.server.interval='5'
omr-tracker.server.mail_alert='0'
omr-tracker.omrvpn=interface
omr-tracker.omrvpn.type='none'
omr-tracker.omrvpn.timeout='10'
omr-tracker.omrvpn.tries='3'
omr-tracker.omrvpn.interval='5'
omr-tracker.omrvpn.mail_alert='0'
omr-tracker.omrvpn.enabled='1'
omr-tracker.omrvpn.server_http_test='1'
omr-tracker.omrvpn.restart_down='0'
omr-tracker.omrvpn.hosts='4.2.2.1' '8.8.8.8'
omr-tracker.omrvpn.count='2'
omr-tracker.omrvpn.wait_test='0'

I tried setting all configuration parameters for the tracker to 1 to see if it would detect the WAN without internet, but it didn't resolve the issue. It just flooded my log with these statuses below, but the master WAN, even if it stays without internet for minutes, it doesn't remove the route for it. Even when the OMR tracker removes the route from the master WAN, the OMR Server tracker continues to try using it to check if the VPS is online, which ends up causing this issue.
And in this OMR, I am using only one VPS configured in it, I am not using a backup VPS.

Jan  5 15:38:45 OpenMPTCProuter user.notice post-tracking-001-post-tracking: wan2 (eth2) switched off because check error and ping from 192.168.0.51 error (114.114.115.115)
Jan  5 15:38:45 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Delete default route to X.X.X.X dev eth2
Jan  5 15:38:46 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Change in routes, set ip route replace default scope global nexthop via 192.168.0.1 dev eth1 weight 1 nexthop via 192.168.2.1 dev eth3 weight 1 nexthop via 192.168.0.1 dev eth4 weight 100 (omrvpn_intf: 
Jan  5 15:38:47 OpenMPTCProuter user.notice post-tracking-001-post-tracking: wan2 (eth2) switched up

vempire-ghost, 2024-01-06:
I did some more tests, my master WAN is WAN4, every time I disconnect its internet, without unplugging the network cable, the OMR is unable to browse unless this entry appears in the log.
Jan  6 04:20:31 OpenMPTCProuter user.notice post-tracking-001-post-tracking: wan4 (eth4) switched off because check error and ping from 192.168.0.52 error (4.2.2.1)
Jan  6 04:20:31 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Delete default route to X.X.X.X dev eth4

Sometimes, the OMR starts working again after deleting the route, but only for a short time until it begins to show this entry in the log.
Jan  6 04:27:24 OpenMPTCProuter daemon.info omr-tracker-xray: Server (X.X.X.X) seems down, no answer to ping
Jan  6 04:27:29 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can't ping server vps on X.X.X.X, no server API answer on X.X.X.X)
Jan  6 04:27:33 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can't ping server vps on X.X.X.X, no server API answer on X.X.X.X)
Jan  6 04:27:40 OpenMPTCProuter user.notice OMR-Tracker-Server: Master server vps down (X.X.X.X)
When this entry appears, the OMR loses connectivity entirely again, even though there are two more functional WANs, and my VPS is working normally.

As it is now, whenever my master WAN experiences a connection drop, the OMR completely loses the ability to navigate the internet until it regains connectivity, which goes entirely against the idea of a system that should provide aggregation and redundancy.

vempire-ghost, 2024-01-06:
What exactly does this configuration do?
[image]

After marking it, the OMR log showed this.
Jan  6 04:36:56 OpenMPTCProuter user.notice post-tracking-001-post-tracking: wan4 (eth4) switched off because check error and ping from 192.168.0.52 error (4.2.2.1)
Jan  6 04:36:56 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Delete default route to X.X.X.X dev eth4
Jan  6 04:36:58 OpenMPTCProuter user.notice post-tracking-001-post-tracking: wan3 (eth3) switched off because check error and ping from 192.168.2.50 error (4.2.2.1)
Jan  6 04:36:58 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Delete default route to X.X.X.X dev eth3
Jan  6 04:36:58 OpenMPTCProuter user.notice post-tracking-001-post-tracking: wan2 (eth2) switched off because check error and ping from 192.168.0.51 error (4.2.2.1)
Jan  6 04:36:58 OpenMPTCProuter user.notice post-tracking-001-post-tracking: wan1 (eth1) switched off because check error and ping from 192.168.0.50 error (4.2.2.1)
Jan  6 04:36:58 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Delete default route to X.X.X.X dev eth1
Jan  6 04:36:58 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Delete default route to X.X.X.X dev eth2
However, everything returned to normal functioning, even though the log showed that all routes were deleted.

Ysurac, 2024-01-06:
"Server http test", add a test that check server API answer via HTTP, if not the connection is set as down.
I will add additional check for server status, as it decide it's down only after one check (and as server is a domain, the issue can be domain resolution instead of connection test).

vempire-ghost, 2024-01-06:
From what I've noticed in my tests, the issue is that the tracker cannot consistently detect whether a WAN is or isn't connected to the internet. If this WAN is the master, the entire system loses internet because the OMR thinks the VPS is offline.

The ideal scenario would be for the OMR server, upon failing a master test, to switch to the next functional WAN until the master passes the test again. However, for this to happen, the tracker needs to be more precise in its detection.
Because the longer the tracker takes to notice that the WAN is without internet, the more time the OMR will fail to create new connections as it continues trying to establish connections through the master.

Could you explain the logic behind the "wan master" issue? OMR always uses the master to check if the VPS is online, and when it can't verify that, shouldn't it use any other online WAN at the moment instead of persistently trying the one that is offline?

vempire-ghost, 2024-01-07:
I found a way to work around the issue by setting the "tun0" interface as the master and the other WANs as "On". This way, it seems that OMR starts using the VPN to communicate with the VPS when it's online and functional. If it's offline, OMR chooses any available WAN until the VPN is active.

Since the VPN is set up using multipath, there is no failure when any connection goes offline, ensuring that OMR doesn't encounter issues checking the status of the VPS or the proxy.

To configure "tun0" as the master, I had to disable multipath on all WANs. OMR then automatically set up "tun0" as the master. After that, I configured all WANs as "ON," and this allowed OMR to maintain "tun0" as the master. Otherwise, OMR would consistently switch off the master on "tun0."

Could you make a simpler option to address this issue? Using VPN for both OMR Proxy Tracker Settings and OMR Server Tracker Settings to check the server instead of any of the WANs has brought significant stability to the system with an unstable WAN master.

Ysurac, 2024-01-07:
I don't know how this can give any good result, it's not a good option in any case.
I'm working on a better WAN detection failure.

vempire-ghost, 2024-01-07:
I don't know how, but in my case, it completely solved my problem. I forced disconnections several times on all WANs alternately, and not once did OMR notify that the VPS was offline or disable the proxy because of it.
However, when I set the WAN as the master, every time it goes offline, OMR loses complete connectivity and starts notifying in the log that it disabled the proxy until the WAN regains internet. For now, I'll leave it like this and continue testing.

vempire-ghost, 2024-01-08:
> I don't know how this can give any good result, it's not a good option in any case. I'm working on a better WAN detection failure.

How is the verification of the OMR tracker for the VPS done today? Is it by sending pings from the WAN master to the server's address, and if there's no response, is it considered offline?

Would it be feasible, after confirming that the VPS is online, to establish a continuous MPTCP connection with the VPS and use it for this verification? This way, in the event that one of the WANs loses internet connectivity, MPTCP would maintain the active connection, allowing OMR to continue verifying that the VPS is online.

xzjq, 2024-01-10:
I experienced this as well. Lost power, which took out 2 out of 3 WANs including the master. Another WAN had battery backup (MPTCP enabled, but not master). Lost all internet.

closed 2024-04-14 by github-actions[bot]