Ysurac/openmptcprouter #3112: VPS Server Is going down randomly..

iamdivyanshverma, 2024-01-06:
## Current Behavior
Hi @Ysurac , My VPS Server is going down randomly , In logread it says Can't Get VPS Token . If i tries to login to the is server its not accessable i have to reboot the VPS Server from the console it order to make it work .
Tried with multiple VPS Provider like AWS,Vultur,etc.
Please Let me know if you need more details .
Thanks

## Specifications

  - OpenMPTCProuter version: v0.60beta1-6.1
  - OpenMPTCProuter VPS version:  OpenMPTCProuter VPS 0.1029-test 
  - OpenMPTCProuter VPS provider: AWS/Vultur
  - OpenMPTCProuter platform: RPI4/x86_64
  - Country: India

Ysurac, 2024-01-06:
Your VPS is down and you don't get any connection anymore ? Or only status display problem ? (token is not always updated and should not make major problem after initial configuration)

iamdivyanshverma, 2024-01-07:
yes my VPS is down , i lost connection ...

Ysurac, 2024-01-07:
When you say console, you mean VNC console from the provider ? If not, try it.
Check on console log if there is a kernel panic (or anything strange in ```dmesg```).
If not:
* check if you can ping outside from the VPS (```ping 1.1.1.1``` for example)
* check firewall configuration, shorewall is used by default so check in ```/etc/shorewall/params.net``` that the correct interface is indicated (should be the interface used to exit on internet)
* check in ```/etc/shorewall/rules``` that you didn't add any rules that can break access
* check server uptime, to check if it doesn't reboot or shutdown

What is the OS used on the VPS ? Should be Debian 12 for 6.1 snapshots.

kmrinal, 2024-01-07:
@Ysurac The servers actually hangs and there is no way to get inside it (via ssh or via VPS provider's access methods). If we reboot the server, it does come back and behaves normally, until it hangs again after few hours. 

The OS is Debian 12.
 
PS: I work with @iamdivyanshverma

Ysurac, 2024-01-07:
Strange that provider console doesn't work, should give a screen even if it's a kernel crash.
What is the kernel release ? ```uname -a``` ?

kmrinal, 2024-01-07:
Yeah, it was quite strange that console would not load up.

Uname Info is:

Linux Bonging-Server 6.1.0-17-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.69-1 (2023-12-30) x86_64 GNU/Linux

Ysurac, 2024-01-07:
I'm launching tests on same kernel with latest snapshot script.
What is the proxy and VPN used ? How many connections ? Anything special when you have the crash ? I will try to reproduce it.

kmrinal, 2024-01-07:
The VPN is Glorytun TCP and Proxy is Shadowsocks. Its a standard configuration and only one OMR device is connected to the server. 

The servers have stopped when in use and also when left idle for sometime. We havent observed any specific pattern for it though.

Thank you Ysurac.

Ysurac, 2024-01-08:
My VPS is still running without crash, do you know how many hours for a crash ?
You have nothing more installed than the VPS script on the server ?

Ysurac, 2024-01-12:
I was able to reproduce the issue.
It was also on 6.1.0-17 kernel (no problem on 6.1.0-13 kernel for now).
I'm now testing with kernel 6.5 on same server: ```apt-get install linux-image-6.5.0-0.deb12.4-cloud-amd64 linux-headers-6.5.0-0.deb12.4-cloud-amd64```

kmrinal, 2024-02-02:
@Ysurac Did server performance/stability improve with 6.5 Kernel ?

iamdivyanshverma, 2024-02-15:
Hi again , looks like we are still having crash in 6.5.0-0.deb12.4-cloud-amd64..

Ysurac, 2024-02-15:
I didn't have any crash on same VM with 6.5 since 33 days.
You can try to follow this guide to enable crash dump: https://www.cyberciti.biz/faq/how-to-on-enable-kernel-crash-dump-on-debian-linux/
This is the only solution I have to get a log that may help to find where is the issue.

iamdivyanshverma, 2024-02-15:
Thanks for the quick reply, i will try this and get back to you if i find something.

closed 2024-05-20 by github-actions[bot]