Ysurac/openmptcprouter #3114: e1000e eno1 Reset adapter unexpectedly

dooley1981, 2024-01-06:
## Current Behavior
I keep getting this error pop up at random times and I will loose connection for about 10 seconds and it goes back online. It will run for hours and did it happens again. I don't recall seeing this until I went to the 6.1 kernel.
[    0.000000] Linux version 6.1.69 (runner@fv-az1019-229) (x86_64-openwrt-linux-musl-gcc (OpenWrt GCC 12.3.0 r0+24699-b196a9f6ce) 12.3.0, GNU ld (GNU Binutils) 2.40.0) #0 SMP PREEMPT_DYNAMIC Thu Dec 28 05:10:32 2023
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz root=PARTUUID=cd8313bb-02 rootwait console=tty0 console=ttyS0,115200n8 noinitrd
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009dbff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009dc00-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000e0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x00000000aab7efff] usable
[    0.000000] BIOS-e820: [mem 0x00000000aab7f000-0x00000000abe7efff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000abe7f000-0x00000000abf7efff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000abf7f000-0x00000000abffefff] ACPI data
[    0.000000] BIOS-e820: [mem 0x00000000abfff000-0x00000000abffffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000ac000000-0x00000000af1fffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000e0000000-0x00000000efffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed10000-0x00000000fed13fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed18000-0x00000000fed19fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed1c000-0x00000000fed1ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff800000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000024edfffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] SMBIOS 2.7 present.
[    0.000000] DMI: Hewlett-Packard HP EliteBook 840 G1/198F, BIOS L71 Ver. 01.49 02/24/2020
[    0.000000] tsc: Fast TSC calibration using PIT
[    0.000000] tsc: Detected 2693.773 MHz processor
[    0.000663] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000667] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000676] last_pfn = 0x24ee00 max_arch_pfn = 0x400000000
[    0.000785] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[    0.001365] last_pfn = 0xac000 max_arch_pfn = 0x400000000
[    0.011293] Using GB pages for direct mapping
[    0.011822] ACPI: Early table checksum verification disabled
[    0.011825] ACPI: RSDP 0x00000000000F2FE0 000024 (v02 HPQOEM)
[    0.011830] ACPI: XSDT 0x00000000ABFFE120 0000B4 (v01 HPQOEM SLIC-MPC 00000001      01000013)
[    0.011836] ACPI: FACP 0x00000000ABFFC000 00010C (v05 HPQOEM 198F     00000001 HP   00000001)
[    0.011842] ACPI: DSDT 0x00000000ABFD1000 02555B (v02 HPQOEM 198F     00000001 INTL 20110112)
[    0.011845] ACPI: FACS 0x00000000ABDE3000 000040

[    0.176037] ACPI: EC: interrupt blocked
[    1.919617] ACPI: EC: EC_CMD/EC_SC=0x66, EC_DATA=0x62
[    1.919912] ACPI: \_SB_.PCI0.LPCB.EC0_: Boot DSDT EC used to handle transactions
[    1.920325] ACPI: Interpreter enabled
[    1.920562] ACPI: PM: (supports S0 S5)
[    1.920794] ACPI: Using IOAPIC for interrupt routing
[    1.921101] PCI: Using host bridge windows from ACPI; if necessary, use "pci=nocrs" and report a bug
[    1.921593] PCI: Using E820 reservations for host bridge windows
[    1.922224] ACPI: Enabled 8 GPEs in block 00 to 7F
[    1.924128] ACPI: PM: Power Resource [APPR]
[    1.927329] ACPI: PM: Power Resource [COMP]
[    1.927820] ACPI: PM: Power Resource [LPP]
[    1.929527] ACPI: PM: Power Resource [PXP6]
[    2.079401] ACPI Error: AE_ERROR, Returned by Handler for [PCI_Config] (20220331/evregion-291)
[    2.079875] ACPI Error: Aborting method \_SB.PCI0.RP06.PXP6._ON due to previous error (AE_ERROR) (20220331/psparse-529)
[    2.082652] ACPI: PCI Root Bridge [PCI0] (domain 0000 [bus 00-fe])
[    2.083002] acpi PNP0A08:00: _OSC: OS supports [Segments MSI HPX-Type3]
[    2.083366] acpi PNP0A08:00: _OSC: not requesting OS control; OS requires [ExtendedConfig ASPM ClockPM MSI]
[    2.084378] acpi PNP0A08:00: FADT indicates ASPM is unsupported, using BIOS configuration
[    2.085359] PCI host bridge to bus 0000:00
[    2.085608] pci_bus 0000:00: root bus resource [io  0x0000-0x0cf7 window]
[    2.085982] pci_bus 0000:00: root bus resource [io  0x0d00-0xffff window]
[    2.086354] pci_bus 0000:00: root bus resource [mem 0x000a0000-0x000bffff window]
[    2.086772] pci_bus 0000:00: root bus resource [mem 0xaf200000-0xdfffffff window]
[    2.087190] pci_bus 0000:00: root bus resource [mem 0xf0000000-0xfed3ffff window]
[    2.087609] pci_bus 0000:00: root bus resource [mem 0xfed45000-0xfedfffff window]
[    2.088027] pci_bus 0000:00: root bus resource [mem 0xfee01000-0xffffffff window]
[    2.088445] pci_bus 0000:00: root bus resource [bus 00-fe]
[    2.088773] pci 0000:00:00.0: [8086:0a04] type 00 class 0x060000
[    2.089188] pci 0000:00:02.0: [8086:0a16] type 00 class 0x030000
[... 702 lines cut ...]
[ 7486.640531]   TDH                  <2a>
[ 7486.640531]   TDT                  <47>
[ 7486.640531]   next_to_use          <47>
[ 7486.640531]   next_to_clean        <28>
[ 7486.640531] buffer_info[next_to_clean]:
[ 7486.640531]   time_stamp           <1000af509>
[ 7486.640531]   next_to_watch        <2a>
[ 7486.640531]   jiffies              <1000af770>
[ 7486.640531]   next_to_watch.status <0>
[ 7486.640531] MAC Status             <80083>
[ 7486.640531] PHY Status             <796d>
[ 7486.640531] PHY 1000BASE-T Status  <3800>
[ 7486.640531] PHY Extended Status    <3000>
[ 7486.640531] PCI Status             <10>
[ 7487.600287] e1000e 0000:00:19.0 eth0: Reset adapter unexpectedly
[ 7491.382544] e1000e 0000:00:19.0 eth0: NIC Link is Up 1000 Mbps Full Duplex, Flow Control: Rx/Tx

## Context (Environment)
I keep getting this error pop up at random times and I will loose connection for about 10 seconds and it goes back online. It will run for hours and did it happens again. I don't recall seeing this until I went to the 6.1 kernel.
## Specifications

  - OpenMPTCProuter version: v0.60beta1-6.1 r0+24699-b19
  - OpenMPTCProuter VPS provider: Cyclone
  - OpenMPTCProuter platform: x86_64  Intel(R) Core(TM) i7-4600U CPU @ 2.10GHz
  - Country: United States

Ysurac, 2024-01-06:
It's under a VM or not ?
Check CPU temperature if it's ok, check memory.
Do you have multiple ethernet interfaces ?

dooley1981, 2024-01-06:
Under VM
OpenMPTCProuter (192.168.2.1)

Version 0.60beta1-6.1

(Latest available version 0.59.1-6.1)

Load: 0.38 0.28 0.27

Core temp: 48.0

Uptime: 14h 19m 38s

Lan address: 192.168.2.1

Lan range: 192.168.2.100-192.168.2.248

Yes I have one ethernet port coming from my laptop connected to a dummy switch to make 8 connections with devices connected to that plus 2 Wan connections.

dooley1981, 2024-01-06:
Now speed changed to 100Mbps max. Look like NIC has change to 100Mbps instead of 1000Mbps

[ 7487.600287] e1000e 0000:00:19.0 eth0: Reset adapter unexpectedly
[ 7491.382544] e1000e 0000:00:19.0 eth0: NIC Link is Up 1000 Mbps Full Duplex, Flow Control: Rx/Tx
[52738.159910] e1000e 0000:00:19.0 eth0: NIC Link is Down
[52751.593264] e1000e 0000:00:19.0 eth0: NIC Link is Up 100 Mbps Full Duplex, Flow Control: Rx/Tx

Ysurac, 2024-01-06:
Do you know the chipset used on the network card ? It's an Intel I21x ?
You can try a ```ethtool -K eth0 tx off rx off``` but not sure this really solve the issue for a long time...

dooley1981, 2024-01-06:
Intel I219-LM 10/100/1000 GbE, you spot on.

Where do I put this ethtool command?

Also is there any other device better to use other than a computer or laptop?

Ysurac, 2024-01-07:
via SSH on the router.
There is lot of devices, what aggregate speed you want ? Do you use LTE modems ?
In most case a small x86_64 box with AES-NI support is great.

dooley1981, 2024-01-07:
Currently im aggregating around 700Mbps. What box you recommend?

Ysurac, 2024-01-07:
A Raspberry PI 5 or any x86_64 box with AES support, there is many on Amazon or Aliexpress for example.

dooley1981, 2024-01-08:
Hi, I tried the ethtool -K eth0 tx off rx off command. It's been working fine since I ran the command. Does this stick with reboot?

Ysurac, 2024-01-08:
You can add the command on ```/etc/rc.local``` to have it after a reboot.

dooley1981, 2024-01-08:
Okay thanks!

closed 2024-01-08 by dooley1981