Ysurac/openmptcprouter #3115: Frequent outages in experienced internet, all green

klucsik, 2024-01-08:
## Expected Behavior
Nice and stable internet connection

## Current Behavior
I get 3-6 seconds of outages and then everything goes normal in regular webbrowsing activity. (torrent seems to be continously working)
System logs are flooded with this kind of unreachable loglines: https://pastebin.com/BPS5Bwyq
When an initial load of a page finishes, I get good connection, but when I visit a new page, it takes quite some time for initial load, and there is a lot of failed request:
For example: 
[image]

## Specifications
Setup: 
[image]
a 5G mobile internet, and a landline, both should be cabale of 300 Mb/s download.

  - OpenMPTCProuter version: 0.60beta-1.61 (same issue present with 0.59.1-6.1)
  - OpenMPTCProuter VPS version: 0.1029-test 6.1.0-13-amd64
  - OpenMPTCProuter VPS provider:  https://alpenhost.at/ 
  - OpenMPTCProuter platform: RPI4

How do I start debugging this? The naked networks produce the same issue... or kinda every heckin network here in Austria... I hoped openmptcprouter would be able to create some stable network out of these

Ysurac, 2024-01-09:
Put network interfaces in WAN firewall zone.
Try to use Shadowsocks-Rust in System->OpenMPTCProuter, Wizard tab, "Advanced settings" checkbox.
Check that you don't have any problem to reach websites by using curl from the VPS.

vempire-ghost, 2024-01-09:
You need to set up the master WAN with the most stable internet connection. The way MPTCP works is that it always establishes a normal TCP connection through the master and then sets up subflows. However, if at the moment it tries to initiate a new connection, and the master has no internet, it will fail to establish the connection, as you are reporting.

If both of your connections are unstable, unfortunately, in the current setup, I believe OMR will only improve the stability of already established connections.

klucsik, 2024-01-13:
Thank you for your responses,
WAN zone: interfaces are in the wan zone, as in the picture. Is this warning case sensitive?
[image]

I try to test the VPS, I found one occassion where it didnt work, maybe thats the proble, I'll try another VPS soon.
[image]

(upload iamges to this post was never succeded at the first try :/ )

I dont find this Shadowsocks-Rust settings, is it available in my not-stable version(0.60beta-1.61)? 

I require a stable connection to work, and would like to be able to host stuff on a stable IP address from my network, if not OMR, what should I try out?