Ysurac/openmptcprouter #3119: omr-bypass

rdmitry0911, 2024-01-11:
I have openmptcprouter v0.60beta1-6.1 r0+23789-ce6ad123e7 installed and running. I'd like to configure omr-bypass for a specific domain to not use mptcp default interface pointed to vps, but to use any of the available interfaces. However, when I put these rules  in Domains section of omr-bypass configuration, the system uses only eth1 from the first rule and if eth1 becomes unavailable, the domain reddit.com can not be reached at all.
```
Enabled      | Domain     | Interface | Note |  
Yes          | reddit.com | eth1      | none |  
Yes          | reddit.com | eth2      | none |  
```
What rules should I put in config to allow using of any available interfaces, but restrict  mptcp default interface?

Ysurac, 2024-01-11:
Don't set any interface and, as indicated on the interface, it will use master interface and failover to another interface when unavailable.

rdmitry0911, 2024-01-11:
> Don't set any interface and, as indicated on the interface, it will use master interface and failover to another interface when unavailable.

Thank you. It does work when the interface is not in the system anymore.  However,  in my case sometimes one of the interfaces looses connectivity. It is still present in the system, but does not have access to the internet.  I can determine this state with an easy ping test. Is it possible to tell to omr-bypass  in this case to mark this interface as unavailable?

Ysurac, 2024-01-11:
"omr-tracker" should detect when interface is down and set it unavailable.
Do you have something in Status->System log when you loose connectivity on an interface ?

rdmitry0911, 2024-01-11:
I started to debug the problem and found out that there is kind of problem in a script responsible for switching interfaces.
I set up both of my interfaces in omr-tracker expecting rotation of the interfaces in default route in case of failure of the master interface

iptables -nvL -t mangle  shows this in the end. 

```
Chain omr-bypass-local (1 references)
 pkts bytes target     prot opt in     out     source               destination         
    0     0 MARK       all  --  *      *       0.0.0.0/0            0.0.0.0/0            match-set omr_dst_bypass_tun0 dst MARK set 0x5391500
    0     0 RETURN     all  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x5391500
    0     0 MARK       all  --  *      *       0.0.0.0/0            0.0.0.0/0            match-set omr_dst_bypass_eth1 dst MARK set 0x5397
    0     0 RETURN     all  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x5397
    0     0 MARK       all  --  *      *       0.0.0.0/0            0.0.0.0/0            match-set omr_dst_bypass_eth2 dst MARK set 0x5398
    0     0 RETURN     all  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x5398
    0     0 MARK       all  --  *      *       0.0.0.0/0            0.0.0.0/0            match-set omr_dst_bypass_eth0 dst MARK set 0x5396
    0     0 RETURN     all  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x5396
    0     0 MARK       all  --  *      *       0.0.0.0/0            0.0.0.0/0            match-set omr_dst_bypass_lo dst MARK set 0x5395
    0     0 RETURN     all  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x5395
    0     0 MARK       udp  --  *      *       0.0.0.0/0            0.0.0.0/0            udp dpts:12346:12352 MARK set 0x539
    0     0 MARK       all  --  *      *       0.0.0.0/0            0.0.0.0/0            match-set omr_dst_bypass_all dst MARK set 0x539
    0     0 RETURN     all  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x539

```
539 is the destination for all packets routed via the rule in omr-bypass without interface specified

root@OpenMPTCProuter:~# ip rules show 
```
0:      from all lookup local
0:      from all oif eth2 lookup 8
0:      from all oif eth1 lookup 7
0:      from 10.255.255.2 lookup 1500
0:      from 192.168.1.100 lookup 8
0:      from 192.168.10.100 lookup 7
0:      from all oif tun0 lookup 1500
1:      from all fwmark 0x539 lookup 991337  # this is one we are looking for
1:      from all fwmark 0x5395 lookup 5
1:      from all fwmark 0x5398 lookup 8
1:      from all fwmark 0x5397 lookup 7
100:    from all lookup lan
10000:  from 192.168.99.1 lookup lan
20000:  from all to 192.168.99.1/24 lookup lan
32766:  from all lookup main
32767:  from all lookup default
90002:  from all iif lo lookup lan
```
in my case 991337 is the table where all packets from omr-bypass rules without specified interface drop.
During test I unplugged the cable from the modem connected to the master interface
in the beginning the script switched dead interface to alive one after some delay, however, after several (5-6) cable unplugging/plugging  back the script stopped working and stuck on this:
```
root@OpenMPTCProuter:~# ip ro sh t 991337
default via 192.168.10.1 dev eth1 linkdown
```
as a workaround
```
/etc/init.d/omr-tracker restart 
```
solves the problem, but this is very inconvenient.

Ysurac, 2024-01-12:
I still need Status->System log and I would also need the result of ```uci show openmptcprouter``` via SSH on the router (remove password/ip) and the name of the interface that should be down.

Ysurac, 2024-01-12:
I found the issue, this will be solve in next snapshot.

rdmitry0911, 2024-01-12:
> I found the issue, this will be solve in next snapshot.

Good! Thank you :)

closed 2024-01-12 by rdmitry0911