Ysurac/openmptcprouter #3124: VPS Fail install

DooMaria, 2024-01-13:
Hi, i get this error when i want to install on VPS:

root@testme:~# wget https://www.openmptcprouter.com/server/debian-x86_64.sh
--2024-01-13 14:09:48--  https://www.openmptcprouter.com/server/debian-x86_64.sh
Resolving www.openmptcprouter.com (www.openmptcprouter.com)... 169.150.247.38, 2400:52e0:1e00::1080:1
Connecting to www.openmptcprouter.com (www.openmptcprouter.com)|169.150.247.38|:443... connected.

root@testme:~# wget https://www.openmptcprouter.com/server/debian-x86_64.sh
--2024-01-13 14:09:48--  https://www.openmptcprouter.com/server/debian-x86_64.sh
Resolving www.openmptcprouter.com (www.openmptcprouter.com)... 169.150.247.38, 2400:52e0:1e00::1080:1
Connecting to www.openmptcprouter.com (www.openmptcprouter.com)|169.150.247.38|:443... connected.

i tried on many diff server , i believe my region blocked downloading this url..
any way to install ?

DooMaria, 2024-01-13:
Last login: Sat Jan 13 14:17:32 2024 from 204.18.107.67
root@testme:~# wget -O - https://www.openmptcprouter.com/server/ubuntu20.04-x86_64.sh | sh
--2024-01-13 14:29:06--  https://www.openmptcprouter.com/server/ubuntu20.04-x86_64.sh
Resolving www.openmptcprouter.com (www.openmptcprouter.com)... 138.199.46.68, 2400:52e0:1500::868:1
Connecting to www.openmptcprouter.com (www.openmptcprouter.com)|138.199.46.68|:443... connected.
Unable to establish SSL connection.

for ubuntu

i will try manual install

DooMaria, 2024-01-13:
okay still not fully install some errors..

root@testme:~# wget -O - https://www.uplooder.net/f/tl/37/b0c7e233e0f426bcd0da00                    b005bd2e9d/debian-x86-64.sh | sh
--2024-01-13 14:29:54--  https://www.uplooder.net/f/tl/37/b0c7e233e0f426bcd0da00                    b005bd2e9d/debian-x86-64.sh
Resolving www.uplooder.net (www.uplooder.net)... 5.9.68.247
Connecting to www.uplooder.net (www.uplooder.net)|5.9.68.247|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 69751 (68K) [text/x-sh]
Saving to: ‘STDOUT’

-                   100%[===================>]  68.12K  41.0KB/s    in 1.7s

2024-01-13 14:30:00 (41.0 KB/s) - written to stdout [69751/69751]

Check user...
Check Linux version...
Check architecture...
Check virtualized environment
Check about broken packages...
Check if OpenMPTCProuter VPS is already installed...
Update mode
Remove lock and update packages list...
Hit:1 http://archive.ubuntu.com/ubuntu focal InRelease
Hit:2 http://archive.ubuntu.com/ubuntu focal-updates InRelease
Hit:3 http://archive.ubuntu.com/ubuntu focal-backports InRelease
Hit:4 http://archive.ubuntu.com/ubuntu focal-security InRelease
Reading package lists... Done
Install apt-transport-https, gnupg and openssh-server...
Reading package lists... Done
Building dependency tree
Reading state information... Done
gnupg is already the newest version (2.2.19-3ubuntu2.2).
gnupg set to manually installed.
openssh-server is already the newest version (1:8.2p1-4ubuntu0.11).
The following packages were automatically installed and are no longer required:
  linux-headers-5.4.0-81 linux-headers-5.4.0-81-generic
  linux-image-5.4.0-81-generic linux-modules-5.4.0-81-generic
  linux-modules-extra-5.4.0-81-generic
Use 'apt autoremove' to remove them.
The following NEW packages will be installed:
  apt-transport-https
0 upgraded, 1 newly installed, 0 to remove and 5 not upgraded.
Need to get 1704 B of archives.
After this operation, 162 kB of additional disk space will be used.
Get:1 http://archive.ubuntu.com/ubuntu focal-updates/universe amd64 apt-transpor                    t-https all 2.0.10 [1704 B]
Fetched 1704 B in 0s (7422 B/s)
Selecting previously unselected package apt-transport-https.
(Reading database ... 144921 files and directories currently installed.)
Preparing to unpack .../apt-transport-https_2.0.10_all.deb ...
Unpacking apt-transport-https (2.0.10) ...
Setting up apt-transport-https (2.0.10) ...
Add OpenMPTCProuter repo...
Explanation: Prefer OpenMPTCProuter provided packages over the Debian native one                    s
Package: *
Pin: origin repo.openmptcprouter.com
Pin-Priority: 1001
--2024-01-13 14:30:17--  https://repo.openmptcprouter.com/openmptcprouter.gpg.ke                    y
Resolving repo.openmptcprouter.com (repo.openmptcprouter.com)... 138.199.46.68,                     2400:52e0:1500::868:1
Connecting to repo.openmptcprouter.com (repo.openmptcprouter.com)|138.199.46.68|                    :443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 1241 (1.2K) [application/octet-stream]
Saving to: 'STDOUT'

-                   100%[===================>]   1.21K  --.-KB/s    in 0s

2024-01-13 14:30:19 (43.0 MB/s) - written to stdout [1241/1241]

OK
Install mptcp kernel and shadowsocks...
Hit:1 http://archive.ubuntu.com/ubuntu focal InRelease
Hit:2 http://archive.ubuntu.com/ubuntu focal-updates InRelease
Hit:3 http://archive.ubuntu.com/ubuntu focal-backports InRelease
Hit:4 http://archive.ubuntu.com/ubuntu focal-security InRelease
Get:5 http://archive.ubuntu.com/ubuntu bionic-backports InRelease [83.3 kB]
Get:6 http://archive.ubuntu.com/ubuntu bionic InRelease [242 kB]
Get:7 http://archive.ubuntu.com/ubuntu bionic-backports/main amd64 Packages [53.                    3 kB]
Get:8 http://archive.ubuntu.com/ubuntu bionic-backports/main Translation-en [14.                    6 kB]
Get:9 http://archive.ubuntu.com/ubuntu bionic/universe amd64 Packages [8570 kB]
Get:10 http://archive.ubuntu.com/ubuntu bionic/universe Translation-en [4941 kB]
Ign:11 https://repo.openmptcprouter.com buster InRelease
Err:12 https://repo.openmptcprouter.com buster Release
  Could not wait for server fd - select (11: Resource temporarily unavailable) [                    IP: 138.199.46.68 443]
Reading package lists... Done
E: The repository 'https://repo.openmptcprouter.com buster Release' does not hav                    e a Release file.
N: Updating from such a repository can't be done securely, and is therefore disa                    bled by default.
N: See apt-secure(8) manpage for repository creation and user configuration deta                    ils.
root@testme:~#

DooMaria, 2024-01-13:
root@testme:~# wget https://www.openmptcprouter.com/server/ubuntu20.04-x86_64.sh
--2024-01-13 14:38:50--  https://www.openmptcprouter.com/server/ubuntu20.04-x86_64.sh
Resolving www.openmptcprouter.com (www.openmptcprouter.com)... 138.199.46.68, 2400:52e0:1500::868:1
Connecting to www.openmptcprouter.com (www.openmptcprouter.com)|138.199.46.68|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 69751 (68K) [application/octet-stream]
Saving to: ‘ubuntu20.04-x86_64.sh’

ubuntu20.04-x86_64.sh    100%[==================================>]  68.12K   222KB/s    in 0.3s

2024-01-13 14:38:50 (222 KB/s) - ‘ubuntu20.04-x86_64.sh’ saved [69751/69751]

root@testme:~# sh ubuntu20.04-x86_64.sh
Check user...
Check Linux version...
Check architecture...
Check virtualized environment
Check about broken packages...
Check if OpenMPTCProuter VPS is already installed...
Update mode
Update OpenMPTCProuter repo key
--2024-01-13 14:38:57--  http://repo.openmptcprouter.com/openmptcprouter.gpg.key
Resolving repo.openmptcprouter.com (repo.openmptcprouter.com)... 138.199.46.68, 2400:52e0:1500::868:1
Connecting to repo.openmptcprouter.com (repo.openmptcprouter.com)|138.199.46.68|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 1241 (1.2K) [application/octet-stream]
Saving to: 'STDOUT'

-                        100%[==================================>]   1.21K  --.-KB/s    in 0s

2024-01-13 14:38:57 (25.1 MB/s) - written to stdout [1241/1241]

OK
Update repo.openmptcprouter.com key
--2024-01-13 14:38:57--  https://repo.openmptcprouter.com/openmptcprouter.gpg.key
Resolving repo.openmptcprouter.com (repo.openmptcprouter.com)... 138.199.46.68, 2400:52e0:1500::868:1
Connecting to repo.openmptcprouter.com (repo.openmptcprouter.com)|138.199.46.68|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 1241 (1.2K) [application/octet-stream]
Saving to: 'STDOUT'

-                        100%[==================================>]   1.21K  --.-KB/s    in 0s

2024-01-13 14:38:58 (59.5 MB/s) - written to stdout [1241/1241]

OK
Remove lock and update packages list...
Hit:1 http://archive.ubuntu.com/ubuntu focal InRelease
Get:2 http://archive.ubuntu.com/ubuntu focal-updates InRelease [114 kB]
Hit:3 http://archive.ubuntu.com/ubuntu focal-backports InRelease
Hit:4 http://archive.ubuntu.com/ubuntu focal-security InRelease
Hit:5 http://archive.ubuntu.com/ubuntu bionic-backports InRelease
Hit:6 http://archive.ubuntu.com/ubuntu bionic InRelease
Ign:7 https://repo.openmptcprouter.com buster InRelease
Err:8 https://repo.openmptcprouter.com buster Release
  Could not wait for server fd - select (11: Resource temporarily unavailable) [IP: 138.199.46.68 443]
Reading package lists... Done
E: The repository 'https://repo.openmptcprouter.com buster Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
root@testme:~#

Ysurac, 2024-01-13:
*.openmptcprouter.com is using a CDN. You can wait a little for CDN update or tell me your country so I can try to check if data are ok.

Ysurac, 2024-01-13:
What is the VPS provider and what is the OS used ?

DooMaria, 2024-01-13:
Server Region: Iran
VPS providers: 
Asiatech, Arvancloud , im testing one by one i let everyone know

for OS i tired everything ( Ubuntu 20.04 , Debian 10,11 )...

Ysurac, 2024-01-14:
For OS you should keep Debian 11 on stable release (Debian 12 if you try snapshot)
My website use a CDN (bunny), should not be blocked but I don't know...
Seems strange that IP is accessible and same IP not reachable a few seconds later here.

I can take a VPS from arvancloud if you still have no success to test, seems to be available from France and quite cheap. What is the VPS you took from them ?

he852100, 2024-01-15:
hosts or proxy
However, the performance after binding is not as good as directly connecting to a single network.

Ysurac, 2024-01-15:
@he852100 you don't give enough information to help. If you are also in Iran and connections is blocked (I don't know if it's the case), there is XRay using VLESS Reality protocol on port 443 that is now available in snapshot and may help in some case.

DooMaria, 2024-01-16:
Hey @Ysurac, you tagged the wrong person. O.o
Sir, I give up!
I discovered that I do not need to set up VPN inside OMPTCP-wrt because networks bond well together by default, even without installing VPN or VPS

DooMaria, 2024-01-16:
Very thankyou sir!

DooMaria, 2024-01-16:
UR A HERO

closed 2024-01-16 by DooMaria

Ysurac, 2024-01-16:
Without VPS side, OpenMPTCProuter is doing load balancing, this doesn't aggregate connections.

DooMaria, 2024-01-16:
@Ysurac if is load balancing alone, how am i getting higher bandwidth?

im confused O.o

Ysurac, 2024-01-16:
You can get better bandwidth due to TCP Congestion Control used (BBR) or speedtest you tried open several connections and can make use of load balancing.

DooMaria, 2024-01-16:
You are right, im using BBR2

DooMaria, 2024-01-16:
Should i play with Fullmesh subflows and other value numbers? @Ysurac

he852100, 2024-01-18:
The load balancing does not increase the daily network bandwidth. It is only effective in some network applications and may cause website abnormalities. You still need the server (VPS) to unify the IP of the network