Ysurac/openmptcprouter #3127: Bandwidth limit for RPI5?

cmsmith81, 2024-01-17:
In the thread below, [Ysurac](https://github.com/Ysurac) noted that the RPI4 should be able to reach around 550Mb/s.

Two Questions:
1) what is the estimated bandwidth limitation if using an RPI5?
2) where does the RPI4 limitation come from?  (ie what is the bottleneck?) Is it USB throughput limitations?

https://github.com/Ysurac/openmptcprouter/issues/2997

Thank you!

Ysurac, 2024-01-17:
Bandwidth limit with a RPI5 is about 900Mb/s, limited by the LAN 1Gb/s port.
RPI4 limit is the CPU.

cmsmith81, 2024-01-17:
I am using 2.5Gb/s ethernet adapters on the RPi5 for my LAN connection and WAN connections.  With that, what would the max bandwidth be?  Thanks for your quick reply!

Ysurac, 2024-01-17:
RPI5 have AES hardware, this also help to get higher speed.
But I don't know the speed it can reach using ethernet adapters.

cmsmith81, 2024-01-17:
Thanks for the information, once I run and test I'll post the results of my max bandwidth here.  (FYI I am using the RPi5 5 version of the nightly builds, and everything works well thus far).

lprhodes, 2024-04-14:
@cmsmith81 what results did you see please?

closed 2024-07-18 by github-actions[bot]