Ysurac/openmptcprouter #3128: How to change in realtime IP address WAN port on exit VPS?

GermanAizek, 2024-01-18:
@Ysurac very thx for active contibuting OpenMPTCProuter,
is it possible to do ip forwarding from router to VPS, with change to selected IP WAN in real time or with restart services?

Ysurac, 2024-01-19:
What do you want to do exactly and why ?
It's possible that I didn't think about a corner case here...

GermanAizek, 2024-01-19:
@Ysurac,
Are there 2 ISP providers, they come to openmptcprouter. Google IP Score rating on VPS is very bad, but he is very cheap, I am forced to constantly go through captchas and cloudflare waf, can I use cleaner IP of one of ISP providers?

Ysurac, 2024-01-19:
You can use OMR-ByPass to use directly a WAN for specified sites/IPs.

GermanAizek, 2024-01-19:
> OMR-ByPass
What is it and is there wiki or documentation on how to use it?

GermanAizek, 2024-01-19:
@Ysurac, Do I understand correctly that I will have to list all visited sites with WAF captchas for OMR bypass?

Ysurac, 2024-01-19:
You will have to list all captchas providers site. But in most case it's not necessary are there is not a lot of captchas providers and after a few times an IP is back in a "whitelist".

GermanAizek, 2024-01-19:
> You will have to list all captchas providers site.

Where can I read about how to create this list, configure, what syntax should I use, etc.

GermanAizek, 2024-02-12:
@Ysurac, how to use OMR-Bypass as example WAN0 - only SSH 22 and FTP 21 connection, WAN1 only HTTP 80 and HTTPS 443?

GermanAizek, 2024-02-12:
the site filtering method does not help, there are too many sites that consider it easier for a robot to bypass the entire protocol

closed 2024-05-17 by github-actions[bot]