Ysurac/openmptcprouter #3132: How to install VPS on Windows?

GermanAizek, 2024-01-22:
@Ysurac,
is it possible to install openmptcp VPS on VDS with Windows. If there are no options, is only installing a virtual machine suitable?

Ysurac, 2024-01-22:
OpenMPTCProuter is an OS, so if you want to run it under windows, you have to put it in a VM via Virtualbox for example.
Not sure on what can be done with WSL, but in any case it's not supported.

GermanAizek, 2024-01-22:
@Ysurac,
you didn't understand, I want to use this as VPS Windows. is it possible? or was it not originally intended that way?
[image]

Ysurac, 2024-01-23:
On the VPS only Debian 12 is supported. Not sure if WSL would also be possible but this is not supported and will not be supported.

closed 2024-01-25 by GermanAizek