Ysurac/openmptcprouter #3134: VPN is not running

WayLAN-2008, 2024-01-24:
## Expected Behavior
That the VPS works and that bonding is active

## Current Behavior
In my openMPTC router, it says the VPN is not running.
## Specifications

  - OpenMPTCProuter version: 
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter VPS provider :My own WIN10SERVER with Hyper-V Debian 10 install
  - OpenMPTCProuter platform: 

https://pastebin.com/A5nYpx5s

WayLAN-2008, 2024-01-24:
Forgot to put the screenshot
[image]

Ysurac, 2024-01-24:
It can't contact the VPS part:
* did you reboot VPS after script install ?
* can you go to https://<VPS_IP>:65500/ and get something ?

WayLAN-2008, 2024-01-25:
Yes i get this
[image]

WayLAN-2008, 2024-01-25:
What ports do i need to open. I did open 65101 and 65401 and 65500.

Ysurac, 2024-01-25:
Open range 65000-65535
Else you have the list here: https://github.com/Ysurac/openmptcprouter/wiki/Install-or-update-the-VPS

WayLAN-2008, 2024-01-25:
As soon as I am home I am going to try it out!

WayLAN-2008, 2024-01-25:
It works now! It still says the VPN is not running but it works thanks!

Ysurac, 2024-01-25:
VPN is used by default for UDP only. It should work.
You can check in Status->System log why it's not running.

closed 2024-04-29 by github-actions[bot]