Ysurac/openmptcprouter #3135: Add DDNS Client update public dns record when active VPS change

Mattieug, 2024-01-27:
Hi, thanks for this awesome solution ! 

I have searched in documentation and issues but I can't find a feature I need :

**Is your feature request related to a problem? Please describe.**
I have configured backup VPS in failover of the main one. It works perfectly. The problem is that I am exposing services to internet which become inaccessible when my external IP (VPS) change. Public DNS remains the same.

**Describe the solution you'd like**
Very concise, I would like to be able to trigger a DDNS update with my external IP when change occurs (VPS1, VPS2, VPSx, Direct WAN...).

**Describe alternatives you've considered**
Perhaps should be possible to add a script launch triggering on external IP change, based on the IP or simply when the outbound interface change ?

**Additional context**
Nop

Thanks !

Mattieug, 2024-01-29:
Hello, I have configured a standalone container which check continuously external adresse ip vs actual dns register. But it would be more convenient if integrated into OpenMPTCPRouter.

Thanks !

Ysurac, 2024-01-29:
You should be able to use ddns: https://openwrt.org/docs/guide-user/services/ddns/client
I can add it as a default package I think.

Mattieug, 2024-01-29:
Ok thanks.

I will keep my external solution waiting ddns available as as standard feature of OpenMPTCPRouter.

thanks

closed 2024-05-04 by github-actions[bot]

Network-Traditions, 2024-06-01:
We had the need to implement a solution for this as well.  We have numerous IPsec VPNs, servers (web, mail, nextcloud, voip, etc) that needed to function regardless of the VPS connection.  The following solution worked well, but does take about 5 minutes to fail-over due to the latency of dynamic dns updating.

We were able to use OpenMPTCProuter's (OMR) "DynamicDNS" under the service menu.  This worked well with our primary public DNS provider namecheap.  For our VPS point of presence public IP we setup an dynamic A record that could be updated by OMR's dynamic DNS service.  We also desired updating A records for OMR's various WAN interfaces in the event we wanted to access service directly.  For those we wrote a script to be called by OMR's dynamic DNS service:

#!/bin/bash
pip=$(uci get openmptcprouter.wan1.publicip 2> /dev/null)
if [ -z "$pip" ]; then echo "0.0.0.0"; else echo $pip; fi

This script get's the public IP address currently assigned to OMR's WAN1 interface.  If one doesn't exist, it passes "0.0.0.0" to OMR's dydnamic DNS to update our NameCheap service.  NameCheap will update the address accordingly and will not set a record if the updated value is "0.0.0.0".