Ysurac/openmptcprouter #3136: AdGuard DNS running on LAN, Is that weird?

adamb, 2024-01-27:
## Expected Behavior

I'm running Home Assistant and I installed AdGuard there.  It's on my LAN 192.168.7.196:53 for DNS queries.  My OMR is configured to use this as a [custom DNS Server](https://github.com/Ysurac/openmptcprouter/wiki/Custom-DNS) but initially this didn't work since the AdGuard DNS server is inside the LAN and the OMR couldn't reach it.  I'm using Eero for my LAN so I configured the Eero to port forward port 53 to this server.  Thus OMR uses 192.168.100.2 (My eero's IP) as its DNS server.

This works but it seems a little weird.  Is there a better way to do this?  I need to use OMR as my DNS server so bypass works.  With this config, OMR Bypass seems to work and AdGuard also seems to work.  DNS queries first hit OMR for bypass logic, then they get sent to AdGuard so that logic also works. 

## Is there a better way?

Does this expose my AdGuard server to the internet?  I don't think so.  Since there is no way to connect to my Eero directly wihtout going through the VPS and the VPS is locked down.  

I was thinking it might be better to run AdGuard on the VPS, but sending DNS queries out and back to my VPS seems like a bad idea.  Bouncing back and forth across my LAN with port forwarding seems better from a latency and bandwidth perspective.  

The only drawback I can see is that I have to remember to keep the Eero at the same LAN IP and the AdGuard (Home Assistant) also at the same IP.  I did put in 8.8.8.8 as a subsequent DNS server, so if my port forwarding or my LAN IPs change, this should keep things going without AdGuard if there is a failure.  

BTW, AdGuard running on Home Assistant has a pretty swanky dashboard.  It also makes figuring out OMR Bypass rules easy.  You can just look at the logs and see what domains are being hit by various apps.

[image]

## Specifications

  - OpenMPTCProuter version:  v0.59.1-5.4 r0+16594-ce92de8c8c
  - OpenMPTCProuter VPS version:  0.128
  - OpenMPTCProuter VPS provider: Linode
  - OpenMPTCProuter platform: RPi4

Ysurac, 2024-01-28:
No problem to use any DNS you want internal or external.
By default OMR use root server DNS because there are accessible from all ISP in the world and data doesn't go to any GAFA. But you are free to use any local or filtered DNS server. To keep OMR-ByPass working, as you are doing, you need to use OMR as IP for DNS.

adamb, 2024-02-05:
Since I'm using and internal server and it uses 94.140.14.14 as its upstream server, should I add this ip to the bypass?

closed 2024-05-10 by github-actions[bot]