Ysurac/openmptcprouter #3137: Installing OMR-VPS on ARM64 (Ubuntu 22.04)

GermanAizek, 2024-01-28:
@Ysurac,
Hello,
I understand correctly that there are no binaries arm64 in your repository and in order to configure omr-vps, I will have to build manually according to your instructions here: https://github.com/Ysurac/openmptcprouter-vps/wiki/Compile-kernel

how relevant is this script, is it suitable for Ubuntu 22.04?
https://github.com/rickytrevor/openmptcprouter-vps-install-arm64
from this issue https://github.com/Ysurac/openmptcprouter/issues/2707

Ysurac, 2024-01-28:
ARM64 is supported on Snapshot VPS but only for Debian (never tested on Ubuntu, not sure I will keep Ubuntu support...)
https://github.com/Ysurac/openmptcprouter/wiki/Snapshots

GermanAizek, 2024-01-28:
I have option to install Debian on server-board, thank you very much. I also respect Debian more myself.
But in mobile distributions, Mobian vs Ubuntu Touch is better in ease operation for user, and more devices are supported thanks to Hallium.

GermanAizek, 2024-01-29:
@Ysurac,
```sh
$ sudo UPSTREAM6="yes" sh debian-x86_64.sh 
Check user...
Check Linux version...
Check architecture...
Check about broken packages...
Check if OpenMPTCProuter VPS is already installed...
Update mode
Update repo.openmptcprouter.com key
Warning: apt-key is deprecated. Manage keyring files in trusted.gpg.d instead (see apt-key(8)).
Remove lock and update packages list...
Hit:1 http://deb.debian.org/debian bullseye InRelease
Hit:2 http://deb.debian.org/debian buster-backports InRelease
Hit:3 https://repo.openmptcprouter.com next InRelease
Hit:4 https://repo.openmptcprouter.com buster InRelease
Hit:5 http://mirrors.tuna.tsinghua.edu.cn/debian bookworm InRelease
Hit:6 http://mirrors.tuna.tsinghua.edu.cn/debian bookworm-updates InRelease
Hit:7 http://mirrors.tuna.tsinghua.edu.cn/debian bookworm-backports InRelease
Hit:8 http://mirrors.tuna.tsinghua.edu.cn/debian-security bookworm-security InRelease
Reading package lists...
Install apt-transport-https, gnupg and openssh-server...
Reading package lists...
Building dependency tree...
Reading state information...
apt-transport-https is already the newest version (2.6.1).
gnupg is already the newest version (2.2.40-1.1).
openssh-server is already the newest version (1:9.2p1-2+deb12u2).
The following packages were automatically installed and are no longer required:
  g++-10 gir1.2-ayatanaappindicator3-0.1 libavdevice58 libavfilter7
  libavformat58 libavresample4 libcbor0 libfftw3-double3 libflac8
  libfluidsynth2 libgdk-pixbuf-xlib-2.0-0 libgdk-pixbuf2.0-0 libglew2.1
  libglu1-mesa libgs9-common libgssdp-1.2-0 libgtksourceview-3.0-1
  libgtksourceview-3.0-common libgupnp-1.2-0 libidn11 libilmbase25 libllvm11
  libmms0 libnautilus-extension1a libntfs-3g883 libofa0 libopenexr25
  libopengl0 libpoppler102 libpostproc55 libprocps8 libpython3.9 libqpdf28
  librest-0.7-0 libsigsegv2 libsoup-gnome2.4-1 libsrt1.4-gnutls
  libstdc++-10-dev libswscale5 libwmf0.2-7 python3-certifi
  python3-charset-normalizer python3-ldb python3-pycurl python3-requests
  python3-talloc ruby2.7
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Add OpenMPTCProuter repo...
Explanation: Prefer OpenMPTCProuter provided packages over the Debian native ones
Package: *
Pin: origin repo.openmptcprouter.com
Pin-Priority: 1001
Explanation: Prefer libuv1 Debian native package
Package: libuv1
Pin: version *
Pin-Priority: 1003
Install mptcp kernel and shadowsocks...
Hit:1 http://deb.debian.org/debian bullseye InRelease
Hit:2 http://deb.debian.org/debian buster-backports InRelease
Hit:3 https://repo.openmptcprouter.com next InRelease
Hit:4 https://repo.openmptcprouter.com buster InRelease
Hit:5 http://mirrors.tuna.tsinghua.edu.cn/debian bookworm InRelease
Hit:6 http://mirrors.tuna.tsinghua.edu.cn/debian bookworm-updates InRelease
Hit:7 http://mirrors.tuna.tsinghua.edu.cn/debian bookworm-backports InRelease
Hit:8 http://mirrors.tuna.tsinghua.edu.cn/debian-security bookworm-security InRelease
Reading package lists...
Reading package lists...
Building dependency tree...
Reading state information...
dirmngr is already the newest version (2.2.40-1.1).
patch is already the newest version (2.7.6-7).
rename is already the newest version (2.01-1).
curl is already the newest version (7.88.1-10+deb12u5).
libcurl4 is already the newest version (7.88.1-10+deb12u5).
unzip is already the newest version (6.0-28).
[... 191 lines cut ...]
install -m 755 mptcpize /usr/bin/
install -d /usr/lib/mptcpize/
install -m 644 libmptcpwrap.so /usr/lib/mptcpize/
Reading package lists...
Building dependency tree...
Reading state information...
iproute2 is already the newest version (6.1.0-3).
The following packages were automatically installed and are no longer required:
  g++-10 gir1.2-ayatanaappindicator3-0.1 libavdevice58 libavfilter7
  libavformat58 libavresample4 libcbor0 libfftw3-double3 libflac8
  libfluidsynth2 libgdk-pixbuf-xlib-2.0-0 libgdk-pixbuf2.0-0 libglew2.1
  libglu1-mesa libgs9-common libgssdp-1.2-0 libgtksourceview-3.0-1
  libgtksourceview-3.0-common libgupnp-1.2-0 libidn11 libilmbase25 libllvm11
  libmms0 libnautilus-extension1a libntfs-3g883 libofa0 libopenexr25
  libopengl0 libpoppler102 libpostproc55 libprocps8 libpython3.9 libqpdf28
  librest-0.7-0 libsigsegv2 libsoup-gnome2.4-1 libsrt1.4-gnutls
  libstdc++-10-dev libswscale5 libwmf0.2-7 python3-certifi
  python3-charset-normalizer python3-ldb python3-pycurl python3-requests
  python3-talloc ruby2.7
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
MPTCPize iperf3...
mptcpize: can't rename /tmp/unit_FDX4hz to /lib/systemd/system/iperf3.service: Invalid cross-device link
```
how to solve this error?

Ysurac, 2024-01-29:
Should be fixed now.

GermanAizek, 2024-01-29:
> Should be fixed now.

same thing
```
MPTCPize iperf3...
mptcpize: can't rename /tmp/unit_Zeis8n to /lib/systemd/system/iperf3.service: Invalid cross-device link
```

Ysurac, 2024-01-30:
```rm debian-x86_64.sh``` and try again https://github.com/Ysurac/openmptcprouter/wiki/Snapshots
This part is removed for ARM64 so the error is not possible (I flushed CDN cache to be sure, maybe I forgot that...)

GermanAizek, 2024-01-31:
@Ysurac, remove sh works.
Again this error on next steps.
```
orangepi@orangepi3-lts:~$ sudo UPSTREAM6="yes" sh debian-x86_64.sh
[sudo] password for orangepi: 
Check user...
Check Linux version...
Check architecture...
Check about broken packages...
Check if OpenMPTCProuter VPS is already installed...
Update mode                                                                                                                                                                                                       
Update repo.openmptcprouter.com key                                                                                                                                                                               
Warning: apt-key is deprecated. Manage keyring files in trusted.gpg.d instead (see apt-key(8)).                                                                                                                   
--2024-01-30 23:57:53--  https://repo.openmptcprouter.com/openmptcprouter.gpg.key                                                                                                                                 
Resolving repo.openmptcprouter.com (repo.openmptcprouter.com)... 93.189.63.146                                                                                                                                    
Connecting to repo.openmptcprouter.com (repo.openmptcprouter.com)|93.189.63.146|:443... connected.                                                                                                                
HTTP request sent, awaiting response... 200 OK                                                                                                                                                                    
Length: 1241 (1.2K) [application/octet-stream]                                                                                                                                                                    
Saving to: '/etc/apt/trusted.gpg.d/openmptcprouter.gpg'                                                                                                                                                           
                                                                                                                                                                                                                  
/etc/apt/trusted.gpg.d/openmptcprouter.gpg           100%[====================================================================================================================>]   1.21K  --.-KB/s    in 0s       
                                                                                                                                                                                                                  
2024-01-30 23:57:54 (3.63 MB/s) - '/etc/apt/trusted.gpg.d/openmptcprouter.gpg' saved [1241/1241]

Remove lock and update packages list...
Hit:1 http://deb.debian.org/debian bullseye InRelease
Get:2 http://deb.debian.org/debian buster-backports InRelease [51.4 kB]
Hit:3 https://repo.openmptcprouter.com next InRelease
Hit:4 https://repo.openmptcprouter.com buster InRelease
Hit:5 http://mirrors.tuna.tsinghua.edu.cn/debian bookworm InRelease
Get:6 http://mirrors.tuna.tsinghua.edu.cn/debian bookworm-updates InRelease [52.1 kB]
Get:7 http://mirrors.tuna.tsinghua.edu.cn/debian bookworm-backports InRelease [56.5 kB]
Get:8 http://mirrors.tuna.tsinghua.edu.cn/debian-security bookworm-security InRelease [48.0 kB]
Get:9 http://mirrors.tuna.tsinghua.edu.cn/debian-security bookworm-security/main arm64 Packages [132 kB]                                                                                                         
Get:10 http://mirrors.tuna.tsinghua.edu.cn/debian-security bookworm-security/main armhf Packages [129 kB]                                                                                                        
Fetched 469 kB in 8s (60.6 kB/s)                                                                                                                                                                                 
Reading package lists... Done
N: Repository 'Debian bookworm' changed its 'non-free component' value from 'non-free' to 'non-free non-free-firmware'
N: More information about this can be found online in the Release notes at: https://www.debian.org/releases/bookworm/arm64/release-notes/ch-information.html#non-free-split
Install apt-transport-https, gnupg and openssh-server...
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
apt-transport-https is already the newest version (2.6.1).
gnupg is already the newest version (2.2.40-1.1).
openssh-server is already the newest version (1:9.2p1-2+deb12u2).
The following packages were automatically installed and are no longer required:
  g++-10 gir1.2-ayatanaappindicator3-0.1 libavdevice58 libavfilter7 libavformat58 libavresample4 libcbor0 libfftw3-double3 libflac8 libfluidsynth2 libgdk-pixbuf-xlib-2.0-0 libgdk-pixbuf2.0-0 libglew2.1
  libglu1-mesa libgs9-common libgssdp-1.2-0 libgtksourceview-3.0-1 libgtksourceview-3.0-common libgupnp-1.2-0 libidn11 libilmbase25 libllvm11 libmms0 libnautilus-extension1a libntfs-3g883 libofa0 libopenexr25
[... cut ...] ..+.+........+......+....+..+...............+....+......+...+..+...+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
-----
Created symlink /etc/systemd/system/multi-user.target.wants/omr-admin-ipv6.service -> /lib/systemd/system/omr-admin-ipv6.service.
Created symlink /etc/systemd/system/multi-user.target.wants/omr-admin.service -> /lib/systemd/system/omr-admin.service.
mptcpize: can't rename /tmp/unit_EhWwVU to /lib/systemd/system/omr-admin.service: Invalid cross-device link
```

GermanAizek, 2024-01-31:
@Ysurac, how to fixes prev error? It is equals previous.

Ysurac, 2024-01-31:
mptcpize is required, I tested on some VPS using ARM64 and the current script is working.
What is the VPS provider ?
There is some customization on the installation ? What is the current kernel (```uname -a```) ?

GermanAizek, 2024-01-31:
@Ysurac, it local deploy on board Orange Pi 3, Allwinner H6, not LTSversion with pci express port.

Ysurac, 2024-01-31:
I think I found a solution, so I will patch mptcpize.

Ysurac, 2024-01-31:
Can you try again ?

GermanAizek, 2024-01-31:
> Can you try again ?

Ok, wait pls.

GermanAizek, 2024-01-31:
> Can you try again ?

Done, but before 'SSH port:' label text is there strange symbol '?m'

[image]

GermanAizek, 2024-01-31:
After installation kernel postfix dont changed and /etc/default/grub.cfg not found.
[image]

GermanAizek, 2024-02-01:
@Ysurac, how to create grub.cfg? where is I can grab examples?

Ysurac, 2024-02-01:
What is used to boot the kernel ? What do you have in ```/boot/``` ?

GermanAizek, 2024-02-01:
> What is used to boot the kernel ? What do you have in `/boot/` ?

@Ysurac,
```
orangepi@orangepi3-lts:~$ ls /boot/                                                                                                                                                                               
boot.bmp  boot.scr                 dtb                   Image                        orangepiEnv.txt                  System.map-5.16.17-sun50iw6  uInitrd-5.16.17-sun50iw6                                      
boot.cmd  config-5.16.17-sun50iw6  dtb-5.16.17-sun50iw6  initrd.img-5.16.17-sun50iw6  orangepi_first_run.txt.template  uInitrd                      vmlinuz-5.16.17-sun50iw6
```

Ysurac, 2024-02-02:
It's not using grub, but another bootloader. I can't help here.
It's a Debian or using Armbian distribution ? In any case, try to update the full system.

GermanAizek, 2024-02-02:
> It's not using grub, but another bootloader. I can't help here. It's a Debian or using Armbian distribution ? In any case, try to update the full system.

It's Debian 11
http://www.orangepi.org/html/hardWare/computerAndMicrocontrollers/service-and-support/Orange-pi-3-LTS.html#downloads
https://drive.google.com/drive/folders/1ctuKgHNN9r517tiAv9GGGaR7UYQgZiXP

closed 2024-05-08 by github-actions[bot]