Ysurac/openmptcprouter #3138: some ERRO log on latest snap 6.1 kernel

fareign, 2024-01-28:
Jan 28 08:10:23 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S99shadowsocks-libev: /tmp/ssrules.nft:38:3-3: Error: syntax error, unexpected -, expecting comma or '}'
Jan 28 08:10:23 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S99shadowsocks-libev: 		-4,
Jan 28 08:10:23 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S99shadowsocks-libev: 		^
Jan 28 08:10:23 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S99shadowsocks-libev: In file included from /tmp/ssrules.nft.chk:1:17-43:
Jan 28 08:10:23 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S99shadowsocks-libev: /tmp/ssrules.nft:39:3-3: Error: syntax error, unexpected '['
Jan 28 08:10:23 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S99shadowsocks-libev: 		[-t,
Jan 28 08:10:23 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S99shadowsocks-libev: 		^
Jan 28 08:10:23 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S99shadowsocks-libev: In file included from /tmp/ssrules.nft.chk:1:17-43:
Jan 28 08:10:23 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S99shadowsocks-libev: /tmp/ssrules.nft:40:10-10: Error: syntax error, unexpected ']', expecting string
Jan 28 08:10:23 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S99shadowsocks-libev: 		timeout],
Jan 28 08:10:23 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S99shadowsocks-libev: 		       ^
Jan 28 08:10:23 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S99shadowsocks-libev: In file included from /tmp/ssrules.nft.chk:1:17-43:
Jan 28 08:10:23 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S99shadowsocks-libev: /tmp/ssrules.nft:41:3-10: Error: syntax error, unexpected string
Jan 28 08:10:23 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S99shadowsocks-libev: 		hostname,
Jan 28 08:10:23 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S99shadowsocks-libev: 		^^^^^^^^
Jan 28 08:10:23 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S99shadowsocks-libev: In file included from /tmp/ssrules.nft.chk:1:17-43:
Jan 28 08:10:23 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S99shadowsocks-libev: /tmp/ssrules.nft:42:3-11: Error: syntax error, unexpected string
Jan 28 08:10:23 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S99shadowsocks-libev: 		resolveip,
Jan 28 08:10:23 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S99shadowsocks-libev: 		^^^^^^^^^
Jan 28 08:10:23 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S99shadowsocks-libev: In file included from /tmp/ssrules.nft.chk:1:17-43:
Jan 28 08:10:23 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S99shadowsocks-libev: /tmp/ssrules.nft:43:3-3: Error: syntax error, unexpected -
Jan 28 08:10:23 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S99shadowsocks-libev: 		-6,
Jan 28 08:10:23 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S99shadowsocks-libev: 		^
Jan 28 08:10:23 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S99shadowsocks-libev: In file included from /tmp/ssrules.nft.chk:1:17-43:
Jan 28 08:10:23 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S99shadowsocks-libev: /tmp/ssrules.nft:44:3-3: Error: syntax error, unexpected '['
Jan 28 08:10:23 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S99shadowsocks-libev: 		[-t,
Jan 28 08:10:23 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S99shadowsocks-libev: 		^
Jan 28 08:10:23 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S99shadowsocks-libev: In file included from /tmp/ssrules.nft.chk:1:17-43:
Jan 28 08:10:23 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S99shadowsocks-libev: /tmp/ssrules.nft:45:10-10: Error: syntax error, unexpected ']', expecting string
Jan 28 08:10:23 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S99shadowsocks-libev: 		timeout],
Jan 28 08:10:23 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S99shadowsocks-libev: 		       ^
Jan 28 08:10:23 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S99shadowsocks-libev: In file included from /tmp/ssrules.nft.chk:1:17-43:
Jan 28 08:10:23 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S99shadowsocks-libev: /tmp/ssrules.nft:46:3-10: Error: syntax error, unexpected string
Jan 28 08:10:23 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S99shadowsocks-libev: 		hostname,
Jan 28 08:10:23 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S99shadowsocks-libev: 		^^^^^^^^
Jan 28 08:10:23 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S99shadowsocks-libev: In file included from /tmp/ssrules.nft.chk:1:17-43:
Jan 28 08:10:23 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S99shadowsocks-libev: /tmp/ssrules.nft:47:3-11: Error: syntax error, unexpected string
Jan 28 08:10:23 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S99shadowsocks-libev: 		resolveip,
Jan 28 08:10:23 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S99shadowsocks-libev: 		^^^^^^^^^
Jan 28 08:10:25 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Jan 28 08:10:25 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S99shadowsocks-rust: Include '/etc/firewall.v2ray-rules' failed with exit code 1
Jan 28 08:10:25 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S99shadowsocks-rust: Include '/etc/firewall.xray-rules' failed with exit code 1
Jan 28 08:10:32 OpenMPTCProuter daemon.info omr-tracker-ss: Shadowsocks sss0 is down (can't contact via http 49.7.115.41, baidu.com, www.quark.cn, cloud.huawei.com)
Jan 28 08:10:33 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Jan 28 08:10:51 OpenMPTCProuter daemon.info omr-tracker-ss: Server sss0 (47.107.32.106) seems down, no answer to ping
Jan 28 08:11:00 OpenMPTCProuter user.notice omr-schedule-010-services: Can't find omr-bypass rules, restart omr-bypass...
Jan 28 08:11:00 OpenMPTCProuter user.notice omr-bypass: Starting OMR-ByPass...
Jan 28 08:11:00 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Tunnel up : Replace default route by 10.255.252.5 dev tun0
Jan 28 08:11:03 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_all_4 Error: No such file or directory
Jan 28 08:11:03 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_all_4 Error: No such file or directory
Jan 28 08:11:03 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_all_4 Error: No such file or directory
Jan 28 08:11:03 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_all_4 Error: No such file or directory

## Specifications

  - OpenMPTCProuter version: latest snap 6.1 kernel
  - OpenMPTCProuter VPS version: 0.1029test
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform: x86
  - Country:

fareign, 2024-01-28:
Jan 28 08:54:38 OpenMPTCProuter daemon.err omr-tracker[6233]: sh: 33.3333: bad number

Ysurac, 2024-01-28:
I would need the result of ```uci show shadowsocks-libev``` and the file ```/tmp/ssrules.nft``` (you may have to comment with  # before the "rm -f "$tmp.nft" in /etc/init.d/shadowsocks-libev to keep it..
The others shadowsocks-libev and firewall errors are related to this problem.
Can you test with ShadowSocks-Rust to check if you have same problem ?

For omr-tracker, I will check where this problem can be.

fareign, 2024-01-28:
root@OpenMPTCProuter:/tmp# cat ssrules.nft

set ss_rules_src_bypass {
        type ipv4_addr;
        flags interval;
        auto-merge;
}
set ss_rules6_src_bypass {
        type ipv6_addr;
        flags interval;
        auto-merge;
}
set ss_rules_src_forward {
        type ipv4_addr;
        flags interval;
        auto-merge;
}
set ss_rules6_src_forward {
        type ipv6_addr;
        flags interval;
        auto-merge;
}
set ss_rules_src_checkdst {
        type ipv4_addr;
        flags interval;
        auto-merge;
}
set ss_rules6_src_checkdst {
        type ipv6_addr;
        flags interval;
        auto-merge;
}
set ss_rules_remote_servers {
        type ipv4_addr;
        flags interval;
        auto-merge;
        elements = {
                47.107.32.106
        }
}
set ss_rules6_remote_servers {
        type ipv6_addr;
        flags interval;
        auto-merge;
}
set ss_rules_dst_bypass {
        type ipv4_addr;
        flags interval;
        auto-merge;
}
set ss_rules6_dst_bypass {
        type ipv6_addr;
        flags interval;
        auto-merge;
}
set ss_rules_dst_bypass_ {
        type ipv4_addr;
        flags interval;
        auto-merge;
        elements = {
                0.0.0.0/8,
                10.0.0.0/8,
                100.64.0.0/10,
                127.0.0.0/8,
                169.254.0.0/16,
                172.16.0.0/12,
                192.0.0.0/24,
                192.0.2.0/24,
                192.168.0.0/16,
                192.31.196.0/24,
                192.52.193.0/24,
                192.88.99.0/24,
                192.175.48.0/24,
                198.18.0.0/15,
                198.51.100.0/24,
                203.0.113.0/24,
                224.0.0.0/4,
                240.0.0.0/4
        }
}
set ss_rules6_dst_bypass_ {
        type ipv6_addr;
        flags interval;
        auto-merge;
        elements = {
                ::1/128,
                ::/128,
                ::ffff:0:0/96,
                64:ff9b:1::/48,
                100::/64,
                fe80::/10,
                2001::/23,
                fc00::/7
        }
}
set ss_rules_dst_forward {
        type ipv4_addr;
        flags interval;
        auto-merge;
}
set ss_rules6_dst_forward {
        type ipv6_addr;
        flags interval;
        auto-merge;
}
set ss_rules_dst_forward_rrst_ {
        type ipv4_addr;
        flags interval;
        auto-merge;
}
set ss_rules6_dst_forward_rrst_ {
        type ipv6_addr;
        flags interval;
        auto-merge;
}

chain ss_rules_pre_tcp {
        type nat hook prerouting priority -1;
        meta l4proto tcp goto ss_rules_pre_src_tcp;
}

chain ss_rules_pre_src_tcp {
        ip daddr @ss_rules_dst_bypass_ accept;
        ip6 daddr @ss_rules6_dst_bypass_ accept;
        goto ss_rules_src_tcp;
}

chain ss_rules_src_tcp {
        ip saddr @ss_rules_src_bypass accept;
        ip saddr @ss_rules_src_forward goto ss_rules_forward_tcp;
        ip saddr @ss_rules_src_checkdst goto ss_rules_dst_tcp;
        ip6 saddr @ss_rules6_src_bypass accept;
        ip6 saddr @ss_rules6_src_forward goto ss_rules_forward_tcp;
        ip6 saddr @ss_rules6_src_checkdst goto ss_rules_dst_tcp;
        goto ss_rules_forward_tcp;
}

chain ss_rules_dst_tcp {
        ip daddr @ss_rules_dst_bypass accept;
        ip daddr @ss_rules_remote_servers accept;
        ip daddr @ss_rules_dst_forward goto ss_rules_forward_tcp;
        ip6 daddr @ss_rules6_dst_bypass accept;
        ip6 daddr @ss_rules6_remote_servers accept;
        ip6 daddr @ss_rules6_dst_forward goto ss_rules_forward_tcp;
        goto ss_rules_forward_tcp;
}

chain ss_rules_forward_tcp {
        meta l4proto tcp  redirect to :1100-1101;
}
chain ss_rules_local_out {
        type nat hook output priority -1;
        meta l4proto != tcp accept;
        ip daddr @ss_rules_remote_servers accept;
        ip daddr @ss_rules_dst_bypass_ accept;
        ip daddr @ss_rules_dst_bypass accept;
        ip6 daddr @ss_rules6_remote_servers accept;
        ip6 daddr @ss_rules6_dst_bypass_ accept;
        ip6 daddr @ss_rules6_dst_bypass accept;
        goto ss_rules_forward_tcp;
}

fareign, 2024-01-28:
root@OpenMPTCProuter:/tmp# uci show shadowsocks-libev
shadowsocks-libev.hi1=ss_redir
shadowsocks-libev.hi1.server='sss0'
shadowsocks-libev.hi1.local_address='::'
shadowsocks-libev.hi1.local_port='1100'
shadowsocks-libev.hi1.mode='tcp_and_udp'
shadowsocks-libev.hi1.timeout='1000'
shadowsocks-libev.hi1.fast_open='1'
shadowsocks-libev.hi1.verbose='0'
shadowsocks-libev.hi1.syslog='1'
shadowsocks-libev.hi1.reuse_port='1'
shadowsocks-libev.hi1.mptcp='1'
shadowsocks-libev.hi1.ipv6_first='1'
shadowsocks-libev.hi1.no_delay='1'
shadowsocks-libev.hi1.ebpf='0'
shadowsocks-libev.hi2=ss_redir
shadowsocks-libev.hi2.server='sss0'
shadowsocks-libev.hi2.local_address='::'
shadowsocks-libev.hi2.local_port='1101'
shadowsocks-libev.hi2.mode='tcp_and_udp'
shadowsocks-libev.hi2.timeout='1000'
shadowsocks-libev.hi2.fast_open='1'
shadowsocks-libev.hi2.verbose='0'
shadowsocks-libev.hi2.syslog='1'
shadowsocks-libev.hi2.reuse_port='1'
shadowsocks-libev.hi2.mptcp='1'
shadowsocks-libev.hi2.ipv6_first='1'
shadowsocks-libev.hi2.no_delay='1'
shadowsocks-libev.hi2.ebpf='0'
shadowsocks-libev.ss_rules=ss_rules
shadowsocks-libev.ss_rules.disabled='0'
shadowsocks-libev.ss_rules.redir_tcp='all'
shadowsocks-libev.ss_rules.src_default='forward'
shadowsocks-libev.ss_rules.dst_default='forward'
shadowsocks-libev.ss_rules.local_default='forward'
shadowsocks-libev.ss_rules.server='sss0'
shadowsocks-libev.sss0=server
shadowsocks-libev.sss0.disabled='0'
shadowsocks-libev.sss0.server='47.***.***.106'
shadowsocks-libev.sss0.server_port='65101'
shadowsocks-libev.sss0.method='aes-256-gcm'
shadowsocks-libev.sss0.obfs='0'
shadowsocks-libev.sss0.key='EuF/imymm2rE/I='
shadowsocks-libev.sss0.obfs_host='8.218.27.117'
shadowsocks-libev.sss0.obfs_plugin='obfs'
shadowsocks-libev.sss0.obfs_type='http'
shadowsocks-libev.dns=ss_tunnel
shadowsocks-libev.dns.disabled='1'
shadowsocks-libev.dns.mode='tcp_and_udp'
shadowsocks-libev.dns.server='sss0'
shadowsocks-libev.dns.local_port='5353'
shadowsocks-libev.dns.tunnel_address='8.8.8.8:53'
shadowsocks-libev.hi3=ss_redir
shadowsocks-libev.hi3.server='sss1'
shadowsocks-libev.hi3.local_address='::'
shadowsocks-libev.hi3.local_port='1101'
shadowsocks-libev.hi3.mode='tcp_and_udp'
shadowsocks-libev.hi3.timeout='1000'
shadowsocks-libev.hi3.fast_open='1'
shadowsocks-libev.hi3.reuse_port='1'
shadowsocks-libev.hi3.syslog='1'
shadowsocks-libev.hi3.mptcp='1'
shadowsocks-libev.hi3.no_delay='1'
shadowsocks-libev.hi3.verbose='0'
shadowsocks-libev.hi3.ebpf='0'
shadowsocks-libev.hi4=ss_redir
shadowsocks-libev.hi4.server='sss1'
shadowsocks-libev.hi4.local_address='::'
shadowsocks-libev.hi4.local_port='1101'
shadowsocks-libev.hi4.mode='tcp_and_udp'
shadowsocks-libev.hi4.timeout='1000'
shadowsocks-libev.hi4.fast_open='1'
shadowsocks-libev.hi4.reuse_port='1'
shadowsocks-libev.hi4.syslog='1'
shadowsocks-libev.hi4.mptcp='1'
shadowsocks-libev.hi4.no_delay='1'
shadowsocks-libev.hi4.verbose='0'
shadowsocks-libev.hi4.ebpf='0'
shadowsocks-libev.sss1=server
shadowsocks-libev.sss1.server_port='65101'
shadowsocks-libev.sss1.method='aes-256-gcm'
shadowsocks-libev.sss1.obfs='0'
shadowsocks-libev.sss1.disabled='1'
shadowsocks-libev.sss1.key='EuF/ifEFm2rE/I='
shadowsocks-libev.sss1.obfs_host='8.218.27.117'
shadowsocks-libev.sss1.server='112.***.39.**9'
shadowsocks-libev.sss1.obfs_plugin='obfs'
shadowsocks-libev.sss1.obfs_type='http'
shadowsocks-libev.tracker_sss0=ss_local
shadowsocks-libev.tracker_sss0.server='sss0'
shadowsocks-libev.tracker_sss0.local_address='127.0.0.1'
shadowsocks-libev.tracker_sss0.local_port='1111'
shadowsocks-libev.tracker_sss0.mode='tcp_and_udp'
shadowsocks-libev.tracker_sss0.timeout='600'
shadowsocks-libev.tracker_sss0.fast_open='1'
shadowsocks-libev.tracker_sss0.syslog='0'
shadowsocks-libev.tracker_sss0.reuse_port='1'
shadowsocks-libev.tracker_sss0.mptcp='1'
shadowsocks-libev.tracker_sss0.verbose='0'

fareign, 2024-01-28:
this problem cause on startup.not working.

Ysurac, 2024-01-30:
I'm not able to reproduce these errors, can you try again with latest snapshot ?
Do you import a previous config ? Do you use snapshot image or custom image ?

closed 2024-05-04 by github-actions[bot]

hhaniel, 2024-05-14:
> Jan 28 08:54:38 OpenMPTCProuter daemon.err omr-tracker[6233]: sh: 33.3333: bad number

Just wondering what this error means - I get this coming up in the log files relatively often but not tracked down what it is yet.

Ysurac, 2024-05-14:
Yes I know, I don't exactly know what is putting this for now...