Ysurac/openmptcprouter #3139: What is the right way to setup wan interfaces ipv4 and ipv6 capable at the same time  via dhcp and dhcpv6

rdmitry0911, 2024-01-28:
## Expected Behavior

Setting up wan1 (eth1) and wan2 (eth2)  protocol to dhcp and wan3 (eth1 or @wan1) and wan4 (eth2 or @wan2) protocol to dhcpv6 should produce working configuration
## Current Behavior

ipv4 works fine, however ipv6 has wrong configuration. I can ping ipv6 addresses via interfaces:
```
root@OpenMPTCProuter:~# ping6 -I eth1 google.com
PING google.com (2a00:1450:4001:812::200e): 56 data bytes
64 bytes from 2a00:1450:4001:812::200e: seq=0 ttl=116 time=34.970 ms
64 bytes from 2a00:1450:4001:812::200e: seq=1 ttl=116 time=34.779 ms
64 bytes from 2a00:1450:4001:812::200e: seq=2 ttl=116 time=34.689 ms
^C
--- google.com ping statistics ---
3 packets transmitted, 3 packets received, 0% packet loss
round-trip min/avg/max = 34.689/34.812/34.970 ms
root@OpenMPTCProuter:~# 
root@OpenMPTCProuter:~# ping6 -I eth2 google.com
PING google.com (2a00:1450:4001:812::200e): 56 data bytes
64 bytes from 2a00:1450:4001:812::200e: seq=0 ttl=64 time=65.513 ms
64 bytes from 2a00:1450:4001:812::200e: seq=1 ttl=64 time=63.421 ms
64 bytes from 2a00:1450:4001:812::200e: seq=2 ttl=64 time=52.708 ms
64 bytes from 2a00:1450:4001:812::200e: seq=3 ttl=64 time=59.217 ms
^C
--- google.com ping statistics ---
4 packets transmitted, 4 packets received, 0% packet loss
round-trip min/avg/max = 52.708/60.214/65.513 ms
```
however I can't ping ipv6 address via default route. Probably, because a strange default route via lan interface occurs in routing table:
```
root@OpenMPTCProuter:~# ip -6 ro sh|grep default
default from fe80::a00:2 dev 6in4-omr6in4 proto static metric 1201 pref medium
default via fd3c:9f67:d5c5:: dev eth0 metric 66 pref medium
default via 2a02:13XX:XXX2:67fd:: dev eth2 metric 68 pref medium
default via fe80::7624:9fff:feb7:80cb dev eth1 metric 69 pref medium
default via fe80::e63a:65ff:fe44:e4de dev eth2 metric 610 pref medium
default via fe80::a00:1 dev 6in4-omr6in4 proto static metric 1201 pref medium
```
after removing this default route:
```
ip -6 route del default via fd3c:9f67:d5c5:: dev eth0
```
tunnel to vps starts and the system looks fine. However, after every network restart this strange default route again appears in  routing table. 

The status picture looks like this:

[image]

## Specifications

  - OpenMPTCProuter version:  [openmptcprouter v0.60beta1-6.1 r0+24843-acf40c022e]
  - OpenMPTCProuter VPS version: 0.1029-test 6.1.0-17-amd64 
  - OpenMPTCProuter VPS provider: Hetzner
  - OpenMPTCProuter platform: x86_64

Ysurac, 2024-01-30:
Should be fixed in latest commit.
I'm still working on mixing IPv4 and IPv6 support.

rdmitry0911, 2024-02-17:
I've just installed beta 2 still doesn't work ipv6 alongside with ipv4. Something has changed in ipv6 configuration in beta 2 ?

Ysurac, 2024-02-17:
You should be able to connect to an IPv6 and IPv4 server and use both without issues. For now IPv6 links can't be aggregated with IPv4 links.
What is your problem ?

rdmitry0911, 2024-02-17:
The problem is that I can't configure omr to connect to vps via ipv6. I'm trying to configure it from the ground up. I have 2 wan interfaces wan1 and wan2 based on eth1 and eth2 connected to a modems with ipv6 capability. When I setup both interfaces in luci settings wizzard I can't do any of 
```
ping6 google.com 
ping6 2a00:1450:4001:810::200e 
ping6 -I eth1 google.com
ping6 -I eth2 google.com

```
However these commands work
```
ping6 -I eth1 2a00:1450:4001:810::200e
ping6 -I eth2 2a00:1450:4001:810::200e
```
Routing table looks like this:
root@OpenMPTCProuter:~# route -A inet6 -n
```
Kernel IPv6 routing table
Destination                                 Next Hop                                Flags Metric Ref    Use Iface
::/0                                        ::                                      !n    -1     4        0 lo      
::/0                                        ::                                      !n    -1     4        0 lo      
::/0                                        ::                                      !n    -1     4        0 lo      
::/0                                        ::                                      !n    -1     4        0 lo      
::/0                                        ::                                      !n    -1     4        0 lo      
::/0                                        ::                                      !n    -1     4        0 lo      
fd13:d888:b31e::/60                         ::                                      U     1024   1        0 eth0    
::/0                                        fd13:d888:b31e::                        UG    1024   2        0 eth0    
::/0                                        ::                                      !n    -1     4        0 lo      
2a0d:3344:1a87:cf10::b44/128                ::                                      U     1024   1        0 eth1    
fd6f:7e1:2556:10::/64                       ::                                      U     1024   1        0 eth1    
::/0                                        fe80::7624:9fff:feb7:80cb               UG    1024   1        0 eth1    
2a02:1388:215e:a239::ca0/128                ::                                      U     1024   1        0 eth2    
fdd1:4d09:c44d::ca0/128                     ::                                      U     1024   1        0 eth2    
::/0                                        fe80::e63a:65ff:fe44:e4de               UG    1024   1        0 eth2    
::/0                                        ::                                      !n    -1     4        0 lo      
::/0                                        ::                                      !n    -1     4        0 lo      
::/0                                        ::                                      U     1201   2        0 6in4-omr6in4
2a02:1388:215e:a239::ca0/128                ::                                      U     610003 1        0 eth2    
2a02:1388:215e:a239::/64                    ::                                      U     256    1        0 eth2    
2a0d:3344:1a87:cf00::/56                    fe80::7624:9fff:feb7:80cb               UG    512    1        0 eth1    
2a0d:3344:1a87:cf00::/56                    fe80::7624:9fff:feb7:80cb               UG    512    1        0 eth1    
2a0d:3344:1a87:cf00::/56                    fe80::7624:9fff:feb7:80cb               UG    512    3        0 eth1    
2a0d:3344:1a87:cf00::/56                    fe80::7624:9fff:feb7:80cb               UG    512    1        0 eth1    
2a0d:3344:1a87:cf10::b44/128                ::                                      U     610002 1        0 eth1    
2a0d:3344:1a87:cf10::/64                    ::                                      U     256    1        0 eth1    
2a0d:3344:1a87:cf10::/64                    ::                                      !n    2147483647 1        0 lo      
fd13:d888:b31e::/64                         ::                                      U     9999   1        0 eth0    
fd13:d888:b31e::/60                         ::                                      U     69999  1        0 eth0    
fd13:d888:b31e::/48                         ::                                      !n    2147483647 4        0 lo      
fd6f:7e1:2556::/48                          fe80::7624:9fff:feb7:80cb               UG    512    1        0 eth1    
fd6f:7e1:2556::/48                          fe80::7624:9fff:feb7:80cb               UG    512    1        0 eth1    
fd6f:7e1:2556::/48                          fe80::7624:9fff:feb7:80cb               UG    512    2        0 eth1    
fd6f:7e1:2556::/48                          fe80::7624:9fff:feb7:80cb               UG    512    1        0 eth1    
fd6f:7e1:2556:10::/64                       ::                                      U     256    1        0 eth1    
fd6f:7e1:2556:10::/64                       ::                                      U     610002 1        0 eth1    
fd6f:7e1:2556:10::/64                       ::                                      !n    2147483647 1        0 lo      
fdd1:4d09:c44d::ca0/128                     ::                                      U     610003 1        0 eth2    
fdd1:4d09:c44d::/64                         ::                                      U     256    1        0 eth2    
fe80::a00:2/128                             ::                                      U     1201   1        0 6in4-omr6in4
[... cut ...]
2a0d:3344:1a87:cf10::/128                   ::                                      Un    0      3        0 eth1    
[... 187 lines cut ...]
	option lookup '1500'

config rule 'all_fw_rule'
	option priority '1'
	option mark '0x4539'
	option lookup '991337'

config rule6 'all_fw_rule6'
	option priority '1'
	option mark '0x6539'
	option lookup '6991337'

config device 'lan_dev'
	option name 'eth0'

config rule '6in4_omr6in4_fw_rule'
	option priority '1'
	option mark '0x45391201'
	option lookup '1201'

config rule6 '6in4_omr6in4_fw_rule6'
	option priority '1'
	option mark '0x65391201'
	option lookup '1201'
```

Ysurac, 2024-02-17:
Did you set both IPv4 and IPv6 address for the VPS ?
Can you give me the output of ```ip -6 r``` (more modern and easier to read than ```route``` command) ?
I think that the 6in4 is here, I will fix that. For now you can do a ```rm -f /etc/init.d/omr-6in4```.

rdmitry0911, 2024-02-18:
vps has ipv6 address only. 
here is ip -6 r result after removing /etc/init.d/omr-6in4:
```
default from fe80::a00:2 dev 6in4-omr6in4 proto static metric 1201 pref medium
2a02:1388:215e:a239::ca0 dev eth2 metric 610003 pref medium
2a02:1388:215e:a239::/64 dev eth2 proto static metric 256 pref medium
2a0d:3344:1a87:cf00::/56 from 2a0d:3344:1a87:cf10::b44 via fe80::7624:9fff:feb7:80cb dev eth1 proto static metric 512 pref medium
2a0d:3344:1a87:cf00::/56 from 2a0d:3344:1a87:cf10::/64 via fe80::7624:9fff:feb7:80cb dev eth1 proto static metric 512 pref medium
2a0d:3344:1a87:cf00::/56 from fd6f:7e1:2556:10::b44 via fe80::7624:9fff:feb7:80cb dev eth1 proto static metric 512 pref medium
2a0d:3344:1a87:cf00::/56 from fd6f:7e1:2556:10::/64 via fe80::7624:9fff:feb7:80cb dev eth1 proto static metric 512 pref medium
2a0d:3344:1a87:cf10::b44 dev eth1 metric 610002 pref medium
2a0d:3344:1a87:cf10::/64 dev eth1 proto static metric 256 pref medium
unreachable 2a0d:3344:1a87:cf10::/64 dev lo proto static metric 2147483647 pref medium
fd13:d888:b31e::/64 dev eth0 proto static metric 9999 pref medium
fd13:d888:b31e::/60 dev eth0 metric 69999 pref medium
unreachable fd13:d888:b31e::/48 dev lo proto static metric 2147483647 pref medium
fd6f:7e1:2556::/48 from 2a0d:3344:1a87:cf10::b44 via fe80::7624:9fff:feb7:80cb dev eth1 proto static metric 512 pref medium
fd6f:7e1:2556::/48 from 2a0d:3344:1a87:cf10::/64 via fe80::7624:9fff:feb7:80cb dev eth1 proto static metric 512 pref medium
fd6f:7e1:2556::/48 from fd6f:7e1:2556:10::b44 via fe80::7624:9fff:feb7:80cb dev eth1 proto static metric 512 pref medium
fd6f:7e1:2556::/48 from fd6f:7e1:2556:10::/64 via fe80::7624:9fff:feb7:80cb dev eth1 proto static metric 512 pref medium
fd6f:7e1:2556:10::/64 dev eth1 proto static metric 256 pref medium
fd6f:7e1:2556:10::/64 dev eth1 metric 610002 pref medium
unreachable fd6f:7e1:2556:10::/64 dev lo proto static metric 2147483647 pref medium
fdd1:4d09:c44d::ca0 dev eth2 metric 610003 pref medium
fdd1:4d09:c44d::/64 dev eth2 proto static metric 256 pref medium
fe80::a00:2 dev 6in4-omr6in4 proto static metric 1201 pref medium
fe80::/64 dev eth0 proto kernel metric 256 pref medium
fe80::/64 dev 6in4-omr6in4 proto kernel metric 256 pref medium
fe80::/64 dev eth2 proto kernel metric 256 pref medium
fe80::/64 dev eth1 proto kernel metric 256 pref medium
default via fe80::a00:1 dev 6in4-omr6in4 proto static metric 1201 pref medium
default via fd13:d888:b31e:: dev eth0 metric 69999 pref medium
default via fe80::7624:9fff:feb7:80cb dev eth1 metric 610002 pref medium
default via fe80::e63a:65ff:fe44:e4de dev eth2 metric 610003 pref medium
```
I can now ping anything by ipv6 address, but ipv4 and dns still doesn't work:
```
root@OpenMPTCProuter:~# nslookup google.com
;; connection timed out; no servers could be reached
```
```
root@OpenMPTCProuter:~# cat /tmp/resolv.conf.d/resolv.conf.auto 
# Interface wan1
# Interface wan2
```

Ysurac, 2024-02-18:
If VPS have IPv6 only, IPv4 can't work if there is no DNS64 and NAT64 available on the provider network.
For DNS, check in Services->Recursive DNS, "Unbound" and "Ressource" tab, that recursive protocol is not set to "IP4 only".

rdmitry0911, 2024-02-18:
How can I check if DNS64 and NAT64 are available? From vps I can resolve domain names and vps ipv6 address is accessible from the internet.

recursive protocol is set to IP4 and IP6

Ysurac, 2024-02-19:
Ask your VPS provider, but I don't think Hetzner provide NAT64 for now.

rdmitry0911, 2024-02-20:
> IPv4 can't work if there is no DNS64 and NAT64 available on the provider network.

I don't quite understand the logic of the need of DNS64 and NAT64. At the moment I can ping6 the vps ipv6 address from omr so the tunnel can be eatablished. From VPS I have access to both ipv4 and ipv6 sites in the internet as well as dns for both ipv4 and ipv6. Why in this case I can't use the tunnel for non-tcp ipv4 traffic?

Ysurac, 2024-02-20:
You said "VPS has ipv6 address only", in this case VPS doesn't have access to IPv4.
If you have both IPv4 and IPv6 on the VPS, you can aggregate IPv4 links and IPv6 links separately.

rdmitry0911, 2024-02-20:
I can access vps from outside only via ipv6, however vps itself has both ipv4 and ipv6 addresses. vps ipv4 address is just not accessible from outside and can't be used for tunneling therefore. What configuration of omr in this case will be more suitable to provide both ipv4 and ipv6 aggregation?

closed 2024-05-25 by github-actions[bot]