Ysurac/openmptcprouter #3141: Adding new target alias names for `aarch64_generic`

GermanAizek, 2024-01-29:
@Ysurac,
```
$ OMR_TARGET="arm64" OMR_FEED_SRC="master" OMR_KERNEL="5.4" ./build.sh
```

```
Target arm64 not found ! You have to configure and compile your kernel manually.
```

Maybe we should add alias target names aarch64_generic?
Example:
`arm64`, `aarch64` and `aarch64_generic`

according to code build.sh, it is necessary to set incomprehensible unknown `r2s` `r4s` for building.

[image]

Ysurac, 2024-01-29:
Generic ARM64 support is "armsr64", but I don't know what you want to do.
The OMR_REAL_TARGET is used only for packages feeds because they can be generic.

GermanAizek, 2024-01-29:
I was trying to build Arm64 image on Arm64 host Ubuntu 20.04

closed 2024-05-04 by github-actions[bot]