Ysurac/openmptcprouter #3142: Failover of the failover ! OpenMptcpRouter backup instance

Mattieug, 2024-01-29:
Hello,

I have implemented OpenMPTCPRouter to failover my internet connexion and I am very happy with that.

Thus, it introduce a new network failure possibility : 
If the unique OpenMPTCPRouter instance crash for any reason, no more dhcp, no more gateway so all my clients would be lost in space.

I am wondering how to have a secondary instance of the router as a backup. Today, I have a second machine ready to start but it implies a manual intervention.

When searching  a way to implement a failover of the failover, I found the possibilty to implement dhcp failover. Would it be a solution ?

How would you manage to have 2 instances of OpenMPTCPRouter (1 primary, 1 backup).

Many thanks.

Ysurac, 2024-01-29:
You should be able to use keepalived: https://openwrt.org/docs/guide-user/network/high-availability

Mattieug, 2024-01-29:
Hello,

excellent solution ! 
So my clients should be able to use VIP pointing seemlessly to the available OpenMPTCPRouter instance.

But, the two OpenMPTCPRouter instances will run both in normal condition. How does the link with the VPSs (I have 2 for reduncy as well) is established ? Ie is it possible to have the 2 OpenMPTCPRouter instances running in the sametime ?

As I understand, packets will follow this path from internet to my client
Internet -> VPS1 or VPS2 -> modem1 or modem2 -> VIP (OpenMPTCPRouter1 or OpenMPTCPRouter2).VIP -> Client. Perfect.
But the 2 OpenMPTCPRouter instances will open a connexion to the active VPS. How will VPS  interpret these 2 tunnel connection requests coming from the same virtual host but 2 different hosts effectivly ?

Sorry, not easy to explain as I am not a network specialist.

thanks

Mattieug, 2024-01-30:
Hello,

I worked all the night to understand how keepalived works and all little configurations surprises it deserves and finally, I just succeeded in setting up a failover system with a primary and a backup OpenMPTCPRouter !

Thanks for the advice.

I don't know if I put in place the state of the art solution but it works.

Could please just confirm it seems ok from your point of view ?

On master router, keepalived maintain a VIP on the lan (192.168.3.1). Its real IP is 192.168.3.2
On secondary router, idem, a VIP on the lan (192.168.3.1). Its real IP is 192.168.3.3
Both server are connected via wan1 to 192.168.2.1 and wan2 192.168.1.1 with the same IP addresses (192.168.x.6)
Both keepalived will launch a script on status change. If MASTER -> ifup wan1 and wan2
If BACKUP -> ifdown wan1 and wan2

In a first option, I was trying to manage wan1 and wan2 as VIP as well but it was unsuccessful.

Other question, is my implementation compliant with futures evolution of your system ?

Thanks !

Ysurac, 2024-01-30:
I never used keepalived yet.
I don't like the WAN down/up part in your config but I think you don't want both router to be connected to the VPS at same time (should work in some case but not tested).
I would like to add keepalived by default, but this not for next release. In any case I will always try to not break a previous configuration.

Mattieug, 2024-01-30:
Ifup/down was the only solution I found. When the 2 routers were connected at the same time, they were using the same wans ip addresses (to be in the dmz of the modems). So they were challenging themselfs to handle the connection and I saw alternativly the vpn mounted on one router or the other…

I think that it would be possible to mount a VIP address on wan1 and another on wan2 but I didn’t succeed…

Ysurac, 2024-01-30:
If you are using Glorytun as VPN, yes it support only one client.

Mattieug, 2024-01-30:
I am using as it was configured out of the box. Better for support.

Thanks.

closed 2024-05-04 by github-actions[bot]