Ysurac/openmptcprouter #3143: Bug adds repo (Bionic) on Ubuntu Touch 20.04 (Focal)

GermanAizek, 2024-01-30:
I decided to check installation omr-vps on arm64 Xiaomi smartphone, Ubuntu Touch works not only on smartphones.

```
phablet@ubuntu-phablet:/tmp$ sudo UPSTREAM6="yes" sh debian-x86_64.sh 
Check user...
Check Linux version...
Check architecture...
Check about broken packages...
Check if OpenMPTCProuter VPS is already installed...
Update mode
Update repo.openmptcprouter.com key
--2024-01-30 12:44:28--  https://repo.openmptcprouter.com/openmptcprouter.gpg.key
Resolving repo.openmptcprouter.com (repo.openmptcprouter.com)... 93.189.63.146
Connecting to repo.openmptcprouter.com (repo.openmptcprouter.com)|93.189.63.146|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 1241 (1.2K) [application/octet-stream]
Saving to: '/etc/apt/trusted.gpg.d/openmptcprouter.gpg'

/etc/apt/trusted.gp 100%[===================>]   1.21K  --.-KB/s    in 0s      

2024-01-30 12:44:28 (9.78 MB/s) - '/etc/apt/trusted.gpg.d/openmptcprouter.gpg' saved [1241/1241]

Remove lock and update packages list...
Hit:1 http://ports.ubuntu.com/ubuntu-ports focal InRelease
Get:2 http://ports.ubuntu.com/ubuntu-ports focal-updates InRelease [114 kB]    
Get:3 http://repo.ubports.com focal InRelease [10.4 kB]                        
Get:4 http://archive.ubuntu.com/ubuntu bionic-backports InRelease [83.3 kB]    
Hit:5 https://repo.openmptcprouter.com next InRelease                          
Hit:6 https://repo.openmptcprouter.com buster InRelease                        
Hit:7 http://ports.ubuntu.com/ubuntu-ports focal-backports InRelease 
Get:8 http://ports.ubuntu.com/ubuntu-ports focal-security InRelease [114 kB]
Get:9 http://archive.ubuntu.com/ubuntu bionic InRelease [242 kB]
Ign:10 http://archive.ubuntu.com/ubuntu bionic-backports/main arm64 Packages
Get:11 http://archive.ubuntu.com/ubuntu bionic-backports/main Translation-en [14.6 kB]
Ign:10 http://archive.ubuntu.com/ubuntu bionic-backports/main arm64 Packages
Ign:10 http://archive.ubuntu.com/ubuntu bionic-backports/main arm64 Packages
Ign:10 http://archive.ubuntu.com/ubuntu bionic-backports/main arm64 Packages
Ign:10 http://archive.ubuntu.com/ubuntu bionic-backports/main arm64 Packages
Err:10 http://archive.ubuntu.com/ubuntu bionic-backports/main arm64 Packages
  404  Not Found [IP: 91.189.91.82 80]
Ign:12 http://archive.ubuntu.com/ubuntu bionic/universe arm64 Packages
Get:13 http://archive.ubuntu.com/ubuntu bionic/universe Translation-en [4941 kB]
Ign:12 http://archive.ubuntu.com/ubuntu bionic/universe arm64 Packages
Ign:12 http://archive.ubuntu.com/ubuntu bionic/universe arm64 Packages
Ign:12 http://archive.ubuntu.com/ubuntu bionic/universe arm64 Packages
Ign:12 http://archive.ubuntu.com/ubuntu bionic/universe arm64 Packages
Err:12 http://archive.ubuntu.com/ubuntu bionic/universe arm64 Packages
  404  Not Found [IP: 91.189.91.82 80]
Fetched 563 kB in 5s (119 kB/s)      
Reading package lists... Done
W: No sandbox user '_apt' on the system, can not drop privileges
E: Failed to fetch http://archive.ubuntu.com/ubuntu/dists/bionic-backports/main/binary-arm64/Packages  404  Not Found [IP: 91.189.91.82 80]
E: Failed to fetch http://archive.ubuntu.com/ubuntu/dists/bionic/universe/binary-arm64/Packages  404  Not Found [IP: 91.189.91.82 80]
E: Some index files failed to download. They have been ignored, or old ones used instead.
```

```
phablet@ubuntu-phablet:/tmp$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.6 LTS
Release:	        20.04
Codename:	focal
```

Ysurac, 2024-01-30:
Ubuntu is not supported for ARM64.
Why do you want VPS part on a smartphone ?

GermanAizek, 2024-01-30:
> Why do you want VPS part on a smartphone ?

VPS locally under another ISP, phone with USB-HDMI-ETH hub is good use as adder omr-vps network bandwidth.

GermanAizek, 2024-02-21:
@Ysurac,
Solution found:

replace from

https://github.com/Ysurac/openmptcprouter-vps/blob/0bd93fb5c8d01aea84303b001d7b6dfa26570b96/debian9-x86_64.sh#L345-L346

to

```sh
echo 'deb http://ports.ubuntu.com/ubuntu-ports bionic-backports main' > /etc/apt/sources.list.d/bionic-backports.list
echo 'deb http://ports.ubuntu.com/ubuntu-ports bionic universe' > /etc/apt/sources.list.d/bionic-universe.list
```

as for your script, you can check username if it is "phablet", then it is Ubuntu Touch. There are no differences between desktop and touch except for Ubuntu repositories

closed 2024-05-27 by github-actions[bot]