Ysurac/openmptcprouter #3144: Attempt building arm64 image OMR on Ubuntu Touch 20.04 (Build dependency: Please install the 'file' package)

GermanAizek, 2024-01-30:
```
$ OMR_TARGET="r2s" OMR_FEED_SRC="master" OMR_KERNEL="5.4" ./build.sh
```

```
WARNING: No feed for package 'libpthread' found
WARNING: No feed for package 'libc' found
WARNING: No feed for package 'librt' found
WARNING: No feed for package 'libpthread' found
WARNING: No feed for package 'libc' found
WARNING: No feed for package 'librt' found
WARNING: No feed for package 'libpthread' found
WARNING: No feed for package 'libc' found
WARNING: No feed for package 'librt' found
WARNING: No feed for package 'libpthread' found
WARNING: No feed for package 'libc' found
WARNING: No feed for package 'librt' found
WARNING: No feed for package 'libpthread' found
WARNING: No feed for package 'libc' found
WARNING: No feed for package 'librt' found
WARNING: No feed for package 'libpthread' found
WARNING: No feed for package 'block-mount' found
WARNING: No feed for package 'blockd' found
WARNING: No feed for package 'fstools' found
Installing package 'luci-proto-gre' from luci
WARNING: No feed for source package 'lua' found
WARNING: No feed for package 'libc' found
WARNING: No feed for package 'librt' found
WARNING: No feed for package 'libpthread' found
WARNING: No feed for package 'gre' found
Installing package 'tcptraceroute' from openmptcprouter
WARNING: No feed for package 'libc' found
WARNING: No feed for package 'librt' found
WARNING: No feed for package 'libpthread' found
WARNING: No feed for package 'libpcap' found
Installing package 'libnet-1.2.x' from packages
WARNING: No feed for package 'libc' found
WARNING: No feed for package 'librt' found
WARNING: No feed for package 'libpthread' found
WARNING: No feed for package 'libpcap' found
WARNING: No feed for package 'kmod-rtl8xxxu' found
WARNING: No feed for package 'kmod-ath9k-htc' found
Installing package 'luci-app-ttyd' from luci
WARNING: No feed for source package 'lua' found
WARNING: No feed for package 'libc' found
WARNING: No feed for package 'librt' found
WARNING: No feed for package 'libpthread' found
Installing package 'ttyd' from packages
WARNING: No feed for package 'libc' found
WARNING: No feed for package 'librt' found
WARNING: No feed for package 'libpthread' found
WARNING: No feed for package 'libopenssl' found
WARNING: No feed for package 'libjson-c' found
WARNING: No feed for package 'zlib' found
Installing package 'libwebsockets' from packages
WARNING: No feed for package 'libc' found
WARNING: No feed for package 'librt' found
WARNING: No feed for package 'libpthread' found
WARNING: No feed for package 'zlib' found
WARNING: No feed for package 'libopenssl' found
WARNING: No feed for package 'libc' found
WARNING: No feed for package 'librt' found
WARNING: No feed for package 'libpthread' found
WARNING: No feed for package 'zlib' found
WARNING: No feed for package 'libmbedtls' found
WARNING: No feed for package 'libc' found
WARNING: No feed for package 'librt' found
WARNING: No feed for package 'libpthread' found
WARNING: No feed for package 'zlib' found
WARNING: No feed for package 'libopenssl' found
[... 897 lines cut ...]
Checking 'seq'... ok.
Checking 'awk'... ok.
Checking 'grep'... ok.
Checking 'egrep'... ok.
Checking 'getopt'... ok.
Checking 'stat'... ok.
Checking 'unzip'... ok.
Checking 'bzip2'... ok.
Checking 'wget'... ok.
Checking 'perl'... ok.
Checking 'python2-cleanup'... ok.
Checking 'python'... ok.
Checking 'python3'... ok.
Checking 'python3-distutils'... ok.
Checking 'git'... ok.
Checking 'file'... failed.
Checking 'rsync'... ok.
Checking 'which'... ok.
Checking 'ldconfig-stub'... ok.

Build dependency: Please install the 'file' package

Prerequisite check failed. Use FORCE=1 to override.
make: *** [/tmp/openmptcprouter/r2s/5.4/source/include/toplevel.mk:180: staging_dir/host/.prereq-build] Error 1
```

Ysurac, 2024-01-30:
You need to install the file package as indicated. ```apt-get install file```

GermanAizek, 2024-01-30:
> You need to install the file package as indicated. `apt-get install file`

It works.

closed 2024-01-30 by GermanAizek