Ysurac/openmptcprouter #3147: Consider testing mullvad's udp over tcp library/binaries as a solution to allow UDP aggregation

ilikenwf, 2024-02-01:
**Is your enhancement proposal related to a problem? Please describe.**

Mullvad has an actively maintained project to facilitate UDP over TCP. If it produces proper TCP, and I would assume it does being by Mullvad, it could allow us to aggregate UDP.

https://github.com/mullvad/udp-over-tcp

**Describe the solution you'd like**

I'd love for someone who knows how better than me to test using the udp-over-tcp system over a multipath setup to see if it works for aggregation and if it does, see it be included in OMR.

**Describe alternatives you've considered**

None, but I'm sure there are others.

ilikenwf, 2024-02-01:
This may be an issue - " So one [Udp2Tcp] instance can handle traffic from a single peer only."

ilikenwf, 2024-02-01:
I'm waiting for hardware to arrive, but once I have my hardware, if I use v2ray, all protocols will route over mptcp, then?

I get mixed messages looking at issues and the wiki.

ilikenwf, 2024-02-01:
There is also this - https://github.com/zehome/MLVPN/

rcboy43, 2024-02-15:
Tested it using Mullvad's official app with the udp-over-tcp enabled. Looking at my network usage on the router shows it's only going over wan interfaces and avoiding the omrvpn. Looks like MPTCP works with the protocol.

ilikenwf, 2024-02-15:
So this could be a new solution?

I ended up going for load balancing for the time being...but still will be watching things here.

Ysurac, 2024-02-17:
I will make some tests. As mullvad have no authentification and no encryption it need to be in a tunnel.

Ysurac, 2024-02-17:
In fact can only be used to pass an UDP VPN over TCP, like doing Wireguard over TCP, because tcp2udp only allow to forward to a fixed listening UDP service.
tcp2udp would require to be patched to support MPTCP with kernel >= 6.1