Ysurac/openmptcprouter #3148: is possible raspberry pi as VPS

R-sajadi, 2024-02-01:
**Is this request related to a missing driver support for a particular hardware platform, SoC or board? Please describe.**

is possible raspberry pi as VPS 
**Describe why you are asking for this support?**
Reduce costs

I want to use the Raspberry Pi as an RTMP server so that only the Raspberry Pi itself plays the same stream received on the HDMI output.
**Additional context**

Ysurac, 2024-02-01:
Snapshot script can be installed on ARM64 under Debian 12.
No idea if it's working or not on RPI.
Remember that the VPS need to have a higher bandwidth than all aggregated connections, so ideally a datacenter or at a place with fiber connection.

GermanAizek, 2024-02-01:
> **Describe why you are asking for this support?**
> Reduce costs

@R-sajadi, with Orange Pi 2 all boards have 1000 Mbit (1Gbe) and more cheapest.

> Remember that the VPS need to have a higher bandwidth than all aggregated connections, so ideally a datacenter or at a place with fiber connection.

@Ysurac, on RPI has ability to connect external 1 GB network card with open drivers, it will come out cheap, but will soc cpu fully use bandwidth via usb 2.0
I'd rather buy another board or mini PC on x64 arch processor (example: ryzen, celeron, atom), they cost same as new rpi 3-4.

GermanAizek, 2024-02-01:
@R-sajadi,
example cheapest external usb 3.0 network card with kernel open source drivers:
- any on AX88179 chipset
https://www.aliexpress.us/item/3256804585756784.html

R-sajadi, 2024-02-01:
> I'd rather buy another board or mini PC on x64 arch processor (example: ryzen, celeron, atom), they cost same as new rpi 3-4.

I will not use usb
The maximum bandwidth that I want to have for video streaming is 4 MB through 3 4G modems.

R-sajadi, 2024-02-01:
[image]
my scenario

R-sajadi, 2024-02-01:
> Snapshot script can be installed on ARM64 under Debian 12.

Which one script should I download for install on ARM64 under Debian 12?

Ysurac, 2024-02-01:
Snapshot documentation is here: https://github.com/Ysurac/openmptcprouter/wiki/Snapshots
You need to use kernel 6.1

R-sajadi, 2024-02-01:
wget -O - http://www.openmptcprouter.com/server-test/debian-x86_64.sh | UPSTREAM6="yes" sh
this Snapshot Not available!!!

Ysurac, 2024-02-01:
What you mean ?

R-sajadi, 2024-02-01:
Which should I download?
1) wget -O - http://www.openmptcprouter.com/server-test/debian-x86_64.sh | UPSTREAM6="yes" sh
or 
2) wget -O - https://www.openmptcprouter.com/server/debian-x86_64.sh | sh

Ysurac, 2024-02-01:
Only snapshot 6.1 VPS script support ARM64, so ```wget -O - http://www.openmptcprouter.com/server-test/debian-x86_64.sh | UPSTREAM6="yes" sh```

R-sajadi, 2024-02-01:
when i use wget -O - http://www.openmptcprouter.com/server-test/debian-x86_64.sh | UPSTREAM6="yes" sh
in terminal this error return:
HTTP request sent, awaiting response... 503 Service Unavailable

R-sajadi, 2024-02-01:
My country has filtered this address!!!!!

Ysurac, 2024-02-01:
What is the country ? I can create another domain.

R-sajadi, 2024-02-01:
Iran

GermanAizek, 2024-02-01:
> What is the country ? I can create another domain.

Mirrors repos list is availabled?

R-sajadi, 2024-02-01:
https://www.openmptcprouter.com/server/debian-x86_64.sh
this is available

Ysurac, 2024-02-01:
You can try to replace www.openmptcprouter.com by wwwopenmptcprouter.b-cdn.net (It's the address of the CDN I use).
You can also check using https instead of http: ```wget -O - https://www.openmptcprouter.com/server-test/debian-x86_64.sh | UPSTREAM6="yes" sh```

R-sajadi, 2024-02-01:
Thank you very much 
this is ok
wget -O - https://www.openmptcprouter.com/server-test/debian-x86_64.sh | UPSTREAM6="yes" sh

Ysurac, 2024-02-01:
ok, thanks for the feedback. I modified the doc to use https instead of http in commands.

R-sajadi, 2024-02-01:
I installed the script how do I know everything is installed correctly?

Ysurac, 2024-02-01:
If the script finished and give you the keys, you rebooted and when you log via SSH you get the info OpenMPTCProuter Server is installed (or check /etc/motd)

R-sajadi, 2024-02-01:
Some packages may not be installed due to filtering!
Which VPN can I use for this?

R-sajadi, 2024-02-01:
security.debian repos

Ysurac, 2024-02-01:
You can try to replace all http URL by https in ```/etc/apt/sources.list``` and in ```/etc/apt/sources.list.d/```

R-sajadi, 2024-02-03:
Hi
when i download this scripc (wget -O - https://www.openmptcprouter.com/server-test/debian-x86_64.sh | UPSTREAM6="yes" sh) Where is it stored?

mahdiyari, 2024-02-03:
I was facing the similar-ish errors coming from the same country.  I assume it is related to the CDN used but not sure.
`wget` to both repo and main domain get stuck due to not being able to make SSL handshake or something like that. I am able to download with `curl -O` though.

Replacing repo and main urls with the HTTP instead of HTTPS works with wget but only if I set an IP address to both domains in the /etc/hosts. I got this IP from outside the country using a VPN. `wget` without changing the hosts on http, returns 503 error on both domains. Changing the hosts doesn't fix https problem.

I also had to edit `/etc/apt/sources.list.d/openmptcprouter.list`  and make it HTTP and uncomment apt-key in the script to not update it during install.

R-sajadi, 2024-02-03:
> I was facing the similar-ish errors coming from the same country. I assume it is related to the CDN used but not sure. `wget` to both repo and main domain get stuck due to not being able to make SSL handshake or something like that. I am able to download with `curl -O` though.
> 
> Replacing repo and main urls with the HTTP instead of HTTPS works with wget but only if I set an IP address to both domains in the /etc/hosts. I got this IP from outside the country using a VPN. `wget` without changing the hosts on http, returns 503 error on both domains. Changing the hosts doesn't fix https problem.
> 
> I also had to edit `/etc/apt/sources.list.d/openmptcprouter.list` and make it HTTP and uncomment apt-key in the script to not update it during install.
> 
> ==========
> 
> Edit: I figured I can set CHINA to yes and it works with just editing VPSURL to http:// and editing /etc/hosts.

If possible, explain in more detail, thank you

mahdiyari, 2024-02-03:
Step by step:

1. Edit `/etc/hosts` and add the following
```
169.150.215.115 www.openmptcprouter.com
169.150.215.115 openmptcprouter.com
169.150.215.115 repo.openmptcprouter.com
```
2. Download the setup script (this is not what you were trying to install above) `wget http://www.openmptcprouter.com/server/debian-x86_64.sh`
3. `sudo chmod +x debian-x86_64.sh`
4. edit debian-x86_64.sh and change VPSURL
```
VPSURL="http://www.openmptcprouter.com/"
```
5. `sudo CHINA=yes ./debian-x86_64.sh`

R-sajadi, 2024-02-03:
what is my problem?
[image]

R-sajadi, 2024-02-04:
I changed ISP and now I can connect to all repositories
root@rpi4-20231108:~# wget -O - https://www.openmptcprouter.com/server-test/debian-x86_64.sh | UPSTREAM6="yes" sh
--2024-02-04 10:22:43--  https://www.openmptcprouter.com/server-test/debian-x86_64.sh
Resolving www.openmptcprouter.com (www.openmptcprouter.com)... 138.199.46.68, 2400:52e0:1500::868:1
Connecting to www.openmptcprouter.com (www.openmptcprouter.com)|138.199.46.68|:443... connected.
GnuTLS: The TLS connection was non-properly terminated.
Unable to establish SSL connection.

R-sajadi, 2024-02-07:
when i use apt update all pakages update succssefully. But why does this problem happen?
Update mode
Remove lock and update packages list...
Hit:1 http://deb.debian.org/debian bookworm InRelease
Hit:2 http://deb.debian.org/debian bookworm-updates InRelease
Hit:3 http://security.debian.org/debian-security bookworm-security InRelease

Ysurac, 2024-02-07:
What is the problem ?

R-sajadi, 2024-02-07:
i can not get these 
Hit:1 http://deb.debian.org/debian bookworm InRelease
Hit:2 http://deb.debian.org/debian bookworm-updates InRelease
Hit:3 http://security.debian.org/debian-security bookworm-security InRelease

R-sajadi, 2024-02-07:
What is the my problem ?
root@rpi4-20231108:~# wget -O - https://www.openmptcprouter.com/server-test/debian-x86_64.sh | UPSTREAM6="yes" sh
--2024-02-07 11:03:39--  https://www.openmptcprouter.com/server-test/debian-x86_64.sh
Resolving www.openmptcprouter.com (www.openmptcprouter.com)... 138.199.46.68, 2400:52e0:1500::868:1
Connecting to www.openmptcprouter.com (www.openmptcprouter.com)|138.199.46.68|:443... connected.
GnuTLS: The TLS connection was non-properly terminated.
Unable to establish SSL connection.

Ysurac, 2024-02-07:
something blocking connections.
If you can do a ```git clone https://github.com/Ysurac/openmptcprouter-vps.git``` and inside the cloned directory run ```UPSTREAM6="yes" LOCALFILES="yes" sh debian9-x86_64.sh``` this may help for some part.

Ysurac, 2024-02-26:
For 6.1 default distribution kernel is used.
Why using wireguard ?
I don't know if your kernel have MPTCP enabled, you can check with ```cat /proc/sys/net/mptcp/enabled```

R-sajadi, 2024-02-27:
@Tillo7 
Hi 
 What kernel did you install? Debian or Ubuntu
Please write the download link of the image file

Tillo7, 2024-02-27:
> For 6.1 default distribution kernel is used. Why using wireguard ? I don't know if your kernel have MPTCP enabled, you can check with `cat /proc/sys/net/mptcp/enabled`

Did I understand correctly that '-mptcp' may not be present with 'uname -a' in version 6.1?"

Version 5.4. works well on virtualBox. Ubuntu. ($ uname -a Linux ttest 5.4.207-mptcp #1 SMP Sun Jul 24 14:39:44 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux).
For good work, you had to turn on "MPTCP over VPN" (Wireguard). And it worked.
In version 6.1. on raspberry pi4 it doesn't work.

~# cat /proc/sys/net/mptcp/enabled
1

Tillo7, 2024-02-27:
> @Tillo7 Hi Which kernel did you install? Debian or Ubuntu I install this one https://raspi.debian.net/tested/20231109_raspi_4_bookworm.img.xz Please write the download link of the image file

I tried some recommended versions from the raspberry pi imager app.
The recommended images are based on Debian.
The result is the same.
The last test was carried out on the full version with a desktop.

Ysurac, 2024-02-27:
@Tillo7:
* yes, no "-mptcp" in the uname -a
* MPTCP over VPN should only be used if the ISP block MPTCP. I don't know if there is any provider that block MPTCP v1 for now...
* What s not working exactly ? Aggregation is working or not at all ? The internal VPN seems to be not working, what is the VPN used ? Did you check log on the VPS ?

I reiterate that RPI is not a supported platform for VPS part: not tested at all and it will not be tested.

R-sajadi, 2024-02-27:
> I tried some recommended versions from the raspberry pi imager app. The recommended images are based on Debian. The result is the same. The last test was carried out on the full version with a desktop.

Does that mean you installed the Raspberry Pi OS ? And then run the script?
[image]

Tillo7, 2024-02-28:
> Does that mean you installed the Raspberry Pi OS ? And then run the script?

Yes. 64 bit
Login.
sudo -i
apt-get update && apt-get upgrade
wget -O - https://www.openmptcprouter.com/server-test/debian-x86_64.sh | UPSTREAM6 = "yes" sh