Ysurac/openmptcprouter #3150: VPS Snapshot 'iperf3' error

psfabio, 2024-02-01:
## Expected Behavior
Install VPS script without error
## Current Behavior
There is an error on install script.

## Possible Solution

## Steps to Reproduce the Problem

got http://www.openmptcprouter.com/server-test/debian-x86_64.sh and install.

## Context (Environment)

## Specifications

OpenMPTCProuter VPS version: script got 01/02/2024
OpenMPTCProuter VPS provider: Oracle
tested both on ubuntu 20.04.6 LTS with vps already running and on a "clean Installation"
kernel 5.4

error message:

`
Install iperf3
Reading package lists... Done
Building dependency tree
Reading state information... Done
iperf3 is already the newest version (3.7-3).
0 upgraded, 0 newly installed, 0 to remove and 62 not upgraded.
chown: invalid user: 'iperf3'

Thanks

Ysurac, 2024-02-01:
Should be fixed, but you should use Debian.

psfabio, 2024-02-01:
> Should be fixed, but you should use Debian.

yes, i know, you  always say that..  ;-)
At oracle, they don´t have debian images, so it is more difficult to deploy, because you need custom images..
But i will change asap.

Thanks again..

vempire-ghost, 2024-02-01:
> > Should be fixed, but you should use Debian.
> 
> yes, i know, you always say that.. ;-) At oracle, they don´t have debian images, so it is more difficult to deploy, because you need custom images.. But i will change asap.
> 
> Thanks again..

Back when I used Oracle, this guide woked fine to transform ubuntu in debian.
https://gist.github.com/ClashTheBunny/8da6f422fc955bfd63ba1b03dd8254c8

psfabio, 2024-02-01:
> Back when I used Oracle, this guide woked fine to transform ubuntu in debian. https://gist.github.com/ClashTheBunny/8da6f422fc955bfd63ba1b03dd8254c8

Thanks

closed 2024-05-06 by github-actions[bot]