Ysurac/openmptcprouter #3154: SIM card in 4G USB Dongle not recognized

filmmaker75, 2024-02-07:
## Expected Behavior
I have two perfectly working HUAWEI 3372-325 dongles, I would like to add the third.

## Current Behavior
I have currently successfully configured two HUAWEI E3372-325 USB dongles with two SIM cards from two different carriers. Everything runs on a Raspberry PI4 and a VPS on Aruba, it works perfectly and the aggregation happens correctly. I purchased a third HUAWEI E3372-325 4G USB dongle and a third SIM from another provider to increase the bandwidth. Unfortunately when I connect the latter to the Raspberry the LED flashes green/red and the interface is not recognised. If I try the dongle in the PC everything works fine. I also tried swapping SIMs and dongles but to no avail. It seems that the new SIM is not recognized by OpenMPTCProuter and consequently the relevant USB interface is not created. Can I solve it somehow? Thank you.

## Specifications

  - OpenMPTCProuter version: 0.59.1-5.4 r0+16594-ce92de8c8c
  - OpenMPTCProuter VPS version: 0.1028.5.4.207-mptcp
  - OpenMPTCProuter VPS provider: ARUBA NETWORK
  - OpenMPTCProuter platform: RPI4 model B

Ysurac, 2024-02-07:
The red light flashing on a RPI indicate an undervoltage problem.
You have this only with one SIM in any dongle ? In this case you should really try a USB powered hub or another RPI power supply.
What is the SIM provider ?

filmmaker75, 2024-02-07:
The red LED that flashes is that of the HUAWEI E3372-325. The RPI works properly, I use its original power supply. Again, if I connect two dongles it works perfectly. The problem is not the dongle but the SIM.
SIM 1 - Iliad
SIM 2 - HO.mobile
SIM 3 - Very mobile
Even trying them one at a time, the first two work while the third (very) doesn't. If I try it on my Windows PC it works correctly.

R-sajadi, 2024-02-07:
LED on RPI or HUAWEI 3372-325 dongles?

filmmaker75, 2024-02-07:
> LED on RPI or HUAWEI 3372-325 dongles?

The LED on the HUAWEI dongle flashes from red to green then red then green endlessly.
[image]

Ysurac, 2024-02-07:
ok, so not a power problem.
This dongle give directly some eth* interfaces or you need to use ModemManager ?

filmmaker75, 2024-02-07:
> ok, so not a power problem.
> This dongle give directly some eth* interfaces or you need to use ModemManager ?

With working SIMs they are seen as USB0 and USB1 on openmptcprouter. However, if I insert the VERY SIM, no interface is created.

Ysurac, 2024-02-07:
What do you have in Status->System log when you plug the dongle ?

filmmaker75, 2024-02-07:
> What do you have in Status->System log when you plug the dongle ?

[system_log_very.txt](https://github.com/Ysurac/openmptcprouter/files/14199594/system_log_very.txt)

This is the log with the VERY SIM connected alone.

Ysurac, 2024-02-07:
I see nothing here. Maybe in Status->kernel log ? Can you plug/unplug the dongle when running ?

filmmaker75, 2024-02-07:
> I see nothing here. Maybe in Status->kernel log ? Can you plug/unplug the dongle when running ?

This is kernel log. I tried inserting and removing several times, even changing the USB port:

[kernel_log_very.txt](https://github.com/Ysurac/openmptcprouter/files/14199726/kernel_log_very.txt)

I also attach the kernel log of the two working sims if it can be useful to you. Thanks for your patience.

[kernel_log_iliad_ho.txt](https://github.com/Ysurac/openmptcprouter/files/14199730/kernel_log_iliad_ho.txt)

Ysurac, 2024-02-07:
Ok, the dongle is not in the good mode (no idea why only on this SIM...).
Can you try, under the working PC, to upgrade the dongle by going to it's web interface (or maybe change mode from here, no idea if it's posssible) ?

filmmaker75, 2024-02-07:
> Can you try, under the working PC, to upgrade the dongle by going to it's web interface (or maybe change mode from here, no idea if it's posssible) ?

From its web interface I don't have many options. If I send you the screenshots can you tell me what to change?

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Ysurac, 2024-02-07:
Can you try to put this file 
[usb-mode.json](https://github.com/Ysurac/openmptcprouter/files/14200082/usb-mode.json) in ```/etc/``` on the router and reboot it ?

filmmaker75, 2024-02-07:
> Can you try to put this file [usb-mode.json](https://github.com/Ysurac/openmptcprouter/files/14200082/usb-mode.json) in `/etc/` on the router and reboot it ?

Of course I can try but you have to explain to me how to do it. I don't know how to do it.

Ysurac, 2024-02-07:
Via SSH, using scp command line or WinSCP under windows

filmmaker75, 2024-02-07:
Done. It asked me to overwrite the old file and I confirmed. Restarted everything but same situation.

Ysurac, 2024-02-07:
I have no more idea...
Really strange that you have issue only with a SIM card...
On this type of dongle, that are using HiLink mode, all is done on modem side. For OpenMPTCProuter it's like an ethernet device.
The json I gave you should force the dongle to switch to modem mode instead of storage mode, but this doesn't work.
It's like if the SIM doesn't allow the dongle to be in modem mode... Don't know if it's possible if the key is not customized by provider...

filmmaker75, 2024-02-08:
> I have no more idea...
> Really strange that you have issue only with a SIM card...
> On this type of dongle, that are using HiLink mode, all is done on modem side. For OpenMPTCProuter it's like an [...]

The key is not personalized. Even by changing the key the situation does not change. By inserting one of the working SIMs into this one, it works too. So the problem is definitely on the SIM. I will do some tests with some Linux distributions to verify how they work but my knowledge of Linux is zero. Thanks anyway for everything.

filmmaker75, 2024-02-08:
I'll update the post with a test I did out of curiosity. I prepared a pendrive with a live debian distro. I started it on my PC and connected the HUAWEI dongle with the Very SIM. And it doesn't work, it has the same behavior as on the RPI4 with OMR. So I assume that there is something blocking it on Linux's part. Obviously I also tried inserting the other two SIMs and, as on OMR, they work perfectly. Evidently there is some different command to give or some sort of incompatibility between Linux and the Very SIM.

closed 2024-05-14 by github-actions[bot]