Ysurac/openmptcprouter #3155: Orange Pi Zero

mitjaya, 2024-02-07:
Good evening. Who will never be able to get vaccinated for Pi zero? I always get an error. I've tried on different systems (Ubuntu and Debian), but I've never tried something else.I am an inexperienced Linux user.
My English with translation
make[1]: *** [package/Makefile:114: /home/lawr1/openmptcprouter/pi/6.1/source/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/lawr1/openmptcprouter/pi/6.1/source'
make: *** [/home/lawr1/openmptcprouter/pi/6.1/source/include/toplevel.mk:232: world] Error 2

Ysurac, 2024-02-07:
The error should be above this logs info.

mitjaya, 2024-02-07:
Thanks for the quick response and for your work.After a while, I'll try to do it again and copy it here

mitjaya, 2024-02-08:
I've tried both make -j3 and make -J1.
This is the beginning

lawr@vm2627147:~/openmptcprouter/zero/5.4/source$ make -j3 V=s
WARNING: Makefile 'package/utils/busybox/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a build dependency on 'libpam', which does not exist
WARNING: Makefile 'package/kernel/linux/Makefile' has a dependency on 'kmod-cdrom', which does not exist
WARNING: Makefile 'package/kernel/linux/Makefile' has a dependency on 'kmod-cdrom', which does not exist
WARNING: Makefile 'package/kernel/linux/Makefile' has a dependency on 'kmod-cdrom', which does not exist
WARNING: Makefile 'package/kernel/linux/Makefile' has a dependency on 'kmod-cdrom', which does not exist
WARNING: Makefile 'package/kernel/linux/Makefile' has a dependency on 'kmod-ledtrig-audio', which does not exist
WARNING: Makefile 'package/feeds/openmptcprouter/openmptcprouter-full/Makefile' has a dependency on 'dnsmasq_full_ipset', which does not exist
WARNING: Makefile 'package/feeds/openmptcprouter/openmptcprouter-full/Makefile' has a dependency on 'kmod-tcp-bbr2', which does not exist
WARNING: Makefile 'package/feeds/openmptcprouter/openmptcprouter-full/Makefile' has a dependency on 'kmod-crypto-lib-blake2s', which does not exist
WARNING: Makefile 'package/utils/policycoreutils/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/policycoreutils/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/policycoreutils/Makefile' has a build dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/wmt/Makefile' has a dependency on 'kmod-mt6625l-wlan-gen2', which does not exist
time: target/linux/prereq#0.26#0.15#0.42

It's been 3 hours

find /home/lawr/openmptcprouter/zero/5.4/source/build_dir/hostpkg/go-1.20.999 -mindepth 1 -maxdepth 1 -not '(' -type f -and -name '.*' -and -size 0 ')' -print0 | xargs -r -0 rm -rf
make[3]: Leaving directory '/home/lawr/openmptcprouter/feeds/openmptcprouter/golang/golang'
time: package/feeds/openmptcprouter/golang/host-compile#1171.14#185.17#2238.15
make[2]: Leaving directory '/home/lawr/openmptcprouter/zero/5.4/source'
make[1]: *** [package/Makefile:110: /home/lawr/openmptcprouter/zero/5.4/source/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/lawr/openmptcprouter/zero/5.4/source'
make: *** [/home/lawr/openmptcprouter/zero/5.4/source/include/toplevel.mk:230: world] Error 2

Ysurac, 2024-02-08:
The error is between all this, here I would say from last log that it's a problem with golang. But here you are compiling for kernel 5.4 and not with develop branch

Ysurac, 2024-02-08:
The real error is often a few lines above end of logs.
You should try with ```OMR_FEED_SRC="develop" OMR_KERNEL="6.1"```

mitjaya, 2024-02-09:
I tried it. The same mistake

if [ -d /home/lawr/openmptcprouter/zero/6.1/source/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/u-boot-Lamobo_R1/u-boot-2023.04/.pkgdir/u-boot-Lamobo_R1 ]; then install -d -m0755 /home/lawr/openmptcprouter/zero/6.1/source/bin/targets/sunxi/cortexa7/u-boot-Lamobo_R1 && cp -fpR /home/lawr/openmptcprouter/zero/6.1/source/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/u-boot-Lamobo_R1/u-boot-2023.04/.pkgdir/u-boot-Lamobo_R1/. /home/lawr/openmptcprouter/zero/6.1/source/bin/targets/sunxi/cortexa7/u-boot-Lamobo_R1/; fi
touch -r /home/lawr/openmptcprouter/zero/6.1/source/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/u-boot-Lamobo_R1/u-boot-2023.04/.built /home/lawr/openmptcprouter/zero/6.1/source/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/u-boot-Lamobo_R1/u-boot-2023.04/.autoremove 2>/dev/null >/dev/null
find /home/lawr/openmptcprouter/zero/6.1/source/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/u-boot-Lamobo_R1/u-boot-2023.04 -mindepth 1 -maxdepth 1 -not '(' -type f -and -name '.*' -and -size 0 ')' -and -not -name '.pkgdir'  -print0 | xargs -r -0 rm -rf
make[3]: Leaving directory '/home/lawr/openmptcprouter/zero/6.1/source/package/boot/uboot-sunxi'
time: package/boot/uboot-sunxi/Lamobo_R1/compile#116.11#19.78#81.63
make[2]: Leaving directory '/home/lawr/openmptcprouter/zero/6.1/source'
make[1]: *** [package/Makefile:122: /home/lawr/openmptcprouter/zero/6.1/source/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/lawr/openmptcprouter/zero/6.1/source'
make: *** [/home/lawr/openmptcprouter/zero/6.1/source/include/toplevel.mk:232: world] Error 2

Ysurac, 2024-02-09:
Not sure if it's the same, you are using -j1 V=s ? in this case the error should be before that. For now I still don't know what is the error.

mitjaya, 2024-02-09:
Yes, I tried both J3 V=s and J1 V=s

mitjaya, 2024-02-09:
Launching make menuconfig. I get errors-
~/openmptcprouter/zero/6.1/source$ make menuconfig
WARNING: Makefile 'package/utils/busybox/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a build dependency on 'libpam', which does not exist
WARNING: Makefile 'package/feeds/openmptcprouter/openmptcprouter-full/Makefile' has a dependency on 'dnsmasq_full_ipset', which does not exist
WARNING: Makefile 'package/feeds/openmptcprouter/openmptcprouter-full/Makefile' has a dependency on 'dnsmasq_full_nftset', which does not exist
WARNING: Makefile 'package/feeds/openmptcprouter/openmptcprouter-full/Makefile' has a dependency on 'kmod-net-rtl8192su', which does not exist
WARNING: Makefile 'package/feeds/openmptcprouter/openmptcprouter-full/Makefile' has a dependency on 'kmod-rtl8187', which does not exist
WARNING: Makefile 'package/feeds/openmptcprouter/openmptcprouter-full/Makefile' has a dependency on 'kmod-crypto-lib-blake2s', which does not exist
WARNING: Makefile 'package/feeds/openmptcprouter/openmptcprouter-full/Makefile' has a dependency on 'kmod-tcp-bbr2', which does not exist
WARNING: Makefile 'package/feeds/openmptcprouter/openmptcprouter-full/Makefile' has a dependency on 'luci-i18n-base-en', which does not exist
WARNING: Makefile 'package/feeds/openmptcprouter/openmptcprouter-mini/Makefile' has a dependency on 'kmod-net-rtl8192su', which does not exist
WARNING: Makefile 'package/feeds/openmptcprouter/openmptcprouter-mini/Makefile' has a dependency on 'kmod-rtl8187', which does not exist
WARNING: Makefile 'package/feeds/openmptcprouter/openmptcprouter-mini/Makefile' has a dependency on 'luci-i18n-base-en', which does not exist
WARNING: Makefile 'package/utils/policycoreutils/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/policycoreutils/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/policycoreutils/Makefile' has a build dependency on 'libpam', which does not exist
WARNING: Makefile 'package/feeds/openmptcprouter/shadowsocks-v2ray-plugin/Makefile' has a dependency on 'golang-golang-x-crypto-dev', which does not exist
WARNING: Makefile 'package/utils/wmt/Makefile' has a dependency on 'kmod-mt6625l-wlan-gen2', which does not exist

*** End of the configuration.
*** Execute 'make' to start the build or try 'make help'.

Ysurac, 2024-02-09:
Warning on menuconfig are not a problem.

mitjaya, 2024-02-09:
Is there a difference in Ubuntu 32 bit or 64bit for compilation? lib32gcc1 is available in the tools. Is it possible to use lib64gcc1? Orange Pi-zero-32bit

Ysurac, 2024-02-10:
Should not have differences for compilation.
I just fixed a bug in Golang compilation.

mitjaya, 2024-02-10:
Without changes. The same mistake

Ysurac, 2024-02-10:
As I said, without more log I can't help.
And a Orange Pi Zero have a A53 CPU, like a RPI3. It's slow...

mitjaya, 2024-02-10:
pi zero- H2

mitjaya, 2024-02-10:
https://github.com/Harwesta/Biforcom-MCR-routers

mitjaya, 2024-02-11:
New errors have appeared

/home/lawr/openmptcprouter/pi/5.4/source/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-sunxi_cortexa7/realtek-r8125-dkms-9.010.01-1/src/r8125_n.c: In function 'rtl8125_suspend':
/home/lawr/openmptcprouter/pi/5.4/source/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-sunxi_cortexa7/realtek-r8125-dkms-9.010.01-1/src/r8125_n.c:15637:9: error: implicit declaration of function 'pci_prepare_to_sleep'; did you mean 'pci_resource_to_user'? [-Werror=implicit-function-declaration]
         pci_prepare_to_sleep(pdev);
         ^~~~~~~~~~~~~~~~~~~~
         pci_resource_to_user
cc1: some warnings being treated as errors
make[5]: *** [scripts/Makefile.build:262: /home/lawr/openmptcprouter/pi/5.4/source/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-sunxi_cortexa7/realtek-r8125-dkms-9.010.01-1/src/r8125_n.o] Error 1
make[4]: *** [Makefile:1746: /home/lawr/openmptcprouter/pi/5.4/source/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-sunxi_cortexa7/realtek-r8125-dkms-9.010.01-1/src] Error 2
make[4]: Leaving directory '/home/lawr/openmptcprouter/pi/5.4/source/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-sunxi_cortexa7/linux-5.4.238'
make[3]: *** [Makefile:40: /home/lawr/openmptcprouter/pi/5.4/source/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-sunxi_cortexa7/realtek-r8125-dkms-9.010.01-1/.built] Error 2
make[3]: Leaving directory '/home/lawr/openmptcprouter/feeds/openmptcprouter/r8125'
time: package/feeds/openmptcprouter/r8125/compile#2.52#0.64#3.57
    ERROR: package/feeds/openmptcprouter/r8125 failed to build.
make[2]: *** [package/Makefile:116: package/feeds/openmptcprouter/r8125/compile] Error 1
make[2]: Leaving directory '/home/lawr/openmptcprouter/pi/5.4/source'
make[1]: *** [package/Makefile:110: /home/lawr/openmptcprouter/pi/5.4/source/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/lawr/openmptcprouter/pi/5.4/source'
make: *** [/home/lawr/openmptcprouter/pi/5.4/source/include/toplevel.mk:230: world] Error 2

closed 2024-05-17 by github-actions[bot]