Ysurac/openmptcprouter #3156: image on ROCK pi 4 C+ --> Erreur DNS: impossible de resoudre des noms d'hotes

yannick2801, 2024-02-09:
Hi, 

what can i check to resolve this issue ? 
(my VPS works on an other support)

  - OpenMPTCProuter version: Linux OpenMPTCProuter 5.4.238 #0 SMP PREEMPT aarch64 GNU/Linux
  - OpenMPTCProuter VPS version: 0.1026 5.4.100
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform: ROCK pi 4C+

image on ROCK pi 4 C+ --> **Erreur DNS: impossible de resoudre des noms d'hotes**

[image]

Thanks
BR
Yannick

Ysurac, 2024-02-09:
It's not a supported platform.
Did you change DNS configuration ? Check what you have in Network->DHCP&DNS
Via SSH on te router check if ```ping 1.1.1.1``` is working, if yes check ```dig google.com``` check if you have an IP in results, if no, try ```dig google.com @1.1.1.1``` check if you have IP in results, and if yes check that dnsmasq and unbound are running (```ps l```)

yannick2801, 2024-02-12:
hi, i don't think i have chnage DNS Configuration,
I have tryed does commands (you can see below): 
But on this router i have no answer to `ps 1`

[image]

Thanks
BR
Yannick

Ysurac, 2024-02-12:
Not ```ps 1``` but ```ps l```.
Make sure you didn't set any DNS on Network->Interfaces, and check if you don't have errors in Status->System log

yannick2801, 2024-02-12:
in `ps l`, i haven't gor unbound, that's strange.

yannick2801, 2024-02-12:
still de same : 
_Feb 12 12:27:10 OpenMPTCProuter user.notice unbound: default protocol configuration
Feb 12 12:27:10 OpenMPTCProuter user.notice unbound: default memory configuration
Feb 12 12:27:10 OpenMPTCProuter user.notice unbound: default recursion configuration
Feb 12 12:27:35 OpenMPTCProuter daemon.info procd: Instance unbound::unbound s in a crash loop 6 crashes, 0 seconds since last crash
Feb 12 12:28:06 OpenMPTCProuter user.notice unbound: default protocol configuration
Feb 12 12:28:06 OpenMPTCProuter user.notice unbound: default memory configuration
Feb 12 12:28:06 OpenMPTCProuter user.notice unbound: default recursion configuration
Feb 12 12:28:31 OpenMPTCProuter daemon.info procd: Instance unbound::unbound s in a crash loop 6 crashes, 0 seconds since last crash_

```
root@OpenMPTCProuter:~# /etc/init.d/unbound restart
unbound: default protocol configuration
unbound: default memory configuration
unbound: default recursion configuration
```

```
root@OpenMPTCProuter:~# uci show unbound
unbound.ub_main=unbound
unbound.ub_main.add_extra_dns='0'
unbound.ub_main.add_local_fqdn='1'
unbound.ub_main.add_wan_fqdn='0'
unbound.ub_main.dhcp_link='none'
unbound.ub_main.dhcp4_slaac6='0'
unbound.ub_main.dns64='0'
unbound.ub_main.dns64_prefix='64:ff9b::/96'
unbound.ub_main.domain='lan'
unbound.ub_main.domain_type='static'
unbound.ub_main.edns_size='1232'
unbound.ub_main.extended_stats='0'
unbound.ub_main.hide_binddata='1'
unbound.ub_main.interface_auto='1'
unbound.ub_main.listen_port='53'
unbound.ub_main.localservice='1'
unbound.ub_main.manual_conf='0'
unbound.ub_main.num_threads='1'
unbound.ub_main.protocol='default'
unbound.ub_main.query_minimize='0'
unbound.ub_main.query_min_strict='0'
unbound.ub_main.rate_limit='0'
unbound.ub_main.rebind_localhost='0'
unbound.ub_main.rebind_protection='1'
unbound.ub_main.recursion='default'
unbound.ub_main.resource='default'
unbound.ub_main.root_age='9'
unbound.ub_main.ttl_min='120'
unbound.ub_main.ttl_neg_max='1000'
unbound.ub_main.unbound_control='0'
unbound.ub_main.validator='0'
unbound.ub_main.validator_ntp='1'
unbound.ub_main.verbosity='1'
unbound.ub_main.iface_trig='lan' 'wan'
unbound.ub_main.iface_wan='wan'
unbound.auth_icann=zone
unbound.auth_icann.enabled='0'
unbound.auth_icann.fallback='1'
unbound.auth_icann.url_dir='https://www.internic.net/domain/'
unbound.auth_icann.zone_type='auth_zone'
unbound.auth_icann.server='lax.xfr.dns.icann.org' 'iad.xfr.dns.icann.org'
unbound.auth_icann.zone_name='.' 'arpa.' 'in-addr.arpa.' 'ip6.arpa.'
unbound.fwd_isp=zone
unbound.fwd_isp.enabled='0'
unbound.fwd_isp.fallback='1'
unbound.fwd_isp.resolv_conf='1'
unbound.fwd_isp.zone_type='forward_zone'
unbound.fwd_isp.zone_name='isp-bill.example.com.' 'isp-mail.example.net.'
unbound.fwd_google=zone
unbound.fwd_google.enabled='0'
unbound.fwd_google.fallback='1'
[... 70 lines cut ...]
unbound.fwd_quad9_unsecured.enabled='0'
unbound.fwd_quad9_unsecured.fallback='1'
unbound.fwd_quad9_unsecured.tls_index='dns10.quad9.net'
unbound.fwd_quad9_unsecured.tls_upstream='1'
unbound.fwd_quad9_unsecured.zone_type='forward_zone'
unbound.fwd_quad9_unsecured.server='9.9.9.10' '149.112.112.10'
unbound.fwd_quad9_unsecured.zone_name='.'
unbound.fwd_quad9_ecs=zone
unbound.fwd_quad9_ecs.enabled='0'
unbound.fwd_quad9_ecs.fallback='1'
unbound.fwd_quad9_ecs.tls_index='dns11.quad9.net'
unbound.fwd_quad9_ecs.tls_upstream='1'
unbound.fwd_quad9_ecs.zone_type='forward_zone'
unbound.fwd_quad9_ecs.server='9.9.9.11' '149.112.112.11'
unbound.fwd_quad9_ecs.zone_name='.'
unbound.fwd_quad9_secured=zone
unbound.fwd_quad9_secured.enabled='0'
unbound.fwd_quad9_secured.fallback='1'
unbound.fwd_quad9_secured.tls_index='dns9.quad9.net'
unbound.fwd_quad9_secured.tls_upstream='1'
unbound.fwd_quad9_secured.zone_type='forward_zone'
unbound.fwd_quad9_secured.server='9.9.9.9' '149.112.112.9'
unbound.fwd_quad9_secured.zone_name='.'
root@OpenMPTCProuter:~#
```

Ysurac, 2024-02-12:
There is a problem here: ```unbound.ub_main.listen_port='53'``` dnsmasq is already listening to this port, by default it's set to 5353 on OpenMPTCProuter.

yannick2801, 2024-02-12:
Thanks

i have added those 2 lines : 
[image]

seems better but i think i have an other probleme: 
[image]

Ysurac, 2024-02-12:
Did you also change port in Services->Recursive DNS to 5353 ?

yannick2801, 2024-02-12:
perfect, thank you very mutch

Trent63, 2024-02-17:
> Hi,
>
> what can i check to resolve this issue ? (my VPS works on an other support)
>
> * OpenMPTCProuter version: Linux OpenMPTCProuter 5.4.238 #0 SMP PREEMPT aarch64 GNU/Linux
> * OpenMPTCProuter VPS [...]

Привет. Как с вами связаться напрямую? у меня есть вопрос по сборке OMR на RockPI

closed 2024-05-23 by github-actions[bot]