Ysurac/openmptcprouter #3159: Wrong wan addresses report on state page

Dragonox77, 2024-02-12:
Hello,
Thank you for your work.

## Expected Behavior

Get valid wan addresses report on the state page

## Current Behavior

Wan addresses are the same on wan1 & wan2

[image]

## Specifications

  - OpenMPTCProuter version: 0.59.1-5.4
  - OpenMPTCProuter VPS version: 0.1028 5.4.207-mptcp
  - OpenMPTCProuter VPS provider: OVH BareMetal
  - OpenMPTCProuter platform: x86_64
  - Country: FR

IPV6 is disabled on wan1 (orange), but I can't disable it on wan2 (starlink). IPV6 is disabled on the OMR router.

I also had to disable ping check on gateways because it keep saying unable to get answer from gateway. (sometimes on wan1, and sometimes on wan2) (my router is very slow, i only use it for testing purpose, so I assume this is the cause :D)

Ysurac, 2024-02-12:
You use macvlan or vlan ? If not it's the problem.

Dragonox77, 2024-02-12:
Thank you!
I can activate back the ping to the gateways, and it show no error, plus, WAN addresses are correct.

closed 2024-02-12 by Dragonox77