Ysurac/openmptcprouter #3160: Installation of OMR Project on RUTX12 Router

Yessine1200, 2024-02-12:
Greetings,

I hope this email finds you well.

I am writing to inquire about the installation process for the OpenMPTCProuter (OMR) software on the Teltonika RUTX12 router. As I am currently working on a project that involves deploying OMR on Teltonika devices, I would greatly appreciate your guidance on how to install OMR specifically on the RUTX12 model.

Could you please provide detailed instructions or documentation on how to install OMR on the Teltonika RUTX12 router? Additionally, if there are any specific considerations or configurations to be aware of during the installation process, I would be grateful for any insights you can provide.

Thank you for your attention to this matter, and I look forward to your prompt response.

Best regards,
Yessine

Ysurac, 2024-02-12:
For current stable release you can install via web interface or RUTX recover mode.
Latest snapshot is not supported as my RUTX12 is dead.

Yessine1200, 2024-02-12:
Can you provide me the link of the stable release ? and how I install it on recover mode ?

Yessine1200, 2024-02-13:
I recently configured the first version of RUTX12 and upgraded the firmware with the openMPTCProuter squashfs image. It's been nearly 3 hours, and the process is still ongoing. Is this normal?
[image]

Ysurac, 2024-02-13:
No, you can reboot it.
Don't forget that port indicated as WAN should now be the new LAN port.

Yessine1200, 2024-02-13:
I reboot it manually? If so, I just did it, and I see nothing happening.

Ysurac, 2024-02-13:
You should have DHCP server enabled on WAN port.
But, if I remember correctly there is in fact 2 versions of RUTX12 model, current stable version only support the first version.

Yessine1200, 2024-02-13:
But I didn't enable the DHCP server on the WAN port. What should I do?
When you mention the modem versions for the RUTX12, are you referring to the firmware version?

Ysurac, 2024-02-13:
When OpenMPTCProuter is installed, DHCP is enabled by default  on WAN port.
I'm referring to the internal board used in RUTX12: one with STM32 and the new one without it.

Yessine1200, 2024-02-13:
I don't have the specifications of my RUTX12 but I think is the new one.
[image]

Ysurac, 2024-02-13:
Maybe, so not supported for now by OpenMPTCProuter as I don't have the device (and it's about 431€...) Sorry

Yessine1200, 2024-02-13:
Could you please provide me with the link to the device? I'm considering making a purchase.
Additionally, I'll inspect my internal board to verify if it's the supported one. I'll get back to you once I've completed this check. Is that acceptable?

Ysurac, 2024-02-13:
https://www.amazon.fr/Teltonika-802-11ac-W125768412-Ethernet-Tabletop/dp/B08C56GGV2
I have no idea how to check on internal board the old/bad one. I think all models after 2021 are the new version.

Yessine1200, 2024-02-13:
I will see that. 
Also,I have a few questions about OMR:
1) Is it possible to retrieve information about each link using a script, such as latency, speed, etc.?
2) Can I automatically select the policy to pass through one link or another based on predefined criteria using a script?

Yessine1200, 2024-02-15:
Hello Ysurac,
I'm following up on my previous inquiries as I haven't received responses yet. Could you please provide answers at your earliest convenience?

Ysurac, 2024-02-15:
For the first point, you can use OpenMPTCProuter router API that is not documented for now. But you can check this doc: https://openwrt.org/docs/techref/ubus
For the second point, I think you mean link between router and VPS, you can create a path manager in C using BPF or mptcpd on 6.1 kernel. I will add some examples and some new path manager/scheduler in next kernel 6.6 release.

Yessine1200, 2024-02-15:
Hello Ysurac,
Regarding the second point, I'm inquiring whether it's possible to programmatically select an alternate link in case of link loss, rather than using a graphical interface like the one provided. Additionally, I have a few more questions, don't hate me :)
In the WAN interface configuration, there are three options for MPTCP (enable, disable, master and backup). Are these options related to selecting the path when a link is lost? Also, if I want to test Glorytun UDP, how can I configure it?

Ysurac, 2024-02-15:
When a link loss an alternate link is automatically used if multipath is enabled on it.
Master is the connection used for initial connection, enable add a connection to multipath "pool", disable remove it, and backup only use the connection when enabled connection are not availables.
Glorytun UDP is available in the advanced checkbox of the wizard.

Yessine1200, 2024-02-15:
When using Glorytun UDP, I can indeed achieve similar multipath functionality as with MPTCP(choosing master or enabled...) . However, with Glorytun UDP, I don't need a proxy like Shadowsocks because Glorytun UDP itself provides the necessary tunneling and encryption for any traffic, right ?

Ysurac, 2024-02-15:
Yes but Glorytun UDP is slower than a proxy and doesn't work correctly in some case.

Yessine1200, 2024-02-15:
So, do you recommend using MPTCP over Glorytun UDP? Because, when using MPTCP, we need to employ other technologies to support UDP traffic, which can potentially introduce bugs in some cases. However, with Glorytun UDP, we can handle both TCP and UDP traffic.

Ysurac, 2024-02-15:
As one of the goal of aggregation is to have higher speed, it's why I use a socks proxy for TCP part. For UDP part, glorytun UDP is a choice, there is several others (even to use V2Ray or XRay that do UDP over TCP).
You can use any available proxy (or disable it and traffic will use VPN) and any available VPN (or disable it and traffic will use load balancing for traffic that doesn't use the proxy).

Yessine1200, 2024-02-16:
Hello Ysurac,
I hope you're doing well. I wanted to share some updates with you regarding my recent testing of OMR. Last night, I spent the entire night testing it out.
Here's what I did: I installed OMR on a virtual machine running on my Mac. I have three routers available (4G Orange, 4G SFR, and 5G Orange), and I configured them all to be on the same LAN as OMR. Since I don't have a switch, I connected all the routers using RJ45 cables and created a bridge to ensure they're on the same LAN. Additionally, I designated the 5G router as the master, as it tends to be more stable compared to the LTE routers. I left all other configurations at their default settings.
Now, I'm trying to understand how OMR operates. If I run a speed test from the internet, will the TCP traffic be distributed equally across the three links, or will it be based on the performance of each link?
Looking forward to your insights.
[image]

Ysurac, 2024-02-16:
I don't understand why you set a bridge. Ideally all WAN should have their own subnet.
Usage is based on latencies: lower latency first.

Yessine1200, 2024-02-16:
To ensure proper functioning, it's recommended to have all my routers on the same LAN, right ?. Since I don't have a switch available, I opted for this alternative solution. You have any other ideas ?
So, does OMR prioritize distributing most of the traffic over the link with the lowest latency?

Ysurac, 2024-02-16:
No, routers shouldn't be on same LAN/subnet. They only need to be reachable from OMR. Add one interface by router on the VM and put a each on dedicated network/subnet.
Yes, OMR prioritize traffic on the lowest latency link.

Ysurac, 2024-02-16:
As it's a VM, use normal on each dedicated real interface. You can change it in the wizard or via Network->Interfaces.

Yessine1200, 2024-02-16:
After implementing the steps you provided, I encountered the following issues:
1. Connectivity Problem: The OMR successfully reaches the 4G router, but it cannot establish connections with the other two routers.
2. Incorrect Server Selection: Additionally, the OMR seems to be selecting a server based on the WAN IP address of the 4G router instead of connecting to my designated VPS.
[image]

Ysurac, 2024-02-16:
"Direct output" mean that it don't use VPS to exit (because it can't contact VPS), so it dispay the current public IP used as fallback.
You need to make sure SFR box and the 5G box are available on IP 192.168.3.1 and 192.168.2.1, if yes check that eth2 and eth3 interface are correctly configured in your hypervisor.

Yessine1200, 2024-02-16:
Yes, all the configurations are correct, and all the interfaces are added to my hypervisor. However, I've realized something: my laptop's configuration is connected via WiFi to the 4G router, not directly to OMR. That's why OMR can reach the 4G router (I tried connecting to another box, and OMR could reach it). So, I need to connect my laptop's configuration directly to OMR, but I'm unsure how to do that because OMR is on a VM.

Yessine1200, 2024-02-16:
It seems I'm limited by the available Ethernet ports to assign each interface to its own router LAN. So, I'll need a switch or stick to my previous method of using bridge mode with MacVLAN to create virtual interfaces. However, I'm concerned about potential congestion with this approach.

Ysurac, 2024-02-16:
I don't understand your configuration... The modems are connected to what ?

Yessine1200, 2024-02-16:
For the initial configuration, I placed all three modems on the same network as OMR:
Router 4G Orange: 192.168.100.2, subnet: 255.255.255.0
Router 4G SFR: 192.168.100.4, subnet: 255.255.255.0
Router 5G: 92.168.100.3, subnet: 255.255.255.0
OMR: 192.168.100.1, subnet: 255.255.255.0
Then, I connected all three modems together. However, it didn't seem ideal to have the routers on the same network.

So, I made a change to the configuration:
Router 4G Orange: 192.168.1.1, subnet: 255.255.255.0
Router 4G SFR: 192.168.3.1, subnet: 255.255.255.0
Router 5G: 92.168.2.1, subnet: 255.255.255.0
OMR: 192.168.100.1, subnet: 255.255.255.0
After this change, the issue persisted: OMR could only reach the Router 4G Orange, despite verifying that all interfaces were available.

Ysurac, 2024-02-16:
How did you connect modem together without any switch and without enough ethernet ports ?

Yessine1200, 2024-02-16:
For the first configuration, I connected each router's LAN interface to another router using Ethernet cables.

Ysurac, 2024-02-16:
There is often a switch integrated to modems/box, this can be used. Else you need a switch or configure the router as a switch.

Yessine1200, 2024-02-18:
> Usage is based on latencies: lower latency first

Hello again Ysurac,
At this juncture, am I able to alter this policy to prioritize distributing traffic according to the highest speed?

Ysurac, 2024-02-18:
No. Or you need to develop a BPF or mptcpd module.

Ysurac, 2024-02-19:
You need to clone this repo and the feeds, develop the module in C, create a package, add it to the image and compile it.

Yessine1200, 2024-02-19:
To accomplish this task, do I need a specific operating system? How exactly do I create and add it to the image? If there is documentation available, could you share it with me?

Ysurac, 2024-02-19:
You need to know how to develop using C language, and for BPF modules you need to learn how these work, for mptcpd you need to follow examples.
There is no specific documentation.

Yessine1200, 2024-02-19:
So, if I understand correctly, I should install the [OMR kernel 6.1](https://download.openmptcprouter.com/release/v0.60beta2-6.1/) on a separate hardware because it supports mptcpd. Then, I make all the changes on it, and after that, I compile the image.

Yessine1200, 2024-02-23:
Hello again Ysurac,
I'm interested in understanding how OMR automatically detects the MTU. Which algorithms does it utilize for this purpose?

Ysurac, 2024-02-23:
ICMP with ```ping -M do -s <packetsize>```

Yessine1200, 2024-02-26:
Hello Ysurac,
Thank you very much for always answering my questions. I'm aware that some of my inquiries might seem basic, but I'm genuinely trying to grasp the concepts behind OMR, as it appears to be a promising project. I'd like to delve deeper into understanding it. I've learned that MPTCP is used to automatically detect potential links for aggregation and to qualify the links for optimal data traffic routing. Here's my question: If one of the links becomes overloaded, resulting in slower speeds, what measures are taken to mitigate this issue?

closed 2024-05-31 by github-actions[bot]