Ysurac/openmptcprouter #3161: Is possible send AT-command to usb 4g dongles on OMR?

R-sajadi, 2024-02-12:
I use raspi with 4  usb 4g modems. I would like to have the AT commands for each of the modems

Ysurac, 2024-02-12:
On latest snapshots you can use ```mmcli --command="AT+..."```, on previous release you will need to add ```--debug``` at the end of the line with ```procd_set_param command``` in ```/etc/init.d/modemmanager```.

R-sajadi, 2024-02-12:
@Ysurac 
Thank you, I will test and report the result

R-sajadi, 2024-02-13:
I download this one openmptcprouter-v0.59.1-5.4-r0+16594-ce92de8c8c-bcm27xx-bcm2711-rpi-4-ext4-factory.img.gz (2022/08/29 - 79.50Mo sha256sum)
and I use 4g modem (E3372s-153) as WAN1 So that I can connect to the Internet
but when  I check modem number with mmcli -L
output is: 
`root@OpenMPTCProuter:~# mmcli -L`
No modems were found
`root@OpenMPTCProuter:#`
Please tell me why??

Ysurac, 2024-02-13:
You configured it in the interface to use ModemManager ?
The model is not a patched to hilink mode version (so doesn't give directly usb*/eth* interfaces with IP) ? If it's the case, then no idea if it's possible.

R-sajadi, 2024-02-13:
I used web interface --->Network/Interfaces/WAN1 
I did not use ModemManager .
The modem is identified as eth1
> root@OpenMPTCProuter:~# ifconfig
> eth0     
 Link encap:Ethernet  HWaddr DC:A6:32:5D:21:F2  
>           inet addr:192.168.100.1  Bcast:192.168.100.255  Mask:255.255.255.0
>           inet6 addr: fe80::dea6:32ff:fe5d:21f2/64 Scope:Link
>           UP BROADCAST RUNNING PROMISC MULTICAST  MTU:1500  Metric:1
>           RX packets:3122 errors:0 dropped:7 overruns:0 frame:0
>           TX packets:3798 errors:0 dropped:0 overruns:0 carrier:0
>           collisions:0 txqueuelen:1000 
>           RX bytes:408100 (398.5 KiB)  TX bytes:2708409 (2.5 MiB)
> 
> eth1      
Link encap:Ethernet  HWaddr 0C:5B:8F:27:9A:64  
>           inet addr:192.168.8.100  Bcast:192.168.8.255  Mask:255.255.255.0
>           UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
>           RX packets:2523 errors:0 dropped:1 overruns:0 frame:0
>           TX packets:1822 errors:0 dropped:0 overruns:0 carrier:0
>           collisions:0 txqueuelen:100 
>           RX bytes:653210 (637.9 KiB)  TX bytes:166672 (162.7 KiB)
> 
> lo        
Link encap:Local Loopback  
>           inet addr:127.0.0.1  Mask:255.0.0.0
>           inet6 addr: ::1/128 Scope:Host
>           UP LOOPBACK RUNNING  MTU:65536  Metric:1
>           RX packets:10418 errors:0 dropped:0 overruns:0 frame:0
>           TX packets:10418 errors:0 dropped:0 overruns:0 carrier:0
>           collisions:0 txqueuelen:1000 
>           RX bytes:628148 (613.4 KiB)  TX bytes:628148 (613.4 KiB)
> 
> wan2      Link encap:Ethernet  HWaddr 12:19:6F:B4:12:34  
>           UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
>           RX packets:0 errors:0 dropped:0 overruns:0 frame:0
>           TX packets:8 errors:0 dropped:0 overruns:0 carrier:0
>           collisions:0 txqueuelen:20 
>           RX bytes:0 (0.0 B)  TX bytes:856 (856.0 B)

Ysurac, 2024-02-13:
So it's using Hilink mode specific to Huawei. No idea if it's possible to send AT commands.

R-sajadi, 2024-02-13:
Do I need to configure my modem in ModemManager?

Ysurac, 2024-02-13:
I think it require to flash the firmware to switch to the stick version. I don't know more...

R-sajadi, 2024-02-13:
I changed the modem to D-Link DWM-157 and
```
root@OpenMPTCProuter:~# mmcli -L
    /org/freedesktop/ModemManager1/Modem/1 [D-Link,Inc] D-Link DWM-157
```
but 
```
root@OpenMPTCProuter:~# mmcli -m 1 --command="AT"
error: command failed: 'GDBus.Error:org.freedesktop.ModemManager1.Error.Core.Unauthorized: Cannot send AT command to modem: operation only allowed in debug mode'
```

Ysurac, 2024-02-13:
Yes, it's what I said before:
`On latest snapshots you can use mmcli -m 0 --command="AT+..." (check modem number with mmcli -L), on previous release you will need to add --debug at the end of the line with procd_set_param command in /etc/init.d/modemmanager.
`

R-sajadi, 2024-02-13:
This is true?
```
. /usr/share/ModemManager/modemmanager.common
        procd_open_instance
        procd_set_param command /usr/sbin/ModemManager-wrapper --debug
        procd_set_param respawn "${respawn_threshold:-3600}" "${respawn_timeout:-5}" "${respawn_retry:-5}"
        procd_set_param pidfile "${MODEMMANAGER_PID_FILE}"
        procd_close_instance
```

R-sajadi, 2024-02-13:
@Ysurac 
Thank you very much for your time, it was ok

R-sajadi, 2024-02-13:
Is this modem added to the list of devices as teql?
[image]

Ysurac, 2024-02-13:
No, teql0 is a virtual interface.
To use this modem you have to change the protocol to ModemManager and then select the device in the list.

closed 2024-05-19 by github-actions[bot]