Ysurac/openmptcprouter #3162: Full speed of switch LAG not recognized by OMR

misterbond007, 2024-02-13:
Hi there! I run OMR on a virtual machine on my NAS. I have created to network interfaces in my Virtual Machine Manager on my NAS. The NAS is connected to the switch via LAG (2 ports, each 1000 MBit/s, so 2000 MBit/s). The Virtual Machine Manager shows, that each created network has the full speed of my LAG (2000 MBit/s). 

In OMR, there are eth0 (VLAN 1), eth1 (VLAN 4) and bond. "bond" has no traffic. It seems that eth0 and eth1 have only 1000 MBit/s each. How can I set the full 2000 MBit/s speed in OMR?

## Current Behavior

## Specifications

  - OpenMPTCProuter version: 
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform:

Ysurac, 2024-02-14:
You can check via SSH on the router with ```ethtool``` the Speed configuration and available mode of an interface.

misterbond007, 2024-02-16:
I receive the following information with ethtool eth0:
Settings for eth0:
Supported ports: [ ]
Supported link modes: Not reported
Supported pause frame use: No
Supports auto-negotiation: No
Supported FEC modes: Not reported
Advertised link modes: Not reported
Advertised pause frame use: No
Advertised auto-negotiation: No
Advertised FEC modes: Not reported
Speed: Unknown!
Duplex: Unknown! (255)
Port: Other
PHYAD: 0
Transceiver. internal
Auto-negotiation: off
Link detected: yes

Ysurac, 2024-02-17:
You can check using ```dmesg``` command or "Status->Kernel log" that NICs are correctly detected with some drivers.

misterbond007, 2024-02-19:
I only find one row with NIC: [    0.439268] VMware vmxnet3 virtual NIC driver - version 1.4.17.0-k-NAPI
The Kernel log is enclosed:
[kernellog.txt](https://github.com/Ysurac/openmptcprouter/files/14336907/kernellog.txt)

closed 2024-05-25 by github-actions[bot]