Ysurac/openmptcprouter #3163: Issues with PPPOE connection taking out whole internet, follows high ping

mr337, 2024-02-13:
I have two internet connections. Startlink, need no explanation, and a wireless P2P that uses PPPOE.

[image]

The P2P PPPOE has a little hiccup, can't ping the VPS, and when is ready will reconnect and rebond the connection. Just like a normal hiccup.
  

## Current Behavior

This P2P link isn't the best and at times goes out. The connection type is PPPOE. When the PPPOE link has any issues including a hiccup with the pings my entire internet stops even though Starlink is just fine. The internet connection drops for 5-10 seconds and then resume. Sometimes I don't get a full network drop, but a high latency. When the PPPOE connection resumes my latencies from from around 100ms and climb into the 300-400ms. The only thing that seems to take care of that is a reboot of the OpenMPTCP Router seems to clear it.

I have a few logs that are interesting when this happens, it looks like the deleting and the recreating of the PPPOE connection could be the cause of it. I did not have this problem until the P2P ISP moved to PPPOE. I'll post more logs as I should have copied them before the restart.

```
Tue Feb 13 16:58:46 2024 daemon.notice netifd: wan2 (11070): udhcpc: sending renew to 100.64.0.1
Tue Feb 13 16:58:46 2024 daemon.notice netifd: wan2 (11070): udhcpc: lease of 100.68.144.99 obtained, lease time 300
Tue Feb 13 17:01:16 2024 daemon.notice netifd: wan2 (11070): udhcpc: sending renew to 100.64.0.1
Tue Feb 13 17:01:16 2024 daemon.notice netifd: wan2 (11070): udhcpc: lease of 100.68.144.99 obtained, lease time 300
Tue Feb 13 17:01:25 2024 daemon.err glorytun[21606]: read: Connection reset by peer
Tue Feb 13 17:01:25 2024 daemon.info glorytun[21606]: STOPPED tun0
Tue Feb 13 17:01:25 2024 daemon.err glorytun[21606]: 216.238.69.208.65001: connected
Tue Feb 13 17:01:25 2024 daemon.info glorytun[21606]: STARTED tun0
Tue Feb 13 17:01:30 2024 user.notice post-tracking-post-tracking: wan3 (pppoe-wan3) switched off because check error and ping from 192.168.34.30 error (8.8.8.8,80.67.169.12,8.8.4.4)
Tue Feb 13 17:01:30 2024 user.notice post-tracking-post-tracking: Delete default route to 216.238.69.208 via 192.168.34.1 dev pppoe-wan3
Tue Feb 13 17:01:39 2024 user.notice post-tracking-post-tracking: wan3 (pppoe-wan3) switched up
Tue Feb 13 17:01:39 2024 user.notice post-tracking-post-tracking: Interface route not yet set, set route ip r add default via 192.168.34.1 dev pppoe-wan3 metric 8
Tue Feb 13 17:03:46 2024 daemon.notice netifd: wan2 (11070): udhcpc: sending renew to 100.64.0.1
Tue Feb 13 17:03:47 2024 daemon.notice netifd: wan2 (11070): udhcpc: lease of 100.68.144.99 obtained, lease time 300
Tue Feb 13 17:06:17 2024 daemon.notice netifd: wan2 (11070): udhcpc: sending renew to 100.64.0.1
Tue Feb 13 17:06:17 2024 daemon.notice netifd: wan2 (11070): udhcpc: lease of 100.68.144.99 obtained, lease time 300
Tue Feb 13 17:08:47 2024 daemon.notice netifd: wan2 (11070): udhcpc: sending renew to 100.64.0.1
Tue Feb 13 17:08:47 2024 daemon.notice netifd: wan2 (11070): udhcpc: lease of 100.68.144.99 obtained, lease time 300
Tue Feb 13 17:11:17 2024 daemon.notice netifd: wan2 (11070): udhcpc: sending renew to 100.64.0.1
Tue Feb 13 17:11:17 2024 daemon.notice netifd: wan2 (11070): udhcpc: lease of 100.68.144.99 obtained, lease time 300
```
```
Tue Feb 13 20:43:26 2024 daemon.notice netifd: wan2 (10820): udhcpc: sending renew to 100.64.0.1
Tue Feb 13 20:43:26 2024 daemon.notice netifd: wan2 (10820): udhcpc: lease of 100.68.144.99 obtained, lease time 300
Tue Feb 13 20:44:18 2024 user.notice post-tracking-post-tracking: Check API configuration...
Tue Feb 13 20:44:23 2024 user.notice post-tracking-post-tracking: Check API configuration... Done
Tue Feb 13 20:45:56 2024 daemon.notice netifd: wan2 (10820): udhcpc: sending renew to 100.64.0.1
Tue Feb 13 20:45:56 2024 daemon.notice netifd: wan2 (10820): udhcpc: lease of 100.68.144.99 obtained, lease time 300
Tue Feb 13 20:46:23 2024 user.notice post-tracking-post-tracking: wan3 (pppoe-wan3) switched off because check error and ping from 192.168.34.30 error (1.2.4.8,80.67.169.40,114.114.114.114)
Tue Feb 13 20:46:23 2024 user.notice post-tracking-post-tracking: Delete default route to 216.238.69.208 via 192.168.34.1 dev pppoe-wan3
Tue Feb 13 20:46:30 2024 user.notice post-tracking-post-tracking: wan3 (pppoe-wan3) switched up
Tue Feb 13 20:46:30 2024 user.notice post-tracking-post-tracking: Interface route not yet set, set route ip r add default via 192.168.34.1 dev pppoe-wan3 metric 8
Tue Feb 13 20:48:26 2024 daemon.notice netifd: wan2 (10820): udhcpc: sending renew to 100.64.0.1
Tue Feb 13 20:48:26 2024 daemon.notice netifd: wan2 (10820): udhcpc: lease of 100.68.144.99 obtained, lease time 300
```

From what I have read and a _hunch_ from other logs it feels like the pppoe-wan3 is an interface, that when a ping check fails the interface is remove/deleted and then recreated. This feels much more different than just a spotty connection vs moving interfaces around.

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.59.1-5.4 r0+16594-ce92de8c8c / LuCI openwrt-21.02 branch git-22.119.37126-a993714
  - OpenMPTCProuter VPS version: VPS 0.1028 5.4.207-mtcp
  - OpenMPTCProuter VPS provider: Vultr
  - OpenMPTCProuter platform: x86_64

Ysurac, 2024-02-14:
How do you check latency for WAN3 ? On status page ?
When an interface doesn't answer correctly, default routes using this interface are removed and added back.

mr337, 2024-02-14:
> How do you check latency for WAN3 ? On status page ? When an interface doesn't answer correctly, default routes using this interface are removed and added back.

Correct, just on the status page. I have been thinking of somehow exporting some of these metrics but don't know of a good method. There are [some options](https://grafana.com/blog/2021/02/09/how-i-monitor-my-openwrt-router-with-grafana-cloud-and-prometheus/) for a [local time series database](https://github.com/benisai/Openwrt-Monitoring) to store these kind of metrics. 

The pppoe-wan3 seems like a virtual interface that is tied to WAN3.

[image]

As part of the interface removal and adding, that process should not impact the overall internet connection, correct? For other setups in the past that were **non** PPPOE when an interface has ping issues I had no problems. It seems adding the PPPOE setup to the configuration started to showing this odd behavior. I am not for sure if tweaking the ping checks or maybe getting a dedicated PPPOE modem would be better (if these are the right settings).

[image]

I'll monitor it and post more logs when I do get a failed internet again, it normally happens at least once a day.

Ysurac, 2024-02-14:
These settings are only related to the Status page display, doesn't change anything on operation.
What is the result of ```ip r``` via SSH on the router when it's working correctly and when it's not the case ?

mr337, 2024-02-15:
And of course after posting this I have no issues. :laughing: 

I'll close this and if the issue persists will re-open. Thanks!

closed 2024-02-15 by mr337