Ysurac/openmptcprouter #3166: DNS issue while using tmate.io

iamdivyanshverma, 2024-02-15:
## Current Behavior
root@OpenMPTCPRouter:/etc/config# tmate -F
Error in auto_init()
To connect to the session locally, run: tmate -S /tmp/tmate-0/dkAMHn attach
fatal: Cannot initialize the DNS lookup service

******************************************

suppose if i add a custom dns server in the file then the result is 
root@OpenMPTCPRouter:/etc/config# tmate -F
Error in auto_init()
To connect to the session locally, run: tmate -S /tmp/tmate-0/nObDGe attach
Connecting to ssh.tmate.io...
Segmentation fault

******************************************

root@OpenMPTCPRouter:/etc/config# ifconfig
br-lan    Link encap:Ethernet  HWaddr E4:5F:01:A9:23:47  
          inet addr:192.168.100.1  Bcast:192.168.100.255  Mask:255.255.255.0
          inet6 addr: fe80::e65f:1ff:fea9:2347/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:21960 errors:0 dropped:0 overruns:0 frame:0
          TX packets:21321 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:3366452 (3.2 MiB)  TX bytes:18360969 (17.5 MiB)

eth0      Link encap:Ethernet  HWaddr E4:5F:01:A9:23:47  
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:15189 errors:0 dropped:0 overruns:0 frame:0
          TX packets:19252 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:1211058 (1.1 MiB)  TX bytes:16168281 (15.4 MiB)

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:9166 errors:0 dropped:0 overruns:0 frame:0
          TX packets:9166 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:541838 (529.1 KiB)  TX bytes:541838 (529.1 KiB)

phy0-ap0  Link encap:Ethernet  HWaddr E6:5F:01:A9:23:48  
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:6779 errors:0 dropped:0 overruns:0 frame:0
          TX packets:7489 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:2000 
          RX bytes:2368456 (2.2 MiB)  TX bytes:2585251 (2.4 MiB)

wwan0     Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  
          inet addr:100.114.194.220  P-t-P:100.114.194.220  Mask:255.255.255.248
          UP POINTOPOINT RUNNING NOARP MULTICAST  MTU:1500  Metric:1
          RX packets:6415 errors:0 dropped:0 overruns:0 frame:0
          TX packets:10075 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:945896 (923.7 KiB)  TX bytes:2029575 (1.9 MiB)

wwan1     Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  
          inet addr:10.23.152.9  P-t-P:10.23.152.9  Mask:255.255.255.252
          UP POINTOPOINT RUNNING NOARP MULTICAST  MTU:1500  Metric:1
          RX packets:6108 errors:0 dropped:0 overruns:0 frame:0
          TX packets:7277 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:1017096 (993.2 KiB)  TX bytes:995265 (971.9 KiB)

wwan2     Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  
          inet addr:100.92.211.45  P-t-P:100.92.211.45  Mask:255.255.255.252
          UP POINTOPOINT RUNNING NOARP MULTICAST  MTU:1500  Metric:1
          RX packets:8529 errors:0 dropped:0 overruns:0 frame:0
          TX packets:9859 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:3748641 (3.5 MiB)  TX bytes:2348834 (2.2 MiB)

****************************************************************

root@OpenMPTCPRouter:/etc/config# route -n
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
0.0.0.0         100.92.211.46   0.0.0.0         UG    0      0        0 wwan2
0.0.0.0         100.114.194.221 0.0.0.0         UG    1      0        0 wwan0
0.0.0.0         100.114.194.221 0.0.0.0         UG    10     0        0 wwan0
0.0.0.0         10.23.152.10    0.0.0.0         UG    11     0        0 wwan1
0.0.0.0         100.92.211.46   0.0.0.0         UG    12     0        0 wwan2
10.23.152.8     0.0.0.0         255.255.255.252 U     11     0        0 wwan1
100.92.211.44   0.0.0.0         255.255.255.252 U     12     0        0 wwan2
100.114.194.216 0.0.0.0         255.255.255.248 U     10     0        0 wwan0
127.0.0.0       0.0.0.0         255.0.0.0       U     8      0        0 lo

******************************************************************

root@OpenMPTCPRouter:/etc/config# cat /etc/resolv.conf

*******************************************************************

## Steps to Reproduce the Problem

  1.opkg update
  2. opkg install tmate
  3. tmate -F

## Specifications

  - OpenMPTCProuter version: v0.60beta1-6.1
  - OpenMPTCProuter VPS version: OpenMPTCProuter VPS 0.1029-test
  - OpenMPTCProuter VPS provider: Digital Ocean
  - OpenMPTCProuter platform: Rpi4
  - Country:

Ysurac, 2024-02-17:
tmate seems to be looking for a nameserver in /etc/resolv.conf, you can add this:
```
search lan
nameserver 127.0.0.1
nameserver ::1
```

iamdivyanshverma, 2024-02-17:
Hi @Ysurac , Thanks for the reply.
Do i need to restart dhsmasq or something else because I tried with the above you mentioned from openwrt dns guide, but no success.
I get segmentation fault everytime .

root@OpenMPTCPRouter:~# tmate -F
Error in auto_init()
To connect to the session locally, run: tmate -S /tmp/tmate-0/KJBLcA attach
Connecting to ssh.tmate.io...
Segmentation fault

Ysurac, 2024-02-17:
No need to restart.
It's a new error, but no idea here... Maybe a problem when connecting to ssh.tmate.io

iamdivyanshverma, 2024-02-17:
I tried with my server where i am hosting tmate-ssh-server. still gives the same error.

root@ZifiLink:~# tmate -F
Error in auto_init()
To connect to the session locally, run: tmate -S /tmp/tmate-0/lnHcEm attach
Connecting to 143.110.250.189...
Segmentation fault

iamdivyanshverma, 2024-02-17:
I have a similar device with OpenWrt 23 . There tmate works just fine.
found one thing in dhcp file 
        
        option noresolv '1'
        
   when i check on openwrt , found something related with split dns , but not able to understand why it is used here.

Ysurac, 2024-02-17:
I don't use tmate, I can't really help, and the "Segmentation fault" doesn't really help to debug here. Maybe there is a debug or verbose mode that can help ?

iamdivyanshverma, 2024-02-18:
there is a verbose option , but the log isn't helping much..!!

root@ZifiLink:~# tmate -F -vv
Error in auto_init()
client started (15165): socket /tmp/tmate-0/dPMoCL, protocol 8
on Linux 6.1.55 #0 SMP PREEMPT_DYNAMIC Mon Sep 25 21:21:09 2023; libevent 2.1.12-stable (epoll)
socket is /tmp/tmate-0/dPMoCL
trying connect
connect failed: No such file or directory
lock file is /tmp/tmate-0/dPMoCL.lock
flock succeeded
got lock (6)
trying connect
connect failed: No such file or directory
server started (15165): socket /tmp/tmate-0/dPMoCL, protocol 8
on Linux 6.1.55 #0 SMP PREEMPT_DYNAMIC Mon Sep 25 21:21:09 2023; libevent 2.1.12-stable (epoll)
continuing cmdq 0x7f5a8f7a6aa0: flags 0, client 0
cmdq 0x7f5a8f7a6aa0: bind-key C-b send-prefix
preparing state for bind-key C-b send-prefix (client 0)
cmd_find_client: no target, return 0
preparing -t state: target none
preparing -s state: target none
continuing cmdq 0x7f5a8f7a6aa0: flags 0, client 0
cmdq 0x7f5a8f7a6aa0: bind-key C-o rotate-window
preparing state for bind-key C-o rotate-window (client 0)
cmd_find_client: no target, return 0
preparing -t state: target none
preparing -s state: target none
continuing cmdq 0x7f5a8f7a6aa0: flags 0, client 0
cmdq 0x7f5a8f7a6aa0: bind-key C-z suspend-client
preparing state for bind-key C-z suspend-client (client 0)
cmd_find_client: no target, return 0
preparing -t state: target none
preparing -s state: target none
continuing cmdq 0x7f5a8f7a6aa0: flags 0, client 0
cmdq 0x7f5a8f7a6aa0: bind-key Space next-layout
preparing state for bind-key Space next-layout (client 0)
cmd_find_client: no target, return 0
preparing -t state: target none
preparing -s state: target none
continuing cmdq 0x7f5a8f7a6aa0: flags 0, client 0
cmdq 0x7f5a8f7a6aa0: bind-key ! break-pane
preparing state for bind-key ! break-pane (client 0)
cmd_find_client: no target, return 0
preparing -t state: target none
preparing -s state: target none
continuing cmdq 0x7f5a8f7a6aa0: flags 0, client 0
cmdq 0x7f5a8f7a6aa0: bind-key " split-window
preparing state for bind-key " split-window (client 0)
cmd_find_client: no target, return 0
preparing -t state: target none
preparing -s state: target none
continuing cmdq 0x7f5a8f7a6aa0: flags 0, client 0
cmdq 0x7f5a8f7a6aa0: bind-key # list-buffers
preparing state for bind-key # list-buffers (client 0)
cmd_find_client: no target, return 0
preparing -t state: target none
preparing -s state: target none
continuing cmdq 0x7f5a8f7a6aa0: flags 0, client 0
cmdq 0x7f5a8f7a6aa0: bind-key $ command-prompt -I#S "rename-session '%%'"
preparing state for bind-key $ command-prompt -I#S "rename-session '%%'" (client 0)
cmd_find_client: no target, return 0
preparing -t state: target none
preparing -s state: target none
continuing cmdq 0x7f5a8f7a6aa0: flags 0, client 0
cmdq 0x7f5a8f7a6aa0: bind-key % split-window -h
preparing state for bind-key % split-window -h (client 0)
cmd_find_client: no target, return 0
preparing -t state: target none
preparing -s state: target none
[... 1060 lines cut ...]
[2] [ssh_connect] ssh_connect: libssh 0.10.4 (c) 2003-2022 Aris Adamantiadis, Andreas Schneider and libssh contributors. Distributed under the LGPL, please refer to COPYING file for information about your rights, using threading threads_pthread
[3] [getai] getai: host 134.209.128.51 matches an IP address
[2] [ssh_socket_connect] ssh_socket_connect: Nonblocking connection socket: 7
[2] [ssh_connect] ssh_connect: Socket connecting, now waiting for the callbacks to work
[3] [ssh_connect] ssh_connect: current state : 1
[2] [ssh_connect] ssh_connect: libssh 0.10.4 (c) 2003-2022 Aris Adamantiadis, Andreas Schneider and libssh contributors. Distributed under the LGPL, please refer to COPYING file for information about your rights, using threading threads_pthread
[3] [getai] getai: host 157.230.72.130 matches an IP address
[2] [ssh_socket_connect] ssh_socket_connect: Nonblocking connection socket: 9
[2] [ssh_connect] ssh_connect: Socket connecting, now waiting for the callbacks to work
[3] [ssh_connect] ssh_connect: current state : 1
[2] [ssh_connect] ssh_connect: libssh 0.10.4 (c) 2003-2022 Aris Adamantiadis, Andreas Schneider and libssh contributors. Distributed under the LGPL, please refer to COPYING file for information about your rights, using threading threads_pthread
[3] [getai] getai: host 206.189.246.93 matches an IP address
[2] [ssh_socket_connect] ssh_socket_connect: Nonblocking connection socket: 10
[2] [ssh_connect] ssh_connect: Socket connecting, now waiting for the callbacks to work
[3] [ssh_connect] ssh_connect: current state : 1
@0 active pane not changed
[3] [ssh_socket_pollcallback] ssh_socket_pollcallback: Received POLLOUT in connecting state
[1] [socket_callback_connected] socket_callback_connected: Socket connection callback: 1 (0)
[3] [ssh_socket_unbuffered_write] ssh_socket_unbuffered_write: Enabling POLLOUT for socket
[3] [ssh_connect] ssh_connect: current state : 2
@0 active pane not changed
[3] [callback_receive_banner] callback_receive_banner: Received banner: SSH-2.0-tmate
[2] [ssh_client_connection_callback] ssh_client_connection_callback: SSH server banner: SSH-2.0-tmate
[2] [ssh_analyze_banner] ssh_analyze_banner: Analyzing banner: SSH-2.0-tmate
Segmentation fault

closed 2024-05-23 by github-actions[bot]