Ysurac/openmptcprouter #3168: Error installing beta 2

patimages, 2024-02-16:
Hi, 

I am getting this error when installing the beta 2 of OMR 60: 

Err:7 http://deb.debian.org/debian stretch-backports Release
  404  Not Found [IP: XXXX]
Err:4 http://deb.debian.org/debian buster-backports InRelease
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY XXXX NO_PUBKEY XXXX
Reading package lists... Done
E: The repository 'http://deb.debian.org/debian stretch-backports Release' no longer has a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: http://deb.debian.org/debian buster-backports InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY XXX NO_PUBKEY XXX

Any idea ? 

Many thanks !

Ysurac, 2024-02-17:
What are the XXXX ?
What is the version installed ? 5.4 or 6.1 ?
What is the Debian release version ?

patimages, 2024-02-17:
I'll get back concerning this but right now, I am trying to get back online. I reinstalled an older version (59.1) and want to upload a backup and get "file not supported"... Is there a way to bypass this warning ?

Ysurac, 2024-02-17:
Can you take a screenshot ?

patimages, 2024-02-17:
[image]

patimages, 2024-02-17:
NAME="openmptcprouter"
VERSION="v0.59.1-5.4"
ID="openmptcprouter"
ID_LIKE="lede openwrt"
PRETTY_NAME="openmptcprouter v0.59.1-5.4"
VERSION_ID="v0.59.1-5.4"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r0+16594-ce92de8c8c"
OPENWRT_BOARD="x86/64"
OPENWRT_ARCH="x86_64"
OPENWRT_TAINTS="busybox override"
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="openmptcprouter v0.59.1-5.4 r0+16594-ce92de8c8c"

The backup is from the same version

Ysurac, 2024-02-17:
No idea where this can be coming from...
You can decompress the file and put ```etc/config/*``` in ```/etc/config/``` on the router.

patimages, 2024-02-18:
Ok, a short summary of the issues. 
- I tried to include a 2.5Gbe adapter to the NUC hosting OMR. It is only supported in OMR 60 beta (5.4). To do that, I had to use the Eth1 and I changed Eth0 to Eth1 exclusively in the wizard. All worked but I lost external access. Any idea why ? 
- Then I migrated to beta 2 (5.4) and mega issue with the VPS update 'NO_PUBKEY 0E98404D386FA1D9 NO_PUBKEY 6ED0E7B82643E1'. 
- I wanted to come back to beta 1 (btw the link is wrong in the homepage -> points to beta 2) but the restore option showed "unsupported file" when using the proper backup (made few minutes ago).
- I came back to 0.59 for now but lost 2.5Gbe speed.
Any idea how to go to beta, use Eth1 and allow for external access ?

Ysurac, 2024-02-18:
* Where did you make changes exactly ? Did you also check if all is ok in Network->Interfaces ?
* I need more info for the NO_PUBKEY as it's about default Debian repo. What is the Debian version used ? It's updated from an old release (maybe some old repo that should be here anymore...) ?
* Beta 1 no longer exist. Did you try to decompress the backup ?

patimages, 2024-02-18:
Where did you make changes exactly ? Did you also check if all is ok in Network->Interfaces ?
-> Changed the physical interfaces to Eth1 for lan et wans : did not go elsewhere, should I ? 

I need more info for the NO_PUBKEY as it's about default Debian repo.
-> It was initially based on debian 10. Worked great for years and many updates. Here is the error : 
An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: http://deb.debian.org/debian buster-backports InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 0E98404D386FA1D9 NO_PUBKEY 6ED0E7B82643E131

-> I actually found beta 1 and decompressed the file but gave up as it was looking a bit like "une usine à gaz" ;-D ! 

Thanks again !

patimages, 2024-02-18:
1/ OK, I ll try all that asap and get back then ! Thanks again for the weekend support ! 
2/ Concerning the stretch-backport come from Debian 9 -> I did nothing weird : just wget -O - http://www.openmptcprouter.com/server-beta/debian.sh | sh ... The script blocked with such error and no further progress. Not sure how to "remove" this issue. 
3/ I tried to copy the content of etc/config/ but wondered about the rest of the folders ? Not required ?

Ysurac, 2024-02-18:
2) ```rm /etc/apt/sources.list.d/stretch-backports.list``` and ```apt-get --allow-unauthenticated install debian-archive-keyring```
3) Not required for the configuration

patimages, 2024-02-18:
Ok, I ll try all that and report. Thanks !

patimages, 2024-02-18:
> 2. `rm /etc/apt/sources.list.d/stretch-backports.list` and `apt-get --allow-unauthenticated install debian-archive-keyring`

Done but still and error ...

Need to get 104 kB of archives.
After this operation, 273 kB of additional disk space will be used.
Get:1 http://security.debian.org/debian-security buster/updates/main amd64 libiperf0 amd64 3.6-2+deb10u1 [77.7 kB]
Get:2 http://security.debian.org/debian-security buster/updates/main amd64 iperf3 amd64 3.6-2+deb10u1 [26.1 kB]
Fetched 104 kB in 0s (896 kB/s)   
Selecting previously unselected package libiperf0:amd64.
(Reading database ... 296542 files and directories currently installed.)
Preparing to unpack .../libiperf0_3.6-2+deb10u1_amd64.deb ...
Unpacking libiperf0:amd64 (3.6-2+deb10u1) ...
Selecting previously unselected package iperf3.
Preparing to unpack .../iperf3_3.6-2+deb10u1_amd64.deb ...
Unpacking iperf3 (3.6-2+deb10u1) ...
Setting up libiperf0:amd64 (3.6-2+deb10u1) ...
Setting up iperf3 (3.6-2+deb10u1) ...
Processing triggers for man-db (2.8.5-2) ...
Processing triggers for libc-bin (2.28-10) ...
chown: invalid user: 'iperf3'
Failed to enable unit: Unit file iperf3.service does not exist.
-                                                      78%[==============================================================================================>                           ]  65.98K  4.95KB/s    in 13s     

Cannot write to '-' (Broken pipe).

Ysurac, 2024-02-19:
It's not the same error. Should be fixed now.

patimages, 2024-02-19:
> It's not the same error. Should be fixed now.

Indeed : all good now ! Thanks !

patimages, 2024-02-19:
In fact, my issues are linked to beta 2 ... I lose external access even on the classical Eth0 (I thought the issue came from going to Eth1 but not the case). When I try to connect from outside, I receive a "VPN tunnel down" error (see attached).
[image]

patimages, 2024-02-19:
I ll try. 
I use : openmptcprouter-v0.60beta2-5.4-r0+16862-170d9e447d-x86-64-generic-squashfs-combined.img.gz

patimages, 2024-02-19:
Not helping : still the same issue (VPN down):

Feb 19 16:53:23 OpenMPTCProuter user.notice post-tracking-020-status: New public ip detected for omrvpn (tun0): 10.255.255.2 (previous: 134.59.116.4)
Feb 19 16:53:23 OpenMPTCProuter user.notice post-tracking-020-status: Reload MPTCP for tun0
Feb 19 17:53:44 OpenMPTCProuter daemon.info dnsmasq-dhcp[19632]: DHCPREQUEST(eth0) 192.168.100.183 34:98:b5:da:2e:12 
Feb 19 17:53:44 OpenMPTCProuter daemon.info dnsmasq-dhcp[19632]: DHCPACK(eth0) 192.168.100.183 34:98:b5:da:2e:12 
Feb 19 17:53:44 OpenMPTCProuter daemon.info dnsmasq-dhcp[19632]: DHCPREQUEST(eth0) 192.168.100.183 34:98:b5:da:2e:12 
Feb 19 17:53:44 OpenMPTCProuter daemon.info dnsmasq-dhcp[19632]: DHCPACK(eth0) 192.168.100.183 34:98:b5:da:2e:12 
Feb 19 17:53:45 OpenMPTCProuter daemon.info dnsmasq-dhcp[19632]: DHCPREQUEST(eth0) 192.168.100.183 34:98:b5:da:2e:12 
Feb 19 17:53:45 OpenMPTCProuter daemon.info dnsmasq-dhcp[19632]: DHCPACK(eth0) 192.168.100.183 34:98:b5:da:2e:12 
Feb 19 17:53:45 OpenMPTCProuter daemon.info dnsmasq-dhcp[19632]: DHCPREQUEST(eth0) 192.168.100.183 34:98:b5:da:2e:12 
Feb 19 17:53:45 OpenMPTCProuter daemon.info dnsmasq-dhcp[19632]: DHCPACK(eth0) 192.168.100.183 34:98:b5:da:2e:12 
Feb 19 17:53:46 OpenMPTCProuter daemon.info dnsmasq-dhcp[19632]: DHCPREQUEST(eth0) 192.168.100.183 34:98:b5:da:2e:12 
Feb 19 17:53:46 OpenMPTCProuter daemon.info dnsmasq-dhcp[19632]: DHCPACK(eth0) 192.168.100.183 34:98:b5:da:2e:12 
Feb 19 17:53:46 OpenMPTCProuter daemon.info dnsmasq-dhcp[19632]: DHCPREQUEST(eth0) 192.168.100.183 34:98:b5:da:2e:12 
Feb 19 17:53:46 OpenMPTCProuter daemon.info dnsmasq-dhcp[19632]: DHCPACK(eth0) 192.168.100.183 34:98:b5:da:2e:12 
Feb 19 17:53:47 OpenMPTCProuter daemon.info dnsmasq-dhcp[19632]: DHCPREQUEST(eth0) 192.168.100.137 60:3e:5f:42:f8:f8 
Feb 19 17:53:47 OpenMPTCProuter daemon.info dnsmasq-dhcp[19632]: DHCPACK(eth0) 192.168.100.137 60:3e:5f:42:f8:f8 PCMac
Feb 19 17:53:47 OpenMPTCProuter daemon.info dnsmasq-dhcp[19632]: DHCPREQUEST(eth0) 192.168.100.183 34:98:b5:da:2e:12 
Feb 19 17:53:47 OpenMPTCProuter daemon.info dnsmasq-dhcp[19632]: DHCPACK(eth0) 192.168.100.183 34:98:b5:da:2e:12 
Feb 19 17:53:51 OpenMPTCProuter daemon.info dnsmasq-dhcp[19632]: DHCPREQUEST(eth0) 192.168.100.183 34:98:b5:da:2e:12 
Feb 19 17:53:51 OpenMPTCProuter daemon.info dnsmasq-dhcp[19632]: DHCPACK(eth0) 192.168.100.183 34:98:b5:da:2e:12 
Feb 19 17:53:53 OpenMPTCProuter daemon.info dnsmasq-dhcp[19632]: DHCPREQUEST(eth0) 192.168.100.183 34:98:b5:da:2e:12 
Feb 19 17:53:53 OpenMPTCProuter daemon.info dnsmasq-dhcp[19632]: DHCPACK(eth0) 192.168.100.183 34:98:b5:da:2e:12 
Feb 19 17:53:55 OpenMPTCProuter daemon.info dnsmasq-dhcp[19632]: DHCPREQUEST(eth0) 192.168.100.183 34:98:b5:da:2e:12 
Feb 19 17:53:55 OpenMPTCProuter daemon.info dnsmasq-dhcp[19632]: DHCPACK(eth0) 192.168.100.183 34:98:b5:da:2e:12 
Feb 19 17:53:57 OpenMPTCProuter daemon.info dnsmasq-dhcp[19632]: DHCPREQUEST(eth0) 192.168.100.183 34:98:b5:da:2e:12 
Feb 19 17:53:57 OpenMPTCProuter daemon.info dnsmasq-dhcp[19632]: DHCPACK(eth0) 192.168.100.183 34:98:b5:da:2e:12 
Feb 19 17:54:02 OpenMPTCProuter daemon.info dnsmasq-dhcp[19632]: DHCPREQUEST(eth0) 192.168.100.183 34:98:b5:da:2e:12 
Feb 19 17:54:02 OpenMPTCProuter daemon.info dnsmasq-dhcp[19632]: DHCPACK(eth0) 192.168.100.183 34:98:b5:da:2e:12 
Feb 19 17:54:04 OpenMPTCProuter daemon.info dnsmasq-dhcp[19632]: DHCPREQUEST(eth0) 192.168.100.183 34:98:b5:da:2e:12 
Feb 19 17:54:04 OpenMPTCProuter daemon.info dnsmasq-dhcp[19632]: DHCPACK(eth0) 192.168.100.183 34:98:b5:da:2e:12 
Feb 19 17:54:09 OpenMPTCProuter daemon.info dnsmasq-dhcp[19632]: DHCPREQUEST(eth0) 192.168.100.183 34:98:b5:da:2e:12 
Feb 19 17:54:09 OpenMPTCProuter daemon.info dnsmasq-dhcp[19632]: DHCPACK(eth0) 192.168.100.183 34:98:b5:da:2e:12 
Feb 19 17:54:21 OpenMPTCProuter daemon.info dnsmasq-dhcp[19632]: DHCPREQUEST(eth0) 192.168.100.183 34:98:b5:da:2e:12 
Feb 19 17:54:21 OpenMPTCProuter daemon.info dnsmasq-dhcp[19632]: DHCPACK(eth0) 192.168.100.183 34:98:b5:da:2e:12 
Feb 19 17:54:22 OpenMPTCProuter daemon.info dnsmasq-dhcp[19632]: DHCPDISCOVER(eth0) a8:51:ab:d5:11:29 
Feb 19 17:54:22 OpenMPTCProuter daemon.info dnsmasq-dhcp[19632]: DHCPOFFER(eth0) 192.168.100.103 a8:51:ab:d5:11:29 
Feb 19 17:54:23 OpenMPTCProuter daemon.info dnsmasq-dhcp[19632]: DHCPREQUEST(eth0) 192.168.100.103 a8:51:ab:d5:11:29 
Feb 19 17:54:23 OpenMPTCProuter daemon.info dnsmasq-dhcp[19632]: DHCPACK(eth0) 192.168.100.103 a8:51:ab:d5:11:29 Apple-TV-Haut
Feb 19 17:55:04 OpenMPTCProuter daemon.info dnsmasq-dhcp[19632]: DHCPREQUEST(eth0) 192.168.100.183 34:98:b5:da:2e:12 
Feb 19 17:55:04 OpenMPTCProuter daemon.info dnsmasq-dhcp[19632]: DHCPACK(eth0) 192.168.100.183 34:98:b5:da:2e:12 
Feb 19 17:55:05 OpenMPTCProuter daemon.info dnsmasq-dhcp[19632]: DHCPREQUEST(eth0) 192.168.100.183 34:98:b5:da:2e:12 
Feb 19 17:55:05 OpenMPTCProuter daemon.info dnsmasq-dhcp[19632]: DHCPACK(eth0) 192.168.100.183 34:98:b5:da:2e:12 
Feb 19 17:55:05 OpenMPTCProuter daemon.info dnsmasq-dhcp[19632]: DHCPREQUEST(eth0) 192.168.100.183 34:98:b5:da:2e:12 
Feb 19 17:55:05 OpenMPTCProuter daemon.info dnsmasq-dhcp[19632]: DHCPACK(eth0) 192.168.100.183 34:98:b5:da:2e:12 
Feb 19 17:55:05 OpenMPTCProuter daemon.info dnsmasq-dhcp[19632]: DHCPREQUEST(eth0) 192.168.100.183 34:98:b5:da:2e:12 
Feb 19 17:55:05 OpenMPTCProuter daemon.info dnsmasq-dhcp[19632]: DHCPACK(eth0) 192.168.100.183 34:98:b5:da:2e:12 
[... cut ...]
Feb 19 17:01:06 OpenMPTCProuter daemon.info glorytun[16319]: STOPPED tun0
Feb 19 17:01:07 OpenMPTCProuter daemon.err glorytun[16319]: 134.59.116.4.65001: connected
Feb 19 17:01:07 OpenMPTCProuter user.notice post-tracking-001-post-tracking: omrvpn down because gateway down, interface have no IPv6, interface have no IPv6 gateway
Feb 19 17:01:08 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Glorytun VPN down, restart it
Feb 19 17:01:08 OpenMPTCProuter daemon.info glorytun: starting glorytun vpn instance vpn

patimages, 2024-02-19:
they are identical

patimages, 2024-02-19:
came back to 0.59 (to keep peace at home ;-D).

patimages, 2024-02-19:
After multiple tries, I can confirm that 0.59 allows for external access while 0.60 prevents it (via port forwarding). I always get "VPN down" as soon as I try "the_VPS_IP:port_I_want". Not sure whether I am alone or not though.

Ysurac, 2024-02-20:
When VPN is down, this seems logical that you don't have external access...
There is several VPN available, by default OpenVPN is now used.

patimages, 2024-02-20:
Totally agreed : there is an issue with glorytun. I'll give try to openvpn then but, in old tests, it was not as efficient as glorytun. I'll report back tonight after trying. Many thanks for all this support !!! Amazing when one considers it is for free !!!

patimages, 2024-02-23:
FYI : When I stopped all other VPN and started openVPN only, I got an alert stating that I had 2 VPN running... I gave up and came back to 0.59.

Ysurac, 2024-02-23:
How do you stop/start VPN ? All must be done via the wizard only.

patimages, 2024-02-23:
wizard only, indeed .... Tried all the VPN but seems as if another one was active although not in the wizard.

Ysurac, 2024-02-23:
You should have the information in Status->System log
This may explain why you have an issue with glorytun.
I would need the full output of ```uci show``` (without key/ips) or via System->OpenMPTCProuter, tab "Show all settings" (already obfuscated)

patimages, 2024-03-07:
RC1 solved all of my issues ! Many thanks Icarus !!!!

closed 2024-03-07 by patimages