Ysurac/openmptcprouter #3169: Manage iptables rules for change wired connection for OMR-VPS

GermanAizek, 2024-02-16:
@Ysurac,
I installed omr-vps locally after launch debian9_x86_64.sh I had an internet connection via wifi. Connecting wired internet to enp1s0 interface and USB-Ethernet adapter generated enx... interface, does not connect to Internet `ping 8.8.8.8` an unreachable host, everything is fine over wifi. Do I understand correctly that I need to configure iptables for interfaces? Is there an example of how to give me access over wired connection for OMR-VPS.

Ysurac, 2024-02-17:
Please use the template.
You can change the Internet interface in ```/etc/shorewall/params.net``` then restart shorewall ```systemctl restart shorewall```

GermanAizek, 2024-02-17:
@Ysurac, many thanks!
and how do I setup wired interface for downloading `opkg update` and `opkg install` for OMR? I want to try create vpn ZeroTier with Moon tunnel between OMR-VPS and VPS to create static ip. I tried to connect via lan and wan, eth0, eth1 and etc. but there is always no internet access.

Ysurac, 2024-02-17:
If you have all green in Status page, you should have Internet access.
What is the result of the commands ?

closed 2024-05-23 by github-actions[bot]

zqx0118, 2024-08-27:
> @Ysurac, I installed omr-vps locally after launch debian9_x86_64.sh I had an internet connection via wifi. Connecting wired internet to enp1s0 interface and USB-Ethernet adapter generated enx... [...]

May I ask if you are conducting router and server testing in a local area network after deploying the server locally? If so, I would greatly appreciate your help. Thank you very much.
My question is how to change the firewall rules or other settings to make this system run on the local area network. I really hope to get your help. Thank you.

GermanAizek, 2024-08-29:
> May I ask if you are conducting router and server testing in a local area network after deploying the server locally? If so, I would greatly appreciate your help. Thank you very much. My question is [...]

@zqx0118 hello, I still haven't figured out how to run OMR-VPS on local network without remote VPS hosting, to be honest, this is the most convenient way, but as I understand it,  author project (@Ysurac) is against such use and does not want to support automatic launch setup OMR-VPS on local network without dedicated IPv4 in any way.

Ysurac, 2024-08-30:
Why installing the VPS on same local network ? This can't aggregate anything then...
Else the VPS script can be installed on any VM or computer under Debian.

GermanAizek, 2024-08-30:
> Why installing the VPS on same local network ? This can't aggregate anything then... Else the VPS script can be installed on any VM or computer under Debian.

I have 3 ISP, I would like to allocate my IP address to one of them for OMR-VPS on local network, and other 2 for summation only.

GermanAizek, 2024-08-30:
Everyone's IP is different, each with 1 Gb optic fiber.

Ysurac, 2024-08-30:
If all ISP have 1GB/s you will then have a max speed of 1Gb/s...
The VPS need to have a bandwidth higher than the aggregated speed you can reach. So in your case to aggregate 3 ISP 1Gb/s, a VPS with at least 3Gb/s connectivity.

GermanAizek, 2024-08-30:
> If all ISP have 1GB/s you will then have a max speed of 1Gb/s... The VPS need to have a bandwidth higher than the aggregated speed you can reach. So in your case to aggregate 3 ISP 1Gb/s, a VPS with at least 3Gb/s connectivity.

Ok i can downgrade 2 ISP to 500 Mb, 3 is stay 1 Gb, is it possible to deploy all this on a local network without remote access and and purchase very specific VPS for OMR-VPS.

Ysurac, 2024-08-30:
Yes, you will be limited by 3 to 1Gb/s and aggregate ISP 1 & 2. (So really no advantages)
You only need to install it on a Debian 12 with network already configured and range 65000-65535 redirected to it.