Ysurac/openmptcprouter #3175: Local routing - Cannot connect from wan1 or wan2 subnet to lan

murateri75, 2024-02-19:
## Expected Behavior

Connect from one device within wan1 or wan2 (other than routers) to lan devices 
Ex : Synology with Surveillance station on wan1, cameras on lan (behind OpenMPTCP router) with ONVIF protocol (port 80)

## Current Behavior

Impossible to connect to IP Camera on lan from Synology on wan1 subnet

## Specifications

  - OpenMPTCProuter version:  openmptcprouter v0.59.1-5.4 r0+16594-ce92de8c8c
  - OpenMPTCProuter VPS version:  5.4.207-mptcp 0.1028
  - OpenMPTCProuter VPS provider: private
  - OpenMPTCProuter platform:  Raspberry Pi 4 Model B Rev 1.4

murateri75, 2024-02-19:
I tried Port Forwarding in Firewall but without success

Ysurac, 2024-02-19:
It's blocked by firewall. In anycase it's not a good idea.

murateri75, 2024-02-19:
That was my assumption
So I need to move my Synology on the router lan correct ?

murateri75, 2024-02-19:
I mean should be only routers on wan1 and wan2

Ysurac, 2024-02-19:
On WAN* you should only have routers/modems/box.

murateri75, 2024-02-19:
Understood

closed 2024-05-24 by github-actions[bot]