Ysurac/openmptcprouter #3181: Error when updating VPS with the latest snapshot.

vempire-ghost, 2024-02-23:
Updating VPS with kernel 5.4 using the command 'wget -O - https://www.openmptcprouter.com/server-test/debian-x86_64.sh | sh' is always returning an error at this stage.

Glorytun UDP
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following packages were automatically installed and are no longer required:
  libfstrm0 libjemalloc2 liblmdb0 libmaxminddb0 libprotobuf-c1 libssl1.1
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 1 reinstalled, 0 to remove and 35 not upgraded.
Need to get 0 B/27.0 kB of archives.
After this operation, 0 B of additional disk space will be used.
(Reading database ... 100109 files and directories currently installed.)
Preparing to unpack .../omr-glorytun_0.3.4-5_amd64.deb ...
Unpacking omr-glorytun (0.3.4-5) over (0.3.4-5) ...
Setting up omr-glorytun (0.3.4-5) ...
Job for glorytun-udp@tun0.service failed because the control process exited with error code.
See "systemctl status glorytun-udp@tun0.service" and "journalctl -xeu glorytun-udp@tun0.service" for details.
dpkg: error processing package omr-glorytun (--configure):
 installed omr-glorytun package post-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 omr-glorytun
E: Sub-process /usr/bin/dpkg returned an error code (1)

vempire-ghost, 2024-02-23:
vempire@instance-3:~$ systemctl status glorytun-udp@tun0.service
× glorytun-udp@tun0.service - Glorytun UDP on tun0
     Loaded: loaded (/lib/systemd/system/glorytun-udp@.service; enabled; preset: enabled)
     Active: failed (Result: exit-code) since Fri 2024-02-23 13:42:54 UTC; 13s ago
   Duration: 4ms
    Process: 4026 ExecStart=/usr/local/bin/glorytun-udp-run /etc/glorytun-udp/tun0 (code=exited, status=2)
    Process: 4027 ExecStartPost=/etc/glorytun-udp/post.sh /etc/glorytun-udp/tun0 (code=exited, status=2)
   Main PID: 4026 (code=exited, status=2)
        CPU: 8ms

Feb 23 13:42:53 instance-3.southamerica-east1-a.c.helical-fin-392321.internal systemd[1]: glorytun-udp@tun0.service: Failed with result 'exit-code'.
Feb 23 13:42:53 instance-3.southamerica-east1-a.c.helical-fin-392321.internal systemd[1]: Failed to start glorytun-udp@tun0.service - Glorytun UDP on tun0.
Feb 23 13:42:54 instance-3.southamerica-east1-a.c.helical-fin-392321.internal systemd[1]: glorytun-udp@tun0.service: Scheduled restart job, restart counter is at 5.
Feb 23 13:42:54 instance-3.southamerica-east1-a.c.helical-fin-392321.internal systemd[1]: Stopped glorytun-udp@tun0.service - Glorytun UDP on tun0.
Feb 23 13:42:54 instance-3.southamerica-east1-a.c.helical-fin-392321.internal systemd[1]: glorytun-udp@tun0.service: Start request repeated too quickly.
Feb 23 13:42:54 instance-3.southamerica-east1-a.c.helical-fin-392321.internal systemd[1]: glorytun-udp@tun0.service: Failed with result 'exit-code'.
Feb 23 13:42:54 instance-3.southamerica-east1-a.c.helical-fin-392321.internal systemd[1]: Failed to start glorytun-udp@tun0.service - Glorytun UDP on tun0.

vempire@instance-3:~$ journalctl -xeu glorytun-udp@tun0.service
░░ Automatic restarting of the unit glorytun-udp@tun0.service has been scheduled, as the result for
░░ the configured Restart= setting for the unit.
Feb 23 13:43:35 instance-3.southamerica-east1-a.c.helical-fin-392321.internal systemd[1]: Stopped glorytun-udp@tun0.service - Glorytun UDP on tun0.
░░ Subject: A stop job for unit glorytun-udp@tun0.service has finished
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░
░░ A stop job for unit glorytun-udp@tun0.service has finished.
░░
░░ The job identifier is 5163 and the job result is done.
Feb 23 13:43:35 instance-3.southamerica-east1-a.c.helical-fin-392321.internal systemd[1]: Starting glorytun-udp@tun0.service - Glorytun UDP on tun0...
░░ Subject: A start job for unit glorytun-udp@tun0.service has begun execution
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░
░░ A start job for unit glorytun-udp@tun0.service has begun execution.
░░
░░ The job identifier is 5163.
Feb 23 13:43:35 instance-3.southamerica-east1-a.c.helical-fin-392321.internal glorytun-udp-run[4818]: /usr/local/bin/glorytun-udp-run: 10: .: cannot open /etc/glorytun-udp/tun0: Permission denied
Feb 23 13:43:35 instance-3.southamerica-east1-a.c.helical-fin-392321.internal systemd[1]: glorytun-udp@tun0.service: Main process exited, code=exited, status=2/INVALIDARGUMENT
░░ Subject: Unit process exited
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░
░░ An ExecStart= process belonging to unit glorytun-udp@tun0.service has exited.
░░
░░ The process' exit code is 'exited' and its exit status is 2.
Feb 23 13:43:35 instance-3.southamerica-east1-a.c.helical-fin-392321.internal post.sh[4819]: /etc/glorytun-udp/post.sh: 3: .: cannot open /etc/glorytun-udp/tun0: Permission denied
Feb 23 13:43:35 instance-3.southamerica-east1-a.c.helical-fin-392321.internal systemd[1]: glorytun-udp@tun0.service: Failed with result 'exit-code'.
░░ Subject: Unit failed
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░
░░ The unit glorytun-udp@tun0.service has entered the 'failed' state with result 'exit-code'.
Feb 23 13:43:35 instance-3.southamerica-east1-a.c.helical-fin-392321.internal systemd[1]: Failed to start glorytun-udp@tun0.service - Glorytun UDP on tun0.
░░ Subject: A start job for unit glorytun-udp@tun0.service has failed
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░
░░ A start job for unit glorytun-udp@tun0.service has finished with a failure.
░░
░░ The job identifier is 5163 and the job result is failed.
Feb 23 13:43:35 instance-3.southamerica-east1-a.c.helical-fin-392321.internal systemd[1]: glorytun-udp@tun0.service: Scheduled restart job, restart counter is at 5.
░░ Subject: Automatic restarting of a unit has been scheduled
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░
░░ Automatic restarting of the unit glorytun-udp@tun0.service has been scheduled, as the result for
░░ the configured Restart= setting for the unit.
Feb 23 13:43:35 instance-3.southamerica-east1-a.c.helical-fin-392321.internal systemd[1]: Stopped glorytun-udp@tun0.service - Glorytun UDP on tun0.
░░ Subject: A stop job for unit glorytun-udp@tun0.service has finished
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░
░░ A stop job for unit glorytun-udp@tun0.service has finished.
░░
░░ The job identifier is 5239 and the job result is done.
Feb 23 13:43:35 instance-3.southamerica-east1-a.c.helical-fin-392321.internal systemd[1]: glorytun-udp@tun0.service: Start request repeated too quickly.
Feb 23 13:43:35 instance-3.southamerica-east1-a.c.helical-fin-392321.internal systemd[1]: glorytun-udp@tun0.service: Failed with result 'exit-code'.
░░ Subject: Unit failed
░░ Defined-By: systemd
[... cut ...]
░░ The unit glorytun-udp@tun0.service has entered the 'failed' state with result 'exit-code'.
Feb 23 13:43:35 instance-3.southamerica-east1-a.c.helical-fin-392321.internal systemd[1]: Failed to start glorytun-udp@tun0.service - Glorytun UDP on tun0.
░░ Subject: A start job for unit glorytun-udp@tun0.service has failed
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░
░░ A start job for unit glorytun-udp@tun0.service has finished with a failure.
░░
░░ The job identifier is 5239 and the job result is failed.

vempire-ghost, 2024-02-23:
vempire@instance-3:~$ ls -la /etc/glorytun-udp/
total 24
drwxr-xr-x  2 vempire vempire 4096 Feb 23 13:30 .
drwxr-xr-x 97 root    root    4096 Feb 23 14:35 ..
-rwxr-xr-x  1 vempire vempire  641 Jun 14  2021 post.sh
-rw-------  1 vempire vempire   76 Feb  1 23:34 tun0
-rw-------  1 vempire vempire   86 Sep  2  2021 tun0.dpkg-dist
-rw-------  1 vempire vempire   65 Nov 27 23:49 tun0.key

Ysurac, 2024-02-25:
Why files belong to user vempire ? Should belong to user root and group root

vempire-ghost, 2024-02-25:
A while ago, I tried to copy the files via SFTP, and as I didn't have the necessary permissions, I used the command 'sudo chown -R vempire:vempire /etc/glorytun-udp.' How can I correct this and restore ownership to the root user?

closed 2024-05-30 by github-actions[bot]