Ysurac/openmptcprouter #3183: Error when using Glorytun TCP on Kernel Log.

vempire-ghost, 2024-02-26:
## Expected Behavior
No errors.

## Current Behavior
"When I select glorytun tcp as the VPN, everything works normally, but these errors appear in the kernel log. Is this something to be concerned about?"
[  472.763709] __mptcp_init4_subsockets: token 0xffe02041 bind() to 10.255.255.2 index 50 failed, error -99
[  472.785368] __mptcp_init4_subsockets: token 0x5c5693fb bind() to 10.255.255.2 index 50 failed, error -99
[  472.785756] __mptcp_init4_subsockets: token 0x778195aa bind() to 10.255.255.2 index 50 failed, error -99
[  472.840589] __mptcp_init4_subsockets: token 0x1111e794 bind() to 10.255.255.2 index 50 failed, error -99
[  472.934623] __mptcp_init4_subsockets: token 0xdeb81ae3 bind() to 10.255.255.2 index 50 failed, error -99

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.60beta2-5.4 r0+16862-170d9 snap from 25/02
  - OpenMPTCProuter VPS version: 5.4.207-mptcp 0.1029-test 
  - OpenMPTCProuter VPS provider: GCP
  - OpenMPTCProuter platform: x86_64
  - Country:

Ysurac, 2024-02-27:
In Network->MPTCP check that MPTCP is not enabled on OMRVPN interface.

vempire-ghost, 2024-02-28:
[image]
There is no interface named OMRVPN.

Ysurac, 2024-02-28:
It can be changed in Network->Interfaces, edit OMRVPN interface and "Advanced parameters" tab

vempire-ghost, 2024-03-01:
[image]
Its disabled.

Ysurac, 2024-03-01:
Do you have anything in Status->System log about multipath ?

vempire-ghost, 2024-03-02:
Only something like this.
Mar  2 15:41:38 OpenMPTCProuter user.notice MPTCP: Set eth1 to master
Mar  2 15:41:38 OpenMPTCProuter user.notice MPTCP: Set eth2 to on
Mar  2 15:41:39 OpenMPTCProuter user.notice MPTCP: Set eth3 to backup
Mar  2 15:41:39 OpenMPTCProuter user.notice MPTCP: Set eth4 to on
Mar  2 15:41:39 OpenMPTCProuter user.notice MPTCP: Set eth5 to on
Mar  2 15:41:39 OpenMPTCProuter user.notice MPTCP: Disabling MPTCP on interface ifb4tun0 not found in enabled multipath list
Mar  2 15:41:40 OpenMPTCProuter user.notice post-tracking-020-status: Check API configuration...
Mar  2 15:41:40 OpenMPTCProuter user.notice post-tracking-020-status: Check API configuration... Done
Mar  2 15:41:41 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Tunnel up : Replace default route by 10.255.255.1 dev tun0
Mar  2 15:41:42 OpenMPTCProuter user.notice post-tracking-020-status: New public ip detected for omrvpn (tun0): xx.215.225.xxx (previous: xx.215.210.xx)
Mar  2 15:41:42 OpenMPTCProuter user.notice post-tracking-020-status: Reload MPTCP for tun0
Mar  2 15:41:47 OpenMPTCProuter daemon.info omr-tracker-xray: xray is up (can contact via http 74.82.42.42)

closed 2024-06-06 by github-actions[bot]