Ysurac/openmptcprouter #3184: Issues getting Connection between VPS / Router

vinc32, 2024-02-27:
## Expected Behavior
Router should connect to VPS getting config via API.

## Current Behavior
5G Connection -> Green - showing Public IP
VPS -> Not able to connect to Server (Admin Script, no result)
-> LOG: "OMR-VPS: Can't get vps token, try later (can ping server vps on %IP%, server API answer on %IP%

Trying to activate VPN Option for 5G Connection -> Gateway Down, No Server ping response.

## Log
Tue Feb 27 11:33:42 2024 daemon.err glorytun[18843]: %IP%.65001: connected
Tue Feb 27 11:35:01 2024 user.notice OMR-VPS: Can't get vps token, try later (can ping server vpse Feb 27 11:33:52 2024 daemon.info glorytun[18843]: STOPPED tun0
Tue Feb 27 11:34:46 2024 user.notice OMR-VPS: Setting server config error, try again
Tue Feb 27 11:35:01 2024 user.notice OMR-VPS: Can't get vps token, try later (can ping server vps on %IP%, no server API answer on %IP%)
Tue Feb 27 11:35:01 2024 daemon.info glorytun: starting glorytun vpn instance vpn
Tue Feb 27 11:35:01 2024 user.notice omr-bypass: Starting OMR-ByPass...
Tue Feb 27 11:35:02 2024 user.notice omr-bypass: OMR-ByPass is running
Tue Feb 27 11:35:02 2024 user.notice omr-tracker: Launching...
Tue Feb 27 11:35:07 2024 user.notice omr-tracker: Launched
Tue Feb 27 11:35:16 2024 user.notice OMR-VPS: Can't get vps token, try later (can ping server vps on %IP%, no server API answer on %IP%)

## Specifications

  - OpenMPTCProuter version: 0.59.1-5.4
  - OpenMPTCProuter VPS version: 0.1028 5.4.207-mptcp
  - OpenMPTCProuter VPS provider: ionos
  - OpenMPTCProuter platform: esxi image x86_64

Ysurac, 2024-02-27:
Did you reboot VPS after install ?
Why do you set a VPN ?

vinc32, 2024-02-27:
reboot done - same issue.

I've thought that the ISP (Telekom 5G Germany) may blocks MPTCP Connections - as workaround it said to activate it ;)

Ysurac, 2024-02-27:
Even we MPTCP blocked, it should be able to contact VPS API.
What is the result when you go with a browser to https://<vpsip>:65500/ ? and using ```curl -k https://<vpsip>:65500/``` from the router via SSH ? and ```curl -k https://127.0.0.1:65500/``` from the VPS ?
If no results, what is the result of ```journalctl -u omr-admin``` from the VPS ?

vinc32, 2024-02-27:
## browser to https://:65500/
- Nothing - not able to open a connection - not to localhost or ip of router / vps

## curl -k https://127.0.0.1:65500/ from VPS
(curl: (7) Failed to connect to 127.0.0.1 port 65500 after 0 ms: Couldn't connect to server

## curl -k https://<vpsip>:65500/ from Router
"Welcome to OpenMPTCProuter Server part"root@OpenMPTCProuter:~# 

## journalctl -u omr-admin
root@localhost:~# journalctl -u omr-admin
-- Journal begins at Wed 2021-09-15 09:02:55 UTC, ends at Tue 2024-02-27 16:22:10 UTC. --
Feb 12 08:25:41 localhost systemd[1]: Started OMR-Admin.
Feb 25 20:03:03 localhost omr-admin.py[1856564]: /bin/sh: 1: dig: not found
Feb 25 20:50:01 localhost systemd[1]: Stopping OMR-Admin...
Feb 25 20:50:04 localhost systemd[1]: omr-admin.service: Succeeded.
Feb 25 20:50:04 localhost systemd[1]: Stopped OMR-Admin.
Feb 25 20:50:04 localhost systemd[1]: omr-admin.service: Consumed 27min 50.544s CPU time.
-- Boot 96d3d98e991349f6b2b88102759e5bb7 --
Feb 25 20:50:31 localhost systemd[1]: Started OMR-Admin.

Ysurac, 2024-02-27:
Can you run ```apt-get install dnsutils && systemctl omr-admin``` on the VPS ? At least this fix the dig error.
You are using an IPv4 server IP ?
You remove the VPN on 5G interface ?

vinc32, 2024-02-27:
systemctl omr-admin -> wasn't found -> not sure how to fix it
"E: Unable to locate package omr-admin"

Using IPv4 Server IP
Removed VPN on 5G Interface

Just went through install routine and noticed something - VPS running Debian 11 with IONOS Image - used  
"wget -O - https://www.openmptcprouter.com/server/debian-x86_64.sh | sh"

To install VPS. Maybe that's the issue?

Ysurac, 2024-02-28:
Sorry it's ```systemctl restart omr-admin```
I have some IONOS VPS and it's working.

Ysurac, 2024-02-28:
What do you have now in "Status->System log" ?
Range 65000-65535 in TCP and UDP, and ICMP are correctly open in Ionos interface ?

vinc32, 2024-02-28:
Added Ports according Wiki, will try to open everything between 65000-65535
[image]

vinc32, 2024-02-28:
Wed Feb 28 17:02:05 2024 daemon.er glorytun[23137]: %IP%:65001: key exchange failed
Wed Feb 28 17:02:05 2024 daemon.info glorytun|23137): STOPPED tunO
Wed Feb 28 17:02:14 2024 user.notice post-tracking-post-tracking: ompn down because gateway down
Wed Feb 28 17:02:14 2024 user.notice post-tracking-post-tracking: Glorytun VPN down, restart it 
Wed Feb 28 17:02:14 2024 daemon.err glorytun[23137]: getpeername: Socket not connected 
Wed Feb 28 17:02:14 2024 daemon.notice netifd: Network device 'tun®" link is down 
Wed Feb 28 17:02:14 2024 daemon.notice netifd: Interface 'ompn' has link connectivity loss 
Wed Feb 28 17:02:14 2024 daemon.notice netifd: Interface 'omrpn' is now down 
Wed Feb 28 17:02:14 2024 daemon.notice netifd: Interface 'ompn' is disabled 
Wed Feb 28 17:02:14 2024 daemon.info glorytun: starting glorytun von instance vpn 
Wed Feb 28 17:02:14 2024 daemon.notice netifd: Interface 'omon' is enabled 
Wed Feb 28 17:02:14 2024 daemon.notice netifd: Network device 'tunO" link is up 
Wed Feb 28 17:02:14 2024 daemon.notice netifd: Interface 'ompn' has link connectivity 
Wed Feb 28 17:02:14 2024 daemon.notice netifd: Interface 'ompn' is setting up now 
Wed Feb 28 17:02:14 2024 daemon.info glorytun[1397]: INITIALIZED tunO 
Wed Feb 28 17:02:14 2024 daemon.notice netifd: Interface 'omvpn' is now up
Wed Feb 28 17:02:14 2024 daemon.warn [7433]: <warn> [SQM_IFB_869a4] invalid systs path read for net/SQM_IFB_869a4
Wed Feb 28 17:02:14 2024 daemon.war [7433]: <warn> [base-manager] couldn't handle kernel event: device net/SQM_IFB_869a4 not found 
Wed Feb 28 17:02:14 2024 daemon.warn [7433]: <warn> [SQM_IFB_9dc2c] invalid sysfs path read for net/SQM_IFB_9dc2c
Wed Feb 28 17:02:14 2024 daemon.warn [7433]: <warn> [base-manager] couldn't handle kernel event: device net/SQM_IFB_9dc2c not found
Wed Feb 28 17:02:14 2024 user notice firewall: Reloading firewall due to ifup of omrvon (tunO)
Wed Feb 28 17:02:15 2024 user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Wed Feb 28 17:02:15 2024 user notice mptcp: Reloading mptcp config due to ifup of omrvon
Wed Feb 28 17:03:54 2024 daemon.err glorytun|13971: %IP%.65001: connected
Wed Feb 28 17:04:04 2024 daemon.err glorytun[13971: %IP%.65001: key exchange failed
Wed Feb 28 17:04:04 2024 daemon.info glorytun[1397]: STOPPED tuno
Wed Feb 28 17:04:04 2024 daemon.err glorytun[1397]: %IP%.65001: connected
Wed Feb 28 17:04:06 2024 user.notice SQM-autorate: Launch on wan1
Wed Feb 28 17:04:08 2024 user.notice SQM-autorate: Launch on wan2
Wed Feb 28 17:04:08 2024 user.notice SQM-autorate: Launch on omrvon|
Wed Feb 28 17:04:08 2024 user.notice SQM-autorate: Launch on lan
Wed Feb 28 17:04:14 2024 daemon.err glorytun|13971: %IP%.65001: key exchange failed
Wed Feb 28 17:04:14 2024 daemon.info glorytun[13971: STOPPED tunO

vinc32, 2024-02-28:
not onsite - so copied out of VMware console - so it copied out of a "picture"

vinc32, 2024-03-01:
Solution found -

5G Router has been the issue - it’s a ZTE MC801A and it’s making the trouble.
Changing the APN manually worked.