Ysurac/openmptcprouter #3188: SQM QoS

Yessine1200, 2024-03-01:
Hello,
I want to understand exactly how to use SQM to improve QoS. In the wizard, there are two boxes: one to enable SQM and the other for SQM autorate. 
1. For the first one, do I need to modify the DL/UL traffic for the links I want to aggregate to avoid link saturation? 
2. For the tunnel created to aggregate the throughput, what values should I set? 
3. How can I prioritize UDP traffic?
[image]

Yessine1200, 2024-03-01:
I see in the documentation you sent: SQM is an integrated system that performs per-packet/per flow network scheduling, active queue length management (AQM), traffic shaping/rate limiting, and QoS (prioritization). If that true I can prioritize ? or I have to install other tool to do that ?

Ysurac, 2024-03-01:
You can use DSCP for priority or install luci-app-qos instead of SQM.

Yessine1200, 2024-03-01:
For example, If I priorise UDP trafic for video diffusion using DSCP: how OMR will react ? The traffic will be send over the link with lower latency ? and OMR will use openVPN(I mean the tunnel created between OMR and VPS) to send the trafic to the server (I am using v6.0beta2)?

Yessine1200, 2024-03-04:
If I use SQM Autorate, OMR utilizes the defined algorithm to automatically detect the maximum/minimum speed based on latency and congestion, especially for mobile networks. My question is: Does SQM Autorate calculate the maximum speed periodically? Because, as you know, mobile networks are not stable in terms of latency and speed.

closed 2024-06-07 by github-actions[bot]