Ysurac/openmptcprouter #3191: understanding OMR

YB1200, 2024-03-03:
Hello, I am new to this project and eager to understand more about the functionalities of OMR. As far as I've grasped, OMR utilizes MPTCP to aggregate throughput for TCP traffic. However, I'm curious about the handling of UDP traffic. Could you please clarify what method is employed for UDP traffic? I've noticed that openVPN TCP is utilized. Additionally, following the aggregation process, what technology is used for the creation of the tunnel between OMR and VPS to pass the aggregated throughput? Also, could you explain when you use shadowsocks and VPN?

Ysurac, 2024-03-03:
For UDP, by default a VPN over TCP is used and aggregated. Some proxy also can do UDP over TCP (V2Ray and XRay). There is Glorytun UDP and MLVPN as VPN that can aggregate UDP over UDP, but performance and stability is often not really good.
Between OMR and the VPS there is a Proxy (Shadowsocks-(libev|rust), V2ray or Xray) used for TCP and a VPN (Glorytun-TCP, OpenVPN, DSVPN,...) for UDP.

YB1200, 2024-03-04:
So if I understand correctly, on the OMR side, MPTCP is used to aggregate both TCP traffic and UDP (VPN over TCP, where VPN options include openVPN TCP and Glorytun TCP). Between OMR and the VPS, shadowsocks is used for TCP, and the same VPN is used for aggregation for UDP, correct?

Ysurac, 2024-03-06:
yes

YB1200, 2024-03-22:
Hello Ysurac,
I have another question about MPTCP. When creating a Subflow between OMR and ISP, does OMR use a proxy for TCP to encrypt data? Or is the proxy only used between OMR and the VPS to transmit the aggregated throughput?

Ysurac, 2024-03-22:
The proxy is used for TCP between OMR and the VPS and is encrypted.