Ysurac/openmptcprouter #3192: omr-bypass doesn't work in omr v0.60rc1

rdmitry0911, 2024-03-04:
## Expected Behavior
I expect that omr-bypass works
When I put anything in omr-bypass settings I expect the script to generate correct rules to bypass

## Current Behavior
Script generates rules for the firewall that can not be executed
I see these messages in firewal log:
```
Section omr_dst_bypass_eth1_dstip_4 (omr_dst_bypass_eth1_rule) option 'set_mark' specifies invalid value '0x453910002'
Section omr_dst_bypass_eth1_dstip_4_accept (omr_dst_bypass_eth1_rule_accept) option 'mark' specifies invalid value '0x453910002'
Section omr_dst_bypass_eth1_srcip_4 (omr_dst_bypass_eth1_srcip) option 'set_xmark' specifies invalid value '0x453910002'
Section omr_dst_bypass_eth1_mac_4 (omr_dst_bypass_eth1_mac) option 'set_xmark' specifies invalid value '0x453910002'
Section omr_dst_bypass_eth1_srcport_tcp_4 (omr_dst_bypass_eth1_srcport) option 'set_xmark' specifies invalid value '0x453910002'
Section omr_dst_bypass_eth1_srcport_udp_4 (omr_dst_bypass_eth1_srcport) option 'set_xmark' specifies invalid value '0x453910002'
Section omr_dst_bypass_eth1_dstport_tcp_4 (omr_dst_bypass_eth1_dstport) option 'set_xmark' specifies invalid value '0x453910002'
Section omr_dst_bypass_eth1_dstport_udp_4 (omr_dst_bypass_eth1_dstport) option 'set_xmark' specifies invalid value '0x453910002'
Section omr_dst_bypass_eth3_dstip_4 (omr_dst_bypass_eth3_rule) option 'set_mark' specifies invalid value '0x453910004'
Section omr_dst_bypass_eth3_dstip_4_accept (omr_dst_bypass_eth3_rule_accept) option 'mark' specifies invalid value '0x453910004'
Section omr_dst_bypass_eth3_srcip_4 (omr_dst_bypass_eth3_srcip) option 'set_xmark' specifies invalid value '0x453910004'
Section omr_dst_bypass_eth3_mac_4 (omr_dst_bypass_eth3_mac) option 'set_xmark' specifies invalid value '0x453910004'
Section omr_dst_bypass_eth3_srcport_tcp_4 (omr_dst_bypass_eth3_srcport) option 'set_xmark' specifies invalid value '0x453910004'
Section omr_dst_bypass_eth3_srcport_udp_4 (omr_dst_bypass_eth3_srcport) option 'set_xmark' specifies invalid value '0x453910004'
Section omr_dst_bypass_eth3_dstport_tcp_4 (omr_dst_bypass_eth3_dstport) option 'set_xmark' specifies invalid value '0x453910004'
Section omr_dst_bypass_eth3_dstport_udp_4 (omr_dst_bypass_eth3_dstport) option 'set_xmark' specifies invalid value '0x453910004'
Section omr_dst_bypass_eth2_dstip_4 (omr_dst_bypass_eth2_rule) option 'set_mark' specifies invalid value '0x453910003'
Section omr_dst_bypass_eth2_dstip_4_accept (omr_dst_bypass_eth2_rule_accept) option 'mark' specifies invalid value '0x453910003'
Section omr_dst_bypass_eth2_srcip_4 (omr_dst_bypass_eth2_srcip) option 'set_xmark' specifies invalid value '0x453910003'
Section omr_dst_bypass_eth2_mac_4 (omr_dst_bypass_eth2_mac) option 'set_xmark' specifies invalid value '0x453910003'
Section omr_dst_bypass_eth2_srcport_tcp_4 (omr_dst_bypass_eth2_srcport) option 'set_xmark' specifies invalid value '0x453910003'
Section omr_dst_bypass_eth2_srcport_udp_4 (omr_dst_bypass_eth2_srcport) option 'set_xmark' specifies invalid value '0x453910003'
Section omr_dst_bypass_eth2_dstport_tcp_4 (omr_dst_bypass_eth2_dstport) option 'set_xmark' specifies invalid value '0x453910003'
Section omr_dst_bypass_eth2_dstport_udp_4 (omr_dst_bypass_eth2_dstport) option 'set_xmark' specifies invalid value '0x453910003'

```
values generated for set_xmark and set_mark are too long 

## Possible Solution
Make shorter the values generated by scripts for expressions in firewall with set_xmark and set_mark statements  

## Steps to Reproduce the Problem

  1. Put 1.1.1.2 in omr-bypass IPs and Networks and chose any interface for this bypass
  2. Save & Apply
  3. Run /etc/inint.d/firewall restart

## Context (Environment)
Different routes for different cases

## Specifications

  - OpenMPTCProuter version: v0.60rc1-6.1 r0+24843-acf40c022e
  - OpenMPTCProuter VPS version: 0.1029-test 6..1.0-18-amd64
  - OpenMPTCProuter VPS provider: Local proxmox vm installation
  - OpenMPTCProuter platform: x86_64
  - Country: GR

Ysurac, 2024-03-04:
You have a problem in your configuration. Can you put the result of ```uci show network``` via SSH from the router ?

Ysurac, 2024-03-04:
ok I found the bug. For now you can manually change the metric for each interface in Network->Interfaces and edit interfaces.

rdmitry0911, 2024-03-04:
> ok I found the bug. For now you can manually change the metric for each interface in Network->Interfaces and edit interfaces.

Unfortunately not.  Changing only metric for the interface doesn't work as restarting omr-bypass restores some how long metrics in /etc/config/network and  long values for marks in /etc/config/firewall

Ysurac, 2024-03-04:
You need also to lower metric in LAN.
A new snapshot is compiling with the fix.

rdmitry0911, 2024-03-05:
Also I've found that /etc/firewall.omr-bypass file that script generates also looks strange. 
```
#!/bin/sh
#nft insert rule inet fw4 ss_rules_dst_tcp ip daddr @omr_dst_bypass_all accept
#nft insert rule inet fw4 ss_rules_local_out ip daddr @omr_dst_bypass_all accept
[ -z "$(nft list ruleset | grep ss_rules)" ] && exit 0
#nft insert rule inet fw4 ss_rules_dst_tcp ip daddr @omr_dst_bypass_all_4 meta mark set 0x00004539 accept
#nft insert rule inet fw4 ss_rules_local_out ip daddr @omr_dst_bypass_all_4 meta mark set 0x00004539 accept
#nft add chain inet fw4 bypass_prerouting '{ type nat hook prerouting priority filter - 5; policy accept; }'
#nft add chain inet fw4 bypass_local '{ type nat hook output priority filter - 5; policy accept; }'
nft insert rule inet fw4 ss_rules_pre_tcp ip daddr @omr_dst_bypass_eth0_4 meta mark set 0x00004539 accept
nft insert rule inet fw4 ss_rules_local_out ip daddr @omr_dst_bypass_eth0_4 meta mark set 0x00004539 accept
nft insert rule inet fw4 ss_rules_pre_tcp ip daddr @omr_dst_bypass_eth1_4 meta mark set 0x00004539 accept
nft insert rule inet fw4 ss_rules_local_out ip daddr @omr_dst_bypass_eth1_4 meta mark set 0x00004539 accept
nft insert rule inet fw4 ss_rules_pre_tcp ip daddr @omr_dst_bypass_eth3_4 meta mark set 0x00004539 accept
nft insert rule inet fw4 ss_rules_local_out ip daddr @omr_dst_bypass_eth3_4 meta mark set 0x00004539 accept
nft insert rule inet fw4 ss_rules_pre_tcp ip daddr @omr_dst_bypass_eth2_4 meta mark set 0x00004539 accept
nft insert rule inet fw4 ss_rules_local_out ip daddr @omr_dst_bypass_eth2_4 meta mark set 0x00004539 accept
nft insert rule inet fw4 ss_rules_pre_tcp ip daddr @omr_dst_bypass_tun0_4 meta mark set 0x00004539 accept
nft insert rule inet fw4 ss_rules_local_out ip daddr @omr_dst_bypass_tun0_4 meta mark set 0x00004539 accept
nft insert rule inet fw4 ss_rules_pre_tcp ip daddr @omr_dst_bypass_all_4 meta mark set 0x00004539 accept
nft insert rule inet fw4 ss_rules_local_out ip daddr @omr_dst_bypass_all_4 meta mark set 0x00004539 accept
```
it puts the same 0x00004539 mark to traffic for all interfaces and does it twice. I think it is a mistake. Probably it should put different marks for ipv4 and ipv6 traffic specific for every interface

Ysurac, 2024-03-05:
Mark use the MTU value that is set automatically and should be different for each interface. Here it's like if MTU is empty. I think it's same problem as previous bug.

rdmitry0911, 2024-03-05:
> Mark use the MTU value that is set automatically and should be different for each interface. Here it's like if MTU is empty. I think it's same problem as previous bug.

You, probably, mean metric, not mtu?

Ysurac, 2024-03-05:
yes sorry, metric

rdmitry0911, 2024-03-05:
When are you plannig to make a commit with a fix of this bug? May be you can share the fix here in advance?

Ysurac, 2024-03-05:
It's already commited.

rdmitry0911, 2024-03-05:
Where can I find it?

Ysurac, 2024-03-05:
You need to put https://github.com/Ysurac/openmptcprouter-feeds/raw/develop/mptcp/files/etc/init.d/mptcp as /etc/init.d/mptcp and do a chmod u+x /etc/init.d/mptcp and "/etc/init.d/mptcp restart"

rdmitry0911, 2024-03-05:
It's getting better now, but still there is a problem with /etc/firewall.omr-bypass. After deleting it and rebooting the system it still looks like this:
root@OpenMPTCProuter:~# cat /etc/firewall.omr-bypass 
```
#!/bin/sh
#nft insert rule inet fw4 ss_rules_dst_tcp ip daddr @omr_dst_bypass_all accept
#nft insert rule inet fw4 ss_rules_local_out ip daddr @omr_dst_bypass_all accept
[ -z "$(nft list ruleset | grep ss_rules)" ] && exit 0
#nft insert rule inet fw4 ss_rules_dst_tcp ip daddr @omr_dst_bypass_all_4 meta mark set 0x00004539 accept
#nft insert rule inet fw4 ss_rules_local_out ip daddr @omr_dst_bypass_all_4 meta mark set 0x00004539 accept
#nft add chain inet fw4 bypass_prerouting '{ type nat hook prerouting priority filter - 5; policy accept; }'
#nft add chain inet fw4 bypass_local '{ type nat hook output priority filter - 5; policy accept; }'
nft insert rule inet fw4 ss_rules_pre_tcp ip daddr @omr_dst_bypass_eth0_4 meta mark set 0x00004539 accept
nft insert rule inet fw4 ss_rules_local_out ip daddr @omr_dst_bypass_eth0_4 meta mark set 0x00004539 accept
nft insert rule inet fw4 ss_rules_pre_tcp ip daddr @omr_dst_bypass_eth1_4 meta mark set 0x00004539 accept
nft insert rule inet fw4 ss_rules_local_out ip daddr @omr_dst_bypass_eth1_4 meta mark set 0x00004539 accept
nft insert rule inet fw4 ss_rules_pre_tcp ip daddr @omr_dst_bypass_eth3_4 meta mark set 0x00004539 accept
nft insert rule inet fw4 ss_rules_local_out ip daddr @omr_dst_bypass_eth3_4 meta mark set 0x00004539 accept
nft insert rule inet fw4 ss_rules_pre_tcp ip daddr @omr_dst_bypass_eth2_4 meta mark set 0x00004539 accept
nft insert rule inet fw4 ss_rules_local_out ip daddr @omr_dst_bypass_eth2_4 meta mark set 0x00004539 accept
nft insert rule inet fw4 ss_rules_pre_tcp ip daddr @omr_dst_bypass_tun0_4 meta mark set 0x00004539 accept
nft insert rule inet fw4 ss_rules_local_out ip daddr @omr_dst_bypass_tun0_4 meta mark set 0x00004539 accept
nft insert rule inet fw4 ss_rules_pre_tcp ip daddr @omr_dst_bypass_all_4 meta mark set 0x00004539 accept
nft insert rule inet fw4 ss_rules_local_out ip daddr @omr_dst_bypass_all_4 meta mark set 0x00004539 accept
```
mark values are the same for all cases

rdmitry0911, 2024-03-05:
And not only this. After /etc/init.d/omr-bypass restart restart all rules with mark target in /etc/config/firewall are diasabled:
```
config rule 'omr_dst_bypass_tun0_dstport_udp_4'
        option name 'omr_dst_bypass_tun0_dstport'
        option src 'lan'
        option dest '*'
        option target 'MARK'
        option enabled '0'
        option set_xmark '0x45391500'

```
I have to enable them manually. And netxt time omr-bypass is restared they are disabled again

Ysurac, 2024-03-05:
Rules are disabled when not used.
Did you rebooted or restarted omr-bypass ?

rdmitry0911, 2024-03-05:
Rebooted

rdmitry0911, 2024-03-05:
bypass rules are not working without manual intervention

Ysurac, 2024-03-05:
What do you have in ```uci show omr-bypass``` ?

rdmitry0911, 2024-03-05:
Here it is:
```
omr-bypass.all=interface
omr-bypass.m6replay=proto
omr-bypass.m6replay.url='m6web.fr' '6play.fr' '6cloud.fr'
omr-bypass.mycanal=proto
omr-bypass.mycanal.url='mycanal.fr' 'canal-plus.com' 'canalplus.com' 'canalplus-cdn.net' 'canalplus.pro' 'canal-plus.net'
omr-bypass.minecraft=proto
omr-bypass.minecraft.url='authserver.mojang.com'
omr-bypass.lesnumeriques=proto
omr-bypass.lesnumeriques.url='lesnumeriques.com' 'botscorner.com' 'app.botscorner.com'
omr-bypass.disneyplus=proto
omr-bypass.disneyplus.url='bamgrid.com' 'disney-plus.net'
omr-bypass.amazonvideo=proto
omr-bypass.amazonvideo.url='cloudfront.net' 'llnw.net'
omr-bypass.free=proto
omr-bypass.free.url='free.fr' 'freebox.fr' 'oqee.tv' 'oqee.net'
omr-bypass.orange=proto
omr-bypass.orange.url='orange.fr' 'sosh.fr' 'liveperson.net' 'liveperson.com' 'lpsn.net' 'lpsnmedia.net' 'francetelecom.fr'
omr-bypass.eth0=interface
omr-bypass.eth0.id='9999'
omr-bypass.eth1=interface
omr-bypass.eth1.id='1002'
omr-bypass.eth2=interface
omr-bypass.eth2.id='1003'
omr-bypass.tun0=interface
omr-bypass.tun0.id='1500'
omr-bypass.global=global
omr-bypass.global.vpn_ipv4_md5='68b329da9893e34099c7d8ad5cb9c940'
omr-bypass.global.vpn_ipv6_md5='68b329da9893e34099c7d8ad5cb9c940'
omr-bypass.eth3=interface
omr-bypass.eth3.id='1004'
omr-bypass.@ips[0]=ips
omr-bypass.@ips[0].ip='1.1.1.2'
omr-bypass.@ips[0].interface='eth3'

```

Ysurac, 2024-03-05:
You should have rules ```omr_dst_bypass_eth3_4``` and ```omr_dst_bypass_eth3_dstip_4``` as enabled in /etc/config/firewall and everything disabled.
What rules you need to enable to make bypass work ?

rdmitry0911, 2024-03-05:
in /etc/firewall.omr-bypass I need 
```
nft insert rule inet fw4 ss_rules_pre_tcp ip daddr @omr_dst_bypass_eth0_4 meta mark set 0x45391001 accept
nft insert rule inet fw4 ss_rules_local_out ip daddr @omr_dst_bypass_eth0_4 meta mark set 0x45391001 accept
nft insert rule inet fw4 ss_rules_pre_tcp ip daddr @omr_dst_bypass_eth1_4 meta mark set 0x45391002 accept
nft insert rule inet fw4 ss_rules_local_out ip daddr @omr_dst_bypass_eth1_4 meta mark set 0x45391002 accept
nft insert rule inet fw4 ss_rules_pre_tcp ip daddr @omr_dst_bypass_eth3_4 meta mark set 0x45391004 accept
nft insert rule inet fw4 ss_rules_local_out ip daddr @omr_dst_bypass_eth3_4 meta mark set 0x45391004 accept
nft insert rule inet fw4 ss_rules_pre_tcp ip daddr @omr_dst_bypass_eth2_4 meta mark set 0x45391003 accept
nft insert rule inet fw4 ss_rules_local_out ip daddr @omr_dst_bypass_eth2_4 meta mark set 0x45391003 accept
nft insert rule inet fw4 ss_rules_pre_tcp ip daddr @omr_dst_bypass_tun0_4 meta mark set 0x45391500 accept
nft insert rule inet fw4 ss_rules_local_out ip daddr @omr_dst_bypass_tun0_4 meta mark set 0x45391500 accept
```
With this correction traffic from lan to 1.1.1.2 goes via right interface. However traffic from omr itself still doesn't go right way. For this some extra rules are required in /etc/firewall.omr-bypass

Also adding other types of rules in omr-bypass, for example Ports source rules doest enable correspondent firewall rule in /etc/config/firewall. I have to put option enabled '1' for this rule manually

rdmitry0911, 2024-03-06:
I've just checked /etc/init.d/omr-bypass and found inconsistency in naming with /etc/config/firewall. In /etc/config/firewall there are separate rules for ipv4 and ipv6 For example there is such a rule:
```
config rule 'omr_dst_bypass_eth2_srcport_tcp_4'
        option name 'omr_dst_bypass_eth2_srcport_tcp_4'
        option proto 'tcp'
        option src 'lan'
        option dest '*'
        option target 'MARK'
        option enabled '1'
        option set_xmark '0x45391003'
        list src_port '22'
```
while in /etc/init.d/omr-bypass there is no such separation in correspendent code:

```
_bypass_src_port() {
.
.
.
                uci -q batch <<-EOF
                        add_list firewall.omr_dst_bypass_${intf}_dstport_tcp.src_port="$sport"
                        set firewall.omr_dst_bypass_${intf}_dstport_tcp.enabled='1'
                EOF

```
this piece of code will try to setup and enable omr_dst_bypass_eth2_dstport_tcp rule which does't exist. To be consistent with firewall rules it should look like this (btw this specific example is messing up dest and src, which is, probably, copy/paste bug) :
```
_bypass_src_port() {
.
.
.
                uci -q batch <<-EOF
                        add_list firewall.omr_dst_bypass_${intf}_srcport_tcp_4.src_port="$sport"
                        set firewall.omr_dst_bypass_${intf}_srcport_tcp_4.enabled='1'
                EOF

```
Or, we can go opposit direction and rules in /etc/config/firewall shouldn't be separated by ipv4 and ipv6 to be consistent with /etc/init.d/omr-bypass. This way, probably, is more easy to implement as omr-bypass luci interface doesn't have ipv4/ipv6 selectors in most bypass rules

rdmitry0911, 2024-03-06:
Where can I get the fix?

rdmitry0911, 2024-03-07:
Ok, I managed to build image from the developer branch with the latest commits and found out that you made substantial changes to omr-bypass logic. Now I see the correct rules in nft tables and bypassed traffic goes via right routes. Will see how it works in more complex configurations

rdmitry0911, 2024-03-07:
Still not all the problems are resolved. When I added another interfaces eth4 and eth5 for bypassing it got the same metric for them as for the previous one. And traffic rules messed up. now my rules look like this:
```
root@OpenMPTCProuter:~# ip ru sh
0:	from all lookup local
0:	from 192.168.50.100 lookup 2
0:	from 192.168.82.151 lookup 1
0:	from all oif eth5 lookup 1
0:	from all fwmark 0x1 lookup 100
0:	from all oif eth2 lookup 3
0:	from all oif tun0 lookup 500
0:	from all oif eth1 lookup 2
0:	from 192.168.80.234 lookup 1
0:	from all oif eth3 lookup 1
0:	from 192.168.81.177 lookup 1
0:	from 192.168.60.100 lookup 3
0:	from 10.255.252.2 lookup 500
1:	from all fwmark 0x45392 lookup 2
1:	from all fwmark 0x45393 lookup 3
1:	from all fwmark 0x4539500 lookup 500
1:	from all fwmark 0x4539 lookup 991337
1:	from all fwmark 0x45391 lookup 1
100:	from all lookup lan
10000:	from 192.168.99.1 lookup lan
20000:	from all to 192.168.99.1/24 lookup lan
32766:	from all lookup main
32767:	from all lookup default
90002:	from all iif lo lookup lan
```
omr-bypass config looks like this:
```
root@OpenMPTCProuter:~# cat /etc/config/omr-bypass 

config interface 'all'

config proto 'm6replay'
	list url 'm6web.fr'
	list url '6play.fr'
	list url '6cloud.fr'

config proto 'mycanal'
	list url 'mycanal.fr'
	list url 'canal-plus.com'
	list url 'canalplus.com'
	list url 'canalplus-cdn.net'
	list url 'canalplus.pro'
	list url 'canal-plus.net'

config proto 'minecraft'
	list url 'authserver.mojang.com'

config proto 'lesnumeriques'
	list url 'lesnumeriques.com'
	list url 'botscorner.com'
	list url 'app.botscorner.com'

config proto 'disneyplus'
	list url 'bamgrid.com'
	list url 'disney-plus.net'

config proto 'amazonvideo'
	list url 'cloudfront.net'
	list url 'llnw.net'

config proto 'free'
	list url 'free.fr'
	list url 'freebox.fr'
	list url 'oqee.tv'
	list url 'oqee.net'

config proto 'orange'
	list url 'orange.fr'
	list url 'sosh.fr'
	list url 'liveperson.net'
	list url 'liveperson.com'
	list url 'lpsn.net'
	list url 'lpsnmedia.net'
	list url 'francetelecom.fr'

config interface 'eth0'
	option id '9999'

config interface 'eth1'
	option id '2'

config interface 'eth2'
	option id '3'

config interface 'tun0'
	option id '500'

config global 'global'
	option vpn_ipv4_md5 '68b329da9893e34099c7d8ad5cb9c940'
	option vpn_ipv6_md5 '68b329da9893e34099c7d8ad5cb9c940'

config interface 'eth3'
	option id '1'

config ips
	option ip '1.1.1.2'
	option interface 'eth3'

config ips
	option ip '1.1.1.3'
	option interface 'eth4'

config src_port
	option sport '33'
	option proto 'tcp'
	option interface 'eth3'

config dest_port
	option dport '22'
	option proto 'tcp'
	option interface 'eth3'

config interface 'eth4'
	option id '1'

config interface 'eth5'
	option id '1'

config ips
	option ip '1.1.1.4'
	option interface 'eth5'
```
As you can see there are many interfaces with id=1 in omr-bypass config file. And strange thing there is no reference to eth4 in ip rules

Ysurac, 2024-03-07:
What is the result of ```uci show network``` ?

rdmitry0911, 2024-03-07:
It's here:
```
root@OpenMPTCProuter:~# uci show network
network.loopback=interface
network.loopback.proto='static'
network.loopback.ipaddr='127.0.0.1'
network.loopback.netmask='255.0.0.0'
network.loopback.multipath='off'
network.loopback.device='lo'
network.loopback.metric='1000'
network.globals=globals
network.globals.ula_prefix='fdcf:f4c9:d08e::/48'
network.globals.multipath='enable'
network.globals.mptcp_path_manager='fullmesh'
network.globals.mptcp_scheduler='blest'
network.globals.congestion='bbr'
network.globals.mptcp_checksum='0'
network.globals.mptcp_debug='0'
network.globals.mptcp_syn_retries='4'
network.globals.mptcp_subflows='8'
network.globals.mptcp_add_addr_accepted='1'
network.globals.mptcp_add_addr_timeout='120'
network.globals.mptcp_pm_type='0'
network.globals.mptcp_disable_initial_config='0'
network.globals.mptcp_force_multipath='1'
network.globals.mptcpd_enable='0'
network.globals.mptcp_fullmesh_num_subflows='1'
network.globals.mptcp_fullmesh_create_on_err='1'
network.globals.mptcp_ndiffports_num_subflows='1'
network.lan=interface
network.lan.proto='static'
network.lan.ipaddr='192.168.99.1'
network.lan.netmask='255.255.255.0'
network.lan.device='eth0'
network.lan.ifname='eth0'
network.lan.metric='9999'
network.lan.ipv6='0'
network.lan.delegate='0'
network.lan.addlatency='0'
network.lan.txqueuelen='2000'
network.lan.multipath='off'
network.lan.ip4table='lan'
network.lan.label='lan'
network.lan.defaultroute='0'
network.lan.peerdns='0'
network.lan_rule=rule
network.lan_rule.lookup='lan'
network.lan_rule.priority='100'
network.wan1=interface
network.wan1.device='eth1'
network.wan1.proto='static'
network.wan1.ip4table='wan'
network.wan1.multipath='master'
network.wan1.defaultroute='0'
network.wan1.delegate='0'
network.wan1.addlatency='0'
network.wan1.metric='2'
network.wan1.peerdns='0'
network.wan1.label='wan1'
network.wan1.ipv6='0'
network.wan1.ipaddr='192.168.50.100'
network.wan1.netmask='255.255.255.0'
network.wan1.gateway='192.168.50.1'
network.wan3=interface
network.wan3.device='eth3'
network.wan3.proto='dhcp'
network.wan3.ip4table='wan'
network.wan3.multipath='off'
network.wan3.defaultroute='0'
network.wan3.metric='1'
[... 139 lines cut ...]
network.eth4_fw_rule6=rule6
network.eth4_fw_rule6.priority='1'
network.eth4_fw_rule6.mark='0x65391'
network.eth4_fw_rule6.lookup='1'
network.wan8_dev=device
network.wan8_dev.name='@us'
network.wan7_dev=device
network.wan7_dev.name='@ru'
network.wan6_dev=device
network.wan6_dev.name='@us'
network.wan5_dev=device
network.wan5_dev.name='eth5'
network.wan4_dev=device
network.wan4_dev.name='eth4'
network.wan3_dev=device
network.wan3_dev.name='eth3'
network.eth5_fw_rule=rule
network.eth5_fw_rule.priority='1'
network.eth5_fw_rule.mark='0x45391'
network.eth5_fw_rule.lookup='1'
network.eth5_fw_rule6=rule6
network.eth5_fw_rule6.priority='1'
network.eth5_fw_rule6.mark='0x65391'
network.eth5_fw_rule6.lookup='1'
```

Ysurac, 2024-03-07:
I would also need ```uci show openmptcprouter.wan1.metric``` to ```uci show openmptcprouter.wan5.metric```

rdmitry0911, 2024-03-07:
Here they are:
```
root@OpenMPTCProuter:~# uci show openmptcprouter.wan1.metric
openmptcprouter.wan1.metric='2'
root@OpenMPTCProuter:~# uci show openmptcprouter.wan2.metric
openmptcprouter.wan2.metric='3'
root@OpenMPTCProuter:~# uci show openmptcprouter.wan3.metric
openmptcprouter.wan3.metric='1'
root@OpenMPTCProuter:~# uci show openmptcprouter.wan4.metric
openmptcprouter.wan4.metric='1'
root@OpenMPTCProuter:~# uci show openmptcprouter.wan5.metric
openmptcprouter.wan5.metric='1'
```

rdmitry0911, 2024-03-09:
I also noticed, that if I delete an interface, for example wan3 and then add a new one using settings wizard, the script will assign a next number to it. In my example it will be wan4. wan3 will never be assigned again. It would be better if the script fills the gaps in numbering

rdmitry0911, 2024-03-09:
I made omr configuration from the ground up. A new version of mptcp commited yesterday generates another, but still wrong id for the interfaces listed in omr-bypass:
```
root@OpenMPTCProuter:~# cat /etc/config/omr-bypass

config interface 'all'

config proto 'm6replay'
	list url 'm6web.fr'
	list url '6play.fr'
	list url '6cloud.fr'

config proto 'mycanal'
	list url 'mycanal.fr'
	list url 'canal-plus.com'
	list url 'canalplus.com'
	list url 'canalplus-cdn.net'
	list url 'canalplus.pro'
	list url 'canal-plus.net'

config proto 'minecraft'
	list url 'authserver.mojang.com'

config proto 'lesnumeriques'
	list url 'lesnumeriques.com'
	list url 'botscorner.com'
	list url 'app.botscorner.com'

config proto 'disneyplus'
	list url 'bamgrid.com'
	list url 'disney-plus.net'

config proto 'amazonvideo'
	list url 'cloudfront.net'
	list url 'llnw.net'

config proto 'free'
	list url 'free.fr'
	list url 'freebox.fr'
	list url 'oqee.tv'
	list url 'oqee.net'

config proto 'orange'
	list url 'orange.fr'
	list url 'sosh.fr'
	list url 'liveperson.net'
	list url 'liveperson.com'
	list url 'lpsn.net'
	list url 'lpsnmedia.net'
	list url 'francetelecom.fr'

config interface 'eth0'
	option id '9999'

config interface 'eth1'
	option id '2'

config interface 'eth2'
	option id '3'

config interface 'tun0'
	option id '1500'

config global 'global'
	option vpn_ipv4_md5 '68b329da9893e34099c7d8ad5cb9c940'
	option vpn_ipv6_md5 '68b329da9893e34099c7d8ad5cb9c940'

config interface 'eth3'
	option id '1'

config interface 'eth4'
	option id '2'

config interface 'eth5'
	option id '2'

```

closed 2024-06-12 by github-actions[bot]