Ysurac/openmptcprouter #3197: Challenges with Achieving Expected Aggregate Throughput

Yessine1200, 2024-03-06:
## Current Behavior

Hi, I expected to achieve an aggregate throughput of 100Mbps, but I didn't achieve that. I have two 4G modems and OpenMPTCProuter installed in a virtual machine. The configuration was done correctly, but I'm unsure what's causing the issue. I'm using Glorytun TCP as the default VPN. Additionally, I have two servers, and I tried to test server failover, but it didn't work. How can I fix that?

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.60rc1-6.1 r0+24
  - OpenMPTCProuter VPS version: 6.1.0-18-amd64 0.1029-test
  - OpenMPTCProuter VPS provider: OVH
  - OpenMPTCProuter platform: x68_64

Ysurac, 2024-03-06:
Several issues here, first try to solve the speed issue.
What do you have on System->OpenMPTCProuter, "Status" page ?
How do you test speed ? Do you use ```omr-test-speed``` via SSH on the router ?

Yessine1200, 2024-03-06:
1. For the status, I have:
[image]
2. I tested the speed using Speedtest on the internet.

Ysurac, 2024-03-06:
You should test with ```omr-test-speed eth1```, ```omr-test-speed eth2``` and ```omr-test-speed``` (stop after about 2 minutes with Ctrl+C)

Yessine1200, 2024-03-06:
I got this error:
[image]

Yessine1200, 2024-03-06:
I just used omr-test-speed, and I got a very low speed.
[image]

Yessine1200, 2024-03-07:
I got this error when testing on the two interfaces of my routers. I don't understand what the problem is; it seems to be firewall-related.
[image]
I also ran omr-test-speed on the VPS, and I got this result. It seems weird because the throughput of my VPS is 1Gbps:
[image]

Ysurac, 2024-03-07:
What is the result of ```omr-test-speed-server``` from the router via SSH ?

Yessine1200, 2024-03-07:
It's weird also:
[image]

Yessine1200, 2024-03-07:
still have the problem, and I don't know how to fix it. The configuration seems okay, and the status page also shows all green.

Ysurac, 2024-03-07:
Can you run ```omr-test-speed-server eth1``` and ```omr-test-speed-server eth2``` ?
Can you also give me Status->System log ?

Yessine1200, 2024-03-07:
I ran omr-test-speed-server eth1: I got nothing, you mean omr-test-speed eth1 ?
**System log:**
Mar  7 15:45:52 OpenMPTCProuter syslog.info syslogd started: BusyBox v1.36.1
Mar  7 15:45:52 OpenMPTCProuter user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Mar  7 15:45:52 OpenMPTCProuter user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Mar  7 15:45:52 OpenMPTCProuter user.notice dnsmasq: Allowing RFC1918 responses for domain plex.direct
Mar  7 15:45:52 OpenMPTCProuter authpriv.info dropbear[3739]: Not backgrounding
Mar  7 15:45:52 OpenMPTCProuter daemon.info dnsmasq[1]: started, version 2.89 cachesize 1000
Mar  7 15:45:52 OpenMPTCProuter daemon.info dnsmasq[1]: DNS service limited to local subnets
Mar  7 15:45:52 OpenMPTCProuter daemon.info dnsmasq[1]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack no-ipset nftset auth cryptohash DNSSEC no-ID loop-detect inotify dumpfile
Mar  7 15:45:52 OpenMPTCProuter daemon.info dnsmasq[1]: UBus support enabled: connected to system bus
Mar  7 15:45:52 OpenMPTCProuter daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5353
Mar  7 15:45:52 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for test
Mar  7 15:45:52 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for onion
Mar  7 15:45:52 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for localhost
Mar  7 15:45:52 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for local
Mar  7 15:45:52 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for invalid
Mar  7 15:45:52 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for bind
Mar  7 15:45:52 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for use-application-dns.net
Mar  7 15:45:52 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for lan
Mar  7 15:45:52 OpenMPTCProuter daemon.info dnsmasq[1]: using 1 more local addresses
Mar  7 15:45:52 OpenMPTCProuter daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Mar  7 15:45:52 OpenMPTCProuter daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 0 names
Mar  7 15:45:52 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Mar  7 15:45:52 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Failed to parse json data: unexpected character
Mar  7 15:45:52 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Failed to parse json data: unexpected character
Mar  7 15:45:52 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Failed to parse json data: unexpected character
Mar  7 15:45:52 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Failed to parse json data: unexpected character
Mar  7 15:45:52 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Failed to parse json data: unexpected character
Mar  7 15:45:52 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Failed to parse json data: unexpected character
Mar  7 15:45:52 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S25packet_steering: uci: Entry not found
Mar  7 15:45:52 OpenMPTCProuter user.notice : Added device handler type: bonding
Mar  7 15:45:52 OpenMPTCProuter user.notice : Added device handler type: 8021ad
Mar  7 15:45:52 OpenMPTCProuter user.notice : Added device handler type: 8021q
Mar  7 15:45:52 OpenMPTCProuter user.notice : Added device handler type: macvlan
Mar  7 15:45:52 OpenMPTCProuter user.notice : Added device handler type: veth
Mar  7 15:45:52 OpenMPTCProuter user.notice : Added device handler type: bridge
Mar  7 15:45:52 OpenMPTCProuter user.notice : Added device handler type: Network device
Mar  7 15:45:52 OpenMPTCProuter user.notice : Added device handler type: tunnel
Mar  7 15:45:52 OpenMPTCProuter daemon.notice wpa_supplicant[4124]: Successfully initialized wpa_supplicant
Mar  7 15:45:52 OpenMPTCProuter daemon.warn netifd: You have delegated IPv6-prefixes but haven't assigned them to any interface. Did you forget to set option ip6assign on your lan-interfaces?
Mar  7 15:45:53 OpenMPTCProuter daemon.notice netifd: Interface 'lan' is enabled
Mar  7 15:45:53 OpenMPTCProuter daemon.notice netifd: Interface 'lan' is setting up now
Mar  7 15:45:53 OpenMPTCProuter daemon.notice netifd: Interface 'lan' is now up
Mar  7 15:45:53 OpenMPTCProuter daemon.notice netifd: Interface 'wan1' is enabled
Mar  7 15:45:53 OpenMPTCProuter daemon.notice netifd: Interface 'wan1' is setting up now
Mar  7 15:45:53 OpenMPTCProuter daemon.notice netifd: Interface 'wan1' is now up
Mar  7 15:45:53 OpenMPTCProuter daemon.notice netifd: Interface 'wan2' is enabled
Mar  7 15:45:53 OpenMPTCProuter daemon.notice netifd: Interface 'wan2' is setting up now
Mar  7 15:45:53 OpenMPTCProuter daemon.notice netifd: Interface 'wan2' is now up
Mar  7 15:45:53 OpenMPTCProuter daemon.notice netifd: Interface 'loopback' is enabled
Mar  7 15:45:53 OpenMPTCProuter daemon.notice netifd: Interface 'loopback' is setting up now
Mar  7 15:45:53 OpenMPTCProuter daemon.notice netifd: Interface 'loopback' is now up
Mar  7 15:45:53 OpenMPTCProuter daemon.notice netifd: Network device 'lo' link is up
Mar  7 15:45:53 OpenMPTCProuter daemon.notice netifd: Interface 'loopback' has link connectivity 
Mar  7 15:45:53 OpenMPTCProuter user.notice MPTCP: Set wan1 to on
[... cut ...]
Mar  7 15:48:29 OpenMPTCProuter user.notice SQM-autorate: Set config for omrvpn
Mar  7 15:49:00 OpenMPTCProuter user.notice omr-schedule-010-services: Can't find omr-bypass rules, restart omr-bypass...
Mar  7 15:49:00 OpenMPTCProuter user.notice omr-bypass: Starting OMR-ByPass...
Mar  7 15:49:01 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Mar  7 15:49:01 OpenMPTCProuter user.notice omr-bypass: OMR-ByPass is running
Mar  7 15:49:06 OpenMPTCProuter user.notice omr-schedule-010-services: Set firewall on server vps

Ysurac, 2024-03-07:
What are the modems ? Some USB keys or ethernet modems ?
There is many up/down of all interfaces in the log.
Can you try a ```ping -B -I eth1 1.1.1.1``` and check if you get data for a few minutes ? Same with eth2

Yessine1200, 2024-03-08:
These are Ethernet modems. The up/down events in the log are due to my failover connection tests. For the ping test, yes, I receive data for both eth1 and eth2. Perhaps the issue lies with the VM where OMR is installed? Indeed, when I run omr-speed-test-server, I get a strange result.

Yessine1200, 2024-03-08:
And as you see here, with omr-test-speed I got only 111Mbps and with normal speedtest I got the right result:
[image]
What can be the problem ?

Ysurac, 2024-03-08:
omr-test-speed give result in MB, so 111*8=888Mbps a bit lower but not so much.

Yessine1200, 2024-03-08:
So what can be the problem ?

Ysurac, 2024-03-08:
I didn't purchase the RUTX50, maybe coming from Amazon.
I would not recommend it for OpenMPTCProuter in most case (or only to use it as a modem).

Yessine1200, 2024-03-08:
But in your documentation, you said that OMR in kernel 6.1 support RUTX50, right ?

Ysurac, 2024-03-08:
Yes it's supported. Not widely tested and still a quite slow device.

Yessine1200, 2024-03-08:
So, you don't recommend to use it ?
For server failover, How configure OMR ?

Ysurac, 2024-03-08:
No.
For server failover, you need to add a server in System->OpenMPTCProuter, "Wizard" tab and set his key. The backup server must answer to ping.

Yessine1200, 2024-03-08:
I added the backup server in the wizard tab. I can ping the backup server via SSH on the router. However, when I shut down the main server, OMR doesn't switch to the backup server; instead, it goes directly Direct Output.

Ysurac, 2024-03-08:
I would need Status->System log again when you switch off master server

Yessine1200, 2024-03-08:
13:49:03 OpenMPTCProuter daemon.err glorytun[23563]: 54.38.xxxxSSss.65001: connected
Mar  8 13:49:03 OpenMPTCProuter daemon.err glorytun[23563]: couldn't set socket option `TCP_KEEPCNT'
Mar  8 13:49:03 OpenMPTCProuter daemon.err glorytun[23563]: couldn't set socket option `TCP_KEEPIDLE'
Mar  8 13:49:03 OpenMPTCProuter daemon.err glorytun[23563]: couldn't set socket option `TCP_KEEPINTVL'
Mar  8 13:49:03 OpenMPTCProuter daemon.err glorytun[23563]: couldn't set socket option `TCP_USER_TIMEOUT'
Mar  8 13:49:04 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Mar  8 13:49:04 OpenMPTCProuter daemon.err glorytun[23563]: 54.38.xxxxx.65001: key exchange failed
Mar  8 13:49:04 OpenMPTCProuter daemon.info glorytun[23563]: STOPPED tun0

Yessine1200, 2024-03-08:
Maybe I have to add rules in shadowsocks-rust for the backup server ?

Ysurac, 2024-03-08:
no, but this log is too short and doesn't have the good part. It's "OMR-Tracker-Server" that do all the job.

Yessine1200, 2024-03-08:
Here is the entire log starting from when the VPS is shutdown:
OMR-Tracker-Server can see the backup server, but it can't failover to it.
8 14:48:22 OpenMPTCProuter daemon.info glorytun[3496]: STOPPED tun0
Mar  8 14:48:58 OpenMPTCProuter user.notice post-tracking-001-post-tracking: omrvpn down because gateway down, interface have no IPv6, interface have no IPv6 gateway
Mar  8 14:48:59 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Glorytun VPN down, restart it
Mar  8 14:48:59 OpenMPTCProuter daemon.err glorytun[3496]: getpeername: Socket not connected
Mar  8 14:48:59 OpenMPTCProuter daemon.notice ttyd[657]: [2024/03/08 14:48:59:0404] N: rops_handle_POLLIN_netlink: DELADDR
Mar  8 14:48:59 OpenMPTCProuter daemon.notice netifd: Network device 'tun0' link is down
Mar  8 14:48:59 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' has link connectivity loss
Mar  8 14:48:59 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is now down
Mar  8 14:48:59 OpenMPTCProuter daemon.notice ttyd[657]: [2024/03/08 14:48:59:0448] N: rops_handle_POLLIN_netlink: DELADDR
Mar  8 14:48:59 OpenMPTCProuter daemon.info glorytun: starting glorytun vpn instance vpn
Mar  8 14:48:59 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is disabled
Mar  8 14:48:59 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is enabled
Mar  8 14:48:59 OpenMPTCProuter daemon.notice netifd: Network device 'tun0' link is up
Mar  8 14:48:59 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' has link connectivity 
Mar  8 14:48:59 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is setting up now
Mar  8 14:48:59 OpenMPTCProuter daemon.info glorytun[4057]: INITIALIZED tun0
Mar  8 14:48:59 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is now up
Mar  8 14:48:59 OpenMPTCProuter user.notice firewall: Reloading firewall due to ifup of omrvpn (tun0)
Mar  8 14:48:59 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Mar  8 14:48:59 OpenMPTCProuter daemon.notice ttyd[657]: [2024/03/08 14:48:59:6005] N: rops_handle_POLLIN_netlink: DELADDR
Mar  8 14:49:00 OpenMPTCProuter user.notice MPTCP: Set eth1 to master
Mar  8 14:49:00 OpenMPTCProuter user.notice MPTCP: Set eth2 to on
Mar  8 14:49:03 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Change in routes, set ip route replace default scope global nexthop via 192.168.3.1 dev eth1 weight 100 nexthop via 192.168.1.1 dev eth2 weight 1 (omrvpn_intf: tun0)
Mar  8 14:49:06 OpenMPTCProuter daemon.info omr-tracker-ss: Shadowsocks sss0 is down (can't contact via http 77.88.55.77, 1.1.1.1, 74.82.42.42)
Mar  8 14:49:07 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Mar  8 14:49:13 OpenMPTCProuter daemon.info omr-tracker-ss: Server sss0 (193.169.104.0) seems down, no answer to ping
Mar  8 14:49:15 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can't ping server vps on 193.169.104.0, no server API answer on 193.169.104.0)
Mar  8 14:49:20 OpenMPTCProuter user.notice OMR-Tracker-Server: Master server vps down (193.169.104.0)
Mar  8 14:49:21 OpenMPTCProuter user.notice OMR-Tracker-Server: Use backup server  (54.38.70.10)
Mar  8 14:49:25 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can't ping server vps on 193.169.104.0, no server API answer on 193.169.104.0)
Mar  8 14:49:29 OpenMPTCProuter daemon.notice netifd: Network device 'tun0' link is down
Mar  8 14:49:29 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' has link connectivity loss
Mar  8 14:49:29 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is now down
Mar  8 14:49:29 OpenMPTCProuter daemon.err glorytun[4057]: getpeername: Socket not connected
Mar  8 14:49:29 OpenMPTCProuter daemon.notice ttyd[657]: [2024/03/08 14:49:29:8344] N: rops_handle_POLLIN_netlink: DELADDR
Mar  8 14:49:29 OpenMPTCProuter daemon.notice ttyd[657]: [2024/03/08 14:49:29:8345] N: rops_handle_POLLIN_netlink: DELADDR
Mar  8 14:49:29 OpenMPTCProuter daemon.info glorytun: starting glorytun vpn instance vpn
Mar  8 14:49:29 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is disabled
Mar  8 14:49:29 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is enabled
Mar  8 14:49:29 OpenMPTCProuter daemon.notice netifd: Network device 'tun0' link is up
Mar  8 14:49:29 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' has link connectivity 
Mar  8 14:49:29 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is setting up now
Mar  8 14:49:29 OpenMPTCProuter daemon.info glorytun[16858]: INITIALIZED tun0
Mar  8 14:49:29 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is now up
Mar  8 14:49:30 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Mar  8 14:49:30 OpenMPTCProuter user.notice firewall: Reloading firewall due to ifup of omrvpn (tun0)
Mar  8 14:49:30 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Mar  8 14:49:31 OpenMPTCProuter daemon.err glorytun[16858]: 54.38.70.10.65001: connected
Mar  8 14:49:31 OpenMPTCProuter daemon.err glorytun[16858]: couldn't set socket option `TCP_KEEPCNT'
Mar  8 14:49:31 OpenMPTCProuter daemon.err glorytun[16858]: couldn't set socket option `TCP_KEEPIDLE'
Mar  8 14:49:31 OpenMPTCProuter daemon.err glorytun[16858]: couldn't set socket option `TCP_KEEPINTVL'
[... cut ...]
Mar  8 14:52:49 OpenMPTCProuter daemon.err glorytun[16858]: couldn't set socket option `TCP_KEEPCNT'
Mar  8 14:52:49 OpenMPTCProuter daemon.err glorytun[16858]: couldn't set socket option `TCP_KEEPIDLE'
Mar  8 14:52:49 OpenMPTCProuter daemon.err glorytun[16858]: couldn't set socket option `TCP_KEEPINTVL'
Mar  8 14:52:49 OpenMPTCProuter daemon.err glorytun[16858]: couldn't set socket option `TCP_USER_TIMEOUT'
Mar  8 14:52:49 OpenMPTCProuter daemon.err glorytun[16858]: 54.38.70.10.65001: key exchange failed
Mar  8 14:52:49 OpenMPTCProuter daemon.info glorytun[16858]: STOPPED tun0

Ysurac, 2024-03-08:
It correctly change to backup server but can't connect.
When you set backup server as master in wizard, all is working correctly ?

Yessine1200, 2024-03-08:
Yes, works correctly.

Yessine1200, 2024-03-11:
I ran the two commands on both servers:
1. For the first command, I obtained the key of glorytun for each server.
2. For the second command, I received information about shadowsocks. The weird thing I noticed is that on the main server, I got duplicate information on port_key.

Ysurac, 2024-03-11:
You get same key for glorytun-tcp on both server ?
You mean the port is duplicated with same key for shadowsocks ? it's normal because Shadowsocks-libev is not multithreaded so it's launched on each CPU.

Yessine1200, 2024-03-11:
No, I didn't get the same key for glorytun-tcp on both servers. Should they be the same?

Ysurac, 2024-03-11:
Should be better with latest snapshot.

Yessine1200, 2024-03-11:
OK, I will try it and get back to you if there is any issue. Thank you.

Yessine1200, 2024-03-12:
It works now for the latest snapshot, but it takes a while to switch to the backup server (nearly 10 seconds), and I lost connection. Is it supposed to be like that? Also, when the main server comes back online, OMR does not return to the main server?

Yessine1200, 2024-03-27:
> I didn't purchase the RUTX50, maybe coming from Amazon. I would not recommend it for OpenMPTCProuter in most case (or only to use it as a modem).

Hello Ysurac,
And for the Rutx12, does the latest version of OMR (0.6rc) now support it?

Ysurac, 2024-03-27:
No. My RUTX12 is died so I can't test and make it work on it

Yessine1200, 2024-03-29:
I have another question: Can OMR also aggregate private 5G connections?

Ysurac, 2024-03-29:
OMR can aggregate any connection if this connection has access to the VPS/Server part.
What you mean with private 5G connections ?

Yessine1200, 2024-03-29:
Private 5G is a platform that you can deploy it by yourself using 5G technology, exclusively owned, managed, and controlled by you. Tools such as OpenAirInterface can facilitate its deployment. So, you have a private 5G which you can connect to it smartphones, tablets and others.

Ysurac, 2024-03-29:
ok, thanks for the information, seems fun to try.
So if the network have access to the same VPS IP as others connections, no problem.

Yessine1200, 2024-04-02:
Hi Ysurac,
Are you aware of any miniPCs in which I could put 4G and 5G SIM cards to put OMR inside and have a single box ?

closed 2024-08-20 by github-actions[bot]