Ysurac/openmptcprouter #3198: Public IPs from Cloud provider routed to cisco router

zee-shany, 2024-03-07:
hello,

Thanks for this great project and enhancment.

I'd like to check if its possible to route a /30 public IP block and /29 public IP block to a Router device after the OpenMPTCP router as shown in diagram below? assuming i'm able to get the /30 and /29 block of IPs from AWS.

[image]

Thanks in advance for your response.

ccmks, 2024-03-13:
I believe by default, OMR is using NAT mode. Where did you get the IP address from?

Ysurac, 2024-03-13:
You should be able to use a GRE tunnel for the full subnet.
Or, easier, set all /29 IPs on the VPS on the same interface, then you can with Shadowsocks-libev and the GRE tunnels that are automatically created use them: https://github.com/Ysurac/openmptcprouter/wiki/VPS-Multi-IPs

closed 2024-06-17 by github-actions[bot]