Ysurac/openmptcprouter #3202: Can't get public IP from ShadowSocks Rust

GermanAizek, 2024-03-10:
@Ysurac, I try similar issue, but not found. I hope I'm not making duplicate.

Ysurac, 2024-03-10:
Please use the template.

GermanAizek, 2024-03-10:
I recheck, but again. I enabled all local instances.
[image]
[image]
[image]
[image]

GermanAizek, 2024-03-11:
@Ysurac, is it possible to get technical journal service? something like `systemctl status shadowsocks-rust`
I setup everything as it was written in Wiki->[Minimum configuration](https://github.com/Ysurac/openmptcprouter/wiki/OpenMPTCProuter-configuration)

closed 2024-06-14 by github-actions[bot]

mvmaksnet, 2025-03-06:
I have the same problem but on version:
  - OpenMPTCProuter version: OpenMPTCProuter v0.62
  - OpenMPTCProuter VPS version: 0.1032-test 6.12.15-x64v3-xanmod1
  - OpenMPTCProuter VPS provider: Maksnet
  - OpenMPTCProuter platform:  x86_64

What can be problem?

Ysurac, 2025-03-06:
The version 0.62 is not available. Only v0.62rc1.
This issue is closed, open a new issue with a screenshot of status page.

mvmaksnet, 2025-03-06:
I have made new issue ticket. Thanks.